Anime4000/RTL960x #105: DFP-34X-2C2 & Mikrotik hEX s unable to get IP EPON

korzhyk, 2022-12-07:
Hello, I faced with problem when my Mikrotik unable to sent DHCP IP Request through SFP interface. SFP ONU connection is established (due PON packets stats) but in PON stats I see 0 unicast messages sent, but has a lot of received.

TheKhack, 2022-12-07:
where u get this MAC_KEY?

korzhyk, 2022-12-07:
Keygen from https://gist.github.com/rajkosto/29c513b96ea6262d2fb1f965a52ce16f

TheKhack, 2022-12-07:
hsgq1.9aMAC_ADDR_UPPERCASE 
maybe you type wrong mac

dexxter00, 2022-12-08:
Which provider is this? Your ISP probably uses VLAN tagging, please check MIB 84
`omcicli mib get 84`

korzhyk, 2022-12-08:
@dexxter00 unfortently `omcicli mib get 84` return empty result on both ONUs and I'm my ISP (Lanet, Ukraine) uses EPON.

```
telnet 192.168.0.1
Trying 192.168.0.1...
Connected to 192.168.0.1.
Escape character is '^]'.
login: admin
Password: 

BusyBox v1.12.4 (2017-02-20 18:14:44 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# omcicli mib get 84
# 
# 

```

korzhyk, 2022-12-09:
Finally, got IP from ISP.

As I know from tech support of my ISP there only ONU MAC address auth on OLT side. So in my case I need only to change `ELAN_MAC_ADDR` and `MAC_KEY` to get my ONU registered. But also need to manually set `PVID` in VLAN settings. Actual `PVID` is a value (**2903**) from l2-table from Picotel `PU-E510`. After this change I finally get Unicast packets received and IP from ISP DHCP.
[image]

korzhyk, 2022-12-29:
UPD2: LLID also must be changed in this case IPS can monitor your SFP ONU, otherwise your ONU will be like a ghost even with Internet access.

closed 2022-12-29 by korzhyk

Anime4000, 2022-12-29:
good find

payk24, 2023-01-07:
Hi, @korzhyk, can you help me, please?
Could you explain to me how to set LLID?

korzhyk, 2023-01-07:
@payk24 you need FW version "[M114_sfp_ODI_Vlan_220414.tar](https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M114_sfp_ODI_Vlan_220414.tar)" there is EPON settings menu that allows to change LLID (mac addr).