Anime4000/RTL960x #107: Explanation of OMCI_VEIP_SLOT_ID flash option is invalid

rajkosto, 2022-12-14:
https://github.com/Anime4000/RTL960x/blob/main/Docs/FLASH_GETSET_INFO.md#omci_veip_slot_id here it implies that it controls the VEIP entityId, and that you should set it to 0x0E01 if that's the entityId of your original ONU, but when this value is read it is read as an uint8 so it can only be 0-255, and it also does NOT control the VEIP entityId, which is always 0x0601 (instead it controls a custom option inside the veip MIB data, which we have no idea what it does, as its not part of standard VEIP mib)

rajkosto, 2022-12-14:
this DOES control the veip entityID ! but the description is woefully incomplete.
first off, the value should be set to the high byte of the wanted entityId, so for 0x0e01 it should be 0x0e or 14 (flash set will only accept 14)
second, the feature that actually applies the hack is not enabled by default, to enable it you must set bit 0x100 in OMCI_CUSTOM_ME to enable the `cf_apply_customized_veip_slot_id` feature, the default value in SFU firmwares for OMCI_CUSTOM_ME in SFU firmwares is 65536, so set it to 65792 to enable the feature.

et voila:

```
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VEIP
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0e01
AdminState: 1
OperState: 1
InterDomainName: 0x00000000000000000000000000000000000000000000000000
TcpUdpPtr: 0x0000
IansAssignPort: 0
CustomizedSlotId: 0x0e
=================================
```

Anime4000, 2022-12-15:
great finding!
how you get VEIP from stock ONT?
mine just from Huawei XML file

rajkosto, 2022-12-15:
omci log opened in wireshark and look for veip

rajkosto, 2022-12-15:
You are able to use treated VLAN in IGD firmware ? Which one ? 
In both IGD and SFU firmware, with VEIP entityid unchanged (0x601) or changed (0xe01), i HAVE to use the "weird" untreated VLAN and not the normal one my ISP ONT uses, in IGD both in bridged mode and even if the stick itself is doing PPP, i cannot use treated vlan id on the other side of the bridge or inside the webgui, only weird vlan.

S8T8, 2022-12-15:
Not an expert like you but my problem seems to be similar, if you want to check more [https://github.com/Anime4000/RTL960x/issues/90](url)
In my case case ISP VLAN is 835, Treatment Inner 3310, IGD firmware is working with VLAN 835, SFU with VLAN 3310.
I can do some tests with latest SFU firmware.

S8T8, 2022-12-15:
Never tried to get PPPoE to the ODI SFP, always to the router but confident that will work

rajkosto, 2022-12-15:
the IGD firmware changes DEVICE_TYPE to 0 (bridge) if you have no wan connections defined in it, otherwise DEVICE_TYPE is 1 (router)... you cant be sure it works on the device if it only works with no WAN connections defined

S8T8, 2022-12-15:
couple of tests with SFU firmware;
- DEVICE_TYPE was set to 0
- SFU default values are:
OMCI_CUSTOM_BDP="258"
OMCI_CUSTOM_RDP="4"
OMCI_CUSTOM_MCAS="0"
OMCI_CUSTOM_ME="65536"
OMCI_VEIP_SLOT_ID="255
- changed default SFU values to IGD ones: 136, 0, 0, 32832, 255, nothing changed
- changed OMCI_CUSTOM_ME to 65792 and 
` omcicli mib get 329
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VEIP
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0xff01
AdminState: 1
OperState: 1
InterDomainName: 0x00000000000000000000000000000000000000000000000000
TcpUdpPtr: 0x0000
IansAssignPort: 0
CustomizedSlotId: 0xff `
- OMCI_VEIP_SLOT_ID changed (what is the suggested value to check?)

rajkosto, 2022-12-15:
try the one i suggested and seen in both Nokia and Huawei ONU, OMCI_VEIP_SLOT_ID = 14

TheKhack, 2022-12-16:
OMCI_WAN_QOS_QUEUE_NUM what it means?

S8T8, 2022-12-16:
> OMCI_VEIP_SLOT_ID = 14

No changes here