Anime4000/RTL960x #113: Setting PVID(Tagging VLAN) does not work on ODI stick

ToTheCLI, 2022-12-30:
I get a WAN connection by PPPoE on VLAN 10 when I set the VLAN on the Mikrotik it works correctly, but when I set it the ODI Stick on PVID 10 no go.
Tried on the following firmwares:
M114_sfp_ODI_Vlan_220414.tar   
M110_sfp_ODI_220817.tar    
M110_sfp_ODI_220923.tar   
all the same PVID (Tagging Mode) NOT working.

Anime4000, 2022-12-31:
this mean OLT send Inner and Outer VLAN is same value, which is does not work to PVID it
my ISP map Inner and Outer same VLAN id, trust, it wont work

ToTheCLI, 2022-12-31:
> this mean OLT send Inner and Outer VLAN is same value, which is does not work to PVID it my ISP map Inner and Outer same VLAN id, trust, it wont work

There is only one VLAN no Outer VLAN on my connection, and on another ISP there is an Outer(VLAN 1003) and Inner(VLAN 9) and it only works when both VLANs are set in Mikrotik when using VLAN 1003 as PVID on ODI and VLAN 9 on Mikrotik it does NOT work.

ToTheCLI, 2023-01-01:
Has anyone got it working on any other stick?

Anime4000, 2023-01-01:
you can use Lantiq based Stick that use OpenWRT

nguyenthanhhong-tg, 2023-01-02:
> I get a WAN connection by PPPoE on VLAN 10 when I set the VLAN on the Mikrotik it works correctly, but when I set the VLAN in the ODI Stick on PVID 10 no go. Tried on the following firmwares: M114_sfp_ODI_Vlan_220414.tar M110_sfp_ODI_220817.tar M110_sfp_ODI_220923.tar all the same PVID (Tagging Mode) NOT working.

I also got the same issue, no PPPoE when setting up VLAN on SFP stick with my V2801F (BaudCom BD-1234 SPM), but on another SFP stick (Hisense LTE3415-SCA+), PPPoE works perfectly.

ToTheCLI, 2023-01-02:
> 

Does the V2801F work when the VLAN is set from Router or not at all?

ToTheCLI, 2023-01-02:
> I also got the same issue, no PPPoE when setting up VLAN on SFP stick with my V2801F (BaudCom BD-1234 SPM), but on another SFP stick (Hisense LTE3415-SCA+), PPPoE works perfectly.

Does the V2801F work when the VLAN is set from Router or not at all?

nguyenthanhhong-tg, 2023-01-02:
> Does the V2801F work when the VLAN is set from Router or not at all?

Yes, it works when i created VLAN on MikroTik. But not work when I moved VLAN to V2801F.

longthanhtran, 2023-11-10:
Same for me, when I choose VLAN tag (35) on ODI stick then Torch from mikrotik shows no VLAN traffic at all. However, the pppoe-client also fails to connect. The only option is to use manual - transparent VLAN on ODI stick and create a VLAN interface for ppppoe-client connection.

On the otherhand, Hisense LTE3415-SCA+ stick works perfectly with VLAN tag so the inner router can dial pppoe without involving VLAN.