Anime4000/RTL960x #114: No OMCI command on V2801F (model BaudCom BD-1234 SPM)

nguyenthanhhong-tg, 2023-01-02:
Hi All,
I got problem on my BD-1234 SPM (V2801F) in OMCI command as below:

Connecting to 192.168.1.1
Connected to 192.168.1.1
login: admin
Password: 

BusyBox v1.12.4 (2019-10-14 09:10:16 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# omcicli mib get 84
# omcicli mib get 131
# omcicli mib get 171
#

Is this my SFP have no option for OMCICLI? My software version is V1.9.0-191015.

Anime4000, 2023-01-02:
Yes, V2801F based firmware wont output omcicli, it seem only print at UART

nguyenthanhhong-tg, 2023-01-03:
> Yes, V2801F based firmware wont output omcicli, it seem only print at UART

Thanks, so sad to hear this.

Anime4000, 2023-01-03:
I suggest you to buy ODI Stick instead

closed 2023-01-03 by Anime4000