Anime4000/RTL960x #115: Null OltVendorId / No VLAN Info

attie, 2023-01-06:
Hi @Anime4000 and all... thank you for collecting so much useful information together.

I originally had an FS GPON-ONU-34-20BI, and the module itself connected well and got an IP on my ISP's management VLAN, but I couldn't get any traffic to the LAN side. The ISP's router also works fine.

I'm now working with a Realtek DFP-34X-2C2, but getting some strange / unexpected results... The Web UI reports ONU State: `O5` / ONU ID: `0` / LOID Status: `Initial Status`, and as below, the OLT Vendor ID is `0x00`, with no VLAN info available. 

I have a friendly / small ISP, so I know a fair amount about their setup and can ask more... The OLT is Dasan V5808, and I know the VLANs they're using for management (101) / internet access (201). It doesn't sound like they're doing much matching / security, but I've cloned the original router's information as the best I can... The `OMCI_SW_VER1` / `OMCI_SW_VER2` values are always reset (I've tried the segfault techique with no luck), but I've just noticed that it was `V1.7.1`, and is now showing as `V1R007C00S001` which was quite unexpected!

Any ideas for what to try or what's going on here?

[image]

<details>
<summary>CLI Output</summary>

```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 329
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VEIP
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0601
AdminState: 1
OperState: 1
InterDomainName: 0x00000000000000000000000000000000000000000000000000
TcpUdpPtr: 0x0000
IansAssignPort: 0
CustomizedSlotId: 0xff
=================================
# omcicli mib get 280
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
TrafficDescriptor
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 131
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
OltG
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x00
OltVendorId: 0x00
EquipId:
Version: 0
ToDInfo:
        Sequence number of GEM superframe: 0x0
        Timestamp: secs 0, nanosecs 0
=================================
# diag gpon show us-flow
gpon show us-flow
============================================================
    GPON ONU MAC U/S Flow Status
Flow ID | GEM Port | Type | TCont
     64 |        9 | OMCI |    16
============================================================
RTK.0> command:#
# diag gpon show ds-flow
gpon show ds-flow
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
     64 |        9 | OMCI |           |
============================================================
```
</details>

attie, 2023-01-06:
As a follow up, it seems that both `V1.7.1` and `V1R007C00S001` are in `/lib/libmib.so` (same as identified by another user)... I wonder what caused the switch...

Anime4000, 2023-01-07:
if you telnet and put this `omcicli mib get 84`
it show anything?

about `V1R007C00S001` is default `OMCI_OLT_MODE 1`
another way is to modify the firmware in Linux [`qemu_test.sh`](https://github.com/Anime4000/RTL960x/tree/main/Tools/emulator):
```
./qemu_test.sh firmware.sh version
```

attie, 2023-01-07:
Apologies - I put the output from a number of commands in the collapsed section!

There's still nothing listed for `omcicli mib get 84` though 😢.
```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
```

> about V1R007C00S001 is default OMCI_OLT_MODE 1

Ahh, I hadn't realised that! Thanks.

Anime4000, 2023-01-07:
try set different OLT version,
since you have relationship with OLT guy, try ask him what alarm OLT print, what case OLT not accepting your ONU Stick

mixanikin, 2023-01-08:
Hi @Anime4000 and @attie , I have a similar problem, I know that my provider has OLT ELTEX, and similarly the status is O5, there is no VLAN and I do not receive a response from OLT, in the web interface I see that the Bytes Sent counter is working, but Bytes Recieved is not, there is a suspicion that OLT does not respond after establishing a connection.

Anime4000, 2023-01-08:
@mixanikin what `omcicli mib get 131` print?
If you have access to OLT, try check OLT alarm,
If no access to OLT, only you can do is fake call ISP saying ONU not working, just to fake it, hoping ISP unlock other ONU to access

mixanikin, 2023-01-09:
@Anime4000 omcicli mib get 131 does not work, since I have a VSOL module, so I think I'll call the provider and try to persuade him. Are there any workarounds? I still have access to the ZTE F670, I can pull the information from it, but I completely changed everything in the module and it still didn't help.

dexxter00, 2023-01-15:
> @Anime4000 omcicli mib get 131 does not work, since I have a VSOL module, so I think I'll call the provider and try to persuade him. Are there any workarounds? I still have access to the ZTE F670, I [...]
So you're trying to emulate Huawei equipment instead of ZTE? Things don't work like that, OLT may or may not have config for your ONT. Go and read the manual in this repo and then copy all the information from your original ONT!

attie, 2023-01-17:
I'm hoping to visit my ISP's test lab in the next week or two, so hopefully I'll be able to shed some light on my specific situation... and I hope it's useful for others too!