Anime4000/RTL960x #117: XPON Adaptor on GPON network

Pukkahq, 2023-01-12:
Can a XPON adaptor be used on a GPON network?

I have a Ufiber UF-Instant which works ok. 

I have also purchased a ODI DFP-34X-2C2 XPON 

When I connect the adaptor and make the same changes I made for UF-Instant I am not getting a connection.
I ran _flash all_ on both devices and changed the OMCC and OMCI setting to be the same, am I missing something else or can the technology's not be mixed?

Pukkahq, 2023-01-12:
Our suppler is BT OpenReach (UK) the ONT is a Nokia but I am not sure about the OLT

Hardware: STICK3V06 
Software: V1.0-220923
 
[image]

Pukkahq, 2023-01-12:
I have had the ONU state O5 before

ToTheCLI, 2023-01-12:
> I have had the ONU state O5 before

Put the flash all of working config and ODI config.
omitting the SN

Pukkahq, 2023-01-12:
Ufiber UF-Instant 

```

<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="admin"/>
  <Value Name="SUPER_PASSWORD" Value="stdONU101"/>
  <Value Name="ELAN_MAC_ADDR" Value="00e04c867001"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="000000000002"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="HW_WLAN0_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN1_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN0_COUNTRYSTR" Value="US"/>
  <Value Name="HW_WLAN1_COUNTRYSTR" Value="US"/>
  <Value Name="VS_AUTH_KEY" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_CWMP_MANUFACTURER" Value="VSOL"/>
  <Value Name="HW_CWMP_PRODUCTCLASS" Value="IGD"/>
  <Value Name="HW_HWVER" Value="RTL960x"/>
  <Value Name="GPON_ONU_MODEL" Value="IGD"/>
  <Value Name="LAN_SDS_MODE" Value="1"/>
  <Value Name="PON_MODE" Value="1"/>
  <Value Name="PON_LED_SPEC" Value="0"/>
  <Value Name="OUI" Value="11,11,11"/>
  <Value Name="PORT_REMAPPING" Value="00,01,02,03"/>
  <Value Name="PON_VENDOR_ID" Value="ALCL"/>
  <Value Name="GPON_SN" Value="!!!!!!!!!!!!!"/>
  <Value Name="EPON_ONU_MODEL" Value="0000"/>
  <Value Name="EPON_EXTONU_MODEL" Value="RTKG"/>
  <Value Name="EPON_SILENT_MODE" Value="1"/>
 </Dir>
</Config>
<Config Name="ROOT">
 <Dir Name="MIB_TABLE">
  <Value Name="WAN_MODE" Value="1"/>
  <Value Name="LAN_IP_ADDR" Value="192.168.254.1"/>
[... 180 lines cut ...]
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=5-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="43"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=6-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="51"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=7-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="PPPOE_SESSION_TBL">
 </Dir>
 <Dir Name="RIP_TBL">
 </Dir>
 <Dir Name="DHCP_RESERVED_IPADDR_TBL">
 </Dir>
 <Dir Name="EPON_LLID_TBL">
 </Dir>
</Config>
#

```

Pukkahq, 2023-01-12:
OID
```

<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="adsl"/>
  <Value Name="SUPER_PASSWORD" Value="realtek"/>
  <Value Name="TEST_MAC_ADDR" Value="realtek"/>
  <Value Name="ELAN_MAC_ADDR" Value="383a2127f932"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="MAC_KEY" Value="7835e20edf25945edb67f18311229714"/>
  <Value Name="RESTORE_DEFAULT_EN" Value="1"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="LASER_POLARITY_TYPE" Value="0"/>
  <Value Name="PON_DETECT_ENABLE" Value="1"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="ALCLF982721E11"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="HW_WLAN0_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN1_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN0_COUNTRYSTR" Value="US"/>
  <Value Name="HW_WLAN1_COUNTRYSTR" Value="US"/>
  <Value Name="HW_CWMP_MANUFACTURER" Value="ALCL"/>
  <Value Name="HW_CWMP_PRODUCTCLASS" Value="DFP-34X-2C2"/>
  <Value Name="HW_HWVER" Value="V2.0"/>
  <Value Name="GPON_ONU_MODEL" Value="DFP-34X-2C2"/>
  <Value Name="LAN_SDS_MODE" Value="0"/>
  <Value Name="PON_MODE" Value="1"/>
  <Value Name="PON_LED_SPEC" Value="0"/>
  <Value Name="OUI" Value="11,11,11"/>
  <Value Name="PORT_REMAPPING" Value="00,01,00,00"/>
  <Value Name="PON_VENDOR_ID" Value="ALCL"/>
  <Value Name="GPON_SN" Value="!!!!!!!!!!!!!!!"/>
  <Value Name="EPON_ONU_MODEL" Value="ALCL"/>
  <Value Name="EPON_EXTONU_MODEL" Value="DFP-34X-2C2"/>
  <Value Name="EPON_SILENT_MODE" Value="1"/>
 </Dir>
</Config>
<Config Name="ROOT">
[... 134 lines cut ...]
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=5-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="43"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=6-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="51"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=7-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="PPPOE_SESSION_TBL">
 </Dir>
 <Dir Name="RIP_TBL">
 </Dir>
 <Dir Name="DHCP_RESERVED_IPADDR_TBL">
 </Dir>
 <Dir Name="EPON_LLID_TBL">
 </Dir>
</Config>

```

ToTheCLI, 2023-01-12:
Try 
flash set OMCI_OLT_MODE 1
Flash set OMCI_FAKE_OK 1

Pukkahq, 2023-01-13:
I tried those changes, but still no go

[image]

ToTheCLI, 2023-01-13:
Put the output of telnet command
omcicli mib get 171

omcicli mib get 84

ToTheCLI, 2023-01-13:
If no output it might be fake O5

Pukkahq, 2023-01-14:
```
omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x01
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 4096, TPID 0, RemoveTags 0
Treatment Inner   : PRI 2,VID 101, TPID 4
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 0
Treatment Inner   : PRI 8,VID 101, TPID 0
INDEX 2
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 101, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 101, TPID 0
INDEX 3
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 4
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x101
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
=================================
=================================
EntityId: 0x02
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x601
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
=================================

```
```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 101
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x04
FilterTbl[0]: PRI 0,CFI 0, VID 3001
FwdOp:  0x10
NumOfEntries: 1
=================================
```

Pukkahq, 2023-01-14:
Looking at this I have manually set the Vlan ID to 101 and it now connected ok.

akeyx, 2023-10-23:
@Pukkahq I'm trying to setup a ODI DFP-34X-2C2 XPON with openreach/BT here in UK as well.
Would you mind sharing your config (priv would be OK as well) or is it literally the one you posted above for the stick + the vlan config?

akeyx, 2023-10-23:
> @akeyx what make and model is your openreach ONT?
It's:

Model Name: NOKIA-ONT 
Model number: G-010G-Q

akeyx, 2023-10-23:
@Pukkahq ^^

Pukkahq, 2023-10-23:
PLOAM Password: 1234567890
GPON SN: ALCLXXXXXXXXX

Vendor ID: ALCL
software version 1: V1.0-220923
OMCC version: 0x80
Product Class: IGD
HW version: RTL960x
OUI: 111111
Device Serial: 000000000002
MAC: 00e04cXXXXX (mac address from ONT)
Fiber Reset: Enable

Under VLAN, Manual PVID: 101

akeyx, 2023-10-23:
@Pukkahq Thanks for this.
I'll try it and report back.

akeyx, 2023-10-24:
@Pukkahq 
Sadly this doesn't work for me.
I tried your settings and various combinations of other things... to no avail. 
In majority of the cases I simply get: `ONU State: O0` and `Tx Power:  -inf dBm` 

A few times I had O5 but there was no output from omcicli for any of the MIBs hence I'd say it was fake. 
All the settings I've put were from the back plate of the original Nokia ONU. This includes the serial number and MAC address. Could you please advise if it's required to get any info by hacking the original ONU? I was considering disassembling it and hooking up the serial console but I'm not convinced it's necessary. 
Thanks in advance for any help.

For reference here's my flash dump:
```<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="adsl"/>
  <Value Name="SUPER_PASSWORD" Value="realtek"/>
  <Value Name="TEST_MAC_ADDR" Value="realtek"/>
  <Value Name="ELAN_MAC_ADDR" Value="cccf83d7XXXX"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="MAC_KEY" Value="XXXX"/>
  <Value Name="RESTORE_DEFAULT_EN" Value="1"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="LASER_POLARITY_TYPE" Value="0"/>
  <Value Name="PON_DETECT_ENABLE" Value="1"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="ALCLXXXXXXXX"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="HW_WLAN0_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN1_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN0_COUNTRYSTR" Value="US"/>
  <Value Name="HW_WLAN1_COUNTRYSTR" Value="US"/>
  <Value Name="HW_CWMP_MANUFACTURER" Value="ALCL"/>
  <Value Name="HW_CWMP_PRODUCTCLASS" Value="IGD"/>
  <Value Name="HW_HWVER" Value="RTL960x"/>
  <Value Name="GPON_ONU_MODEL" Value="IGD"/>
  <Value Name="LAN_SDS_MODE" Value="0"/>
  <Value Name="PON_MODE" Value="1"/>
  <Value Name="PON_LED_SPEC" Value="0"/>
  <Value Name="OUI" Value="cc,cf,83"/>
  <Value Name="PORT_REMAPPING" Value="00,01,00,00"/>
  <Value Name="PON_VENDOR_ID" Value="ALCL"/>
[... 139 lines cut ...]
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=4-->
  <Value Name="VPI" Value="0"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=5-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="43"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=6-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="51"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=7-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="PPPOE_SESSION_TBL">
 </Dir>
 <Dir Name="RIP_TBL">
 </Dir>
 <Dir Name="DHCP_RESERVED_IPADDR_TBL">
 </Dir>
 <Dir Name="EPON_LLID_TBL">
 </Dir>
</Config>```

Anime4000, 2023-10-25:
ONU State: O0 show a invalid mac key

Pukkahq, 2023-10-25:
here is my config, dont know if its going to be any help. I do have some brand new modules here so I could try one of them
```
<Config Name="ROOT">
 <Dir Name="MIB_TABLE">
  <Value Name="WAN_MODE" Value="7"/>
  <Value Name="LAN_IP_ADDR" Value="192.168.254.1"/>
  <Value Name="LAN_SUBNET" Value="255.255.255.0"/>
  <Value Name="LAN_ENABLE_IP2" Value="0"/>
  <Value Name="LAN_IP_ADDR2" Value="192.168.100.1"/>
  <Value Name="LAN_SUBNET2" Value="255.255.255.0"/>
  <Value Name="LAN_RIP" Value="0"/>
  <Value Name="LAN_AUTOSEARCH" Value="0"/>
  <Value Name="DNS_MODE" Value="0"/>
  <Value Name="DNS1" Value="0.0.0.0"/>
  <Value Name="DNS2" Value="0.0.0.0"/>
  <Value Name="DNS3" Value="0.0.0.0"/>
  <Value Name="ADSL_MODE" Value="83"/>
  <Value Name="ADSL_OLR" Value="3"/>
  <Value Name="USER_NAME" Value="user"/>
  <Value Name="USER_PASSWORD" Value="user"/>
  <Value Name="DEVICE_TYPE" Value="1"/>
  <Value Name="INIT_LINE" Value="1"/>
  <Value Name="INIT_SCRIPT" Value="1"/>
  <Value Name="SNMP_SYS_NAME" Value="Modem/Router"/>
  <Value Name="VC_AUTOSEARCH" Value="0"/>
  <Value Name="BR_AGEING_TIME" Value="7200"/>
  <Value Name="BR_STP_ENABLED" Value="0"/>
  <Value Name="MP_MODE" Value="4"/>
  <Value Name="SPC_ENABLED" Value="0"/>
  <Value Name="SPC_IPTYPE" Value="0"/>
  <Value Name="MP_FIN" Value="1"/>
  <Value Name="MULTI_LINGUAL_STATE" Value="en"/>
  <Value Name="TFTP_SERVER_ADDR" Value="tftp://0.0.0.0"/>
  <Value Name="POSIX_TZ_STRING" Value=""/>
  <Value Name="BOOT_FILENAME" Value=""/>
  <Value Name="UPNP" Value="0"/>
  <Value Name="UPNP_EXT_ITF" Value="255"/>
  <Value Name="SYSLOG_LOG_LEVEL" Value="0"/>
  <Value Name="SYSLOG_DISPLAY_LEVEL" Value="0"/>
  <Value Name="SUSER_NAME" Value="admin"/>
  <Value Name="SUSER_PASSWORD" Value="admin"/>
  <Value Name="E8BDUSER_NAME" Value="administrator"/>
  <Value Name="E8BDUSER_PASSWORD" Value="SXXXXXXXX"/>
  <Value Name="ADSL_TONE" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="DIRECT_BRIDGE_MODE" Value="1"/>
  <Value Name="ADSL_HIGH_INP" Value="0"/>
  <Value Name="SYSLOG" Value="0"/>
  <Value Name="MAXLOGLEN" Value="16384"/>
  <Value Name="ADSLDBG" Value="0"/>
  <Value Name="MIB_ETH_MAC_CTRL" Value="0"/>
  <Value Name="MIB_WLAN_MAC_CTRL" Value="0"/>
  <Value Name="SNOOPING_ENABLED" Value="0"/>
  <Value Name="LOID" Value=""/>
  <Value Name="LOID_PASSWD" Value=""/>
  <Value Name="LOID_OLD" Value=""/>
  <Value Name="LOID_PASSWD_OLD" Value=""/>
  <Value Name="PROVINCE_TRAP_PPPOE_TRAFFIC" Value="0"/>
  <Value Name="GPON_PLOAM_FORMAT" Value="1"/>
  <Value Name="GPON_PLOAM_PASSWD" Value="31323334353637383930"/>
  <Value Name="OMCI_DBGLVL" Value="0"/>
  <Value Name="OMCI_LOGFILE" Value="0"/>
  <Value Name="OMCI_PORT_TYPE" Value="1"/>
  <Value Name="DUAL_MGMT_MODE" Value="1"/>
  <Value Name="OMCI_CUSTOM_BDP" Value="258"/>
  <Value Name="OMCI_CUSTOM_RDP" Value="4"/>
  <Value Name="OMCI_CUSTOM_MCAST" Value="0"/>
  <Value Name="OMCI_CUSTOM_ME" Value="65536"/>
  <Value Name="OMCI_VENDOR_PRODUCT_CODE" Value="15"/>
  <Value Name="OMCI_FAKE_OK" Value="1"/>
  <Value Name="OMCI_VEIP_SLOT_ID" Value="255"/>
[... 111 lines cut ...]
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=5-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="43"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=6-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="51"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=7-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="PPPOE_SESSION_TBL">
 </Dir>
 <Dir Name="RIP_TBL">
 </Dir>
 <Dir Name="DHCP_RESERVED_IPADDR_TBL">
 </Dir>
 <Dir Name="EPON_LLID_TBL">
 </Dir>
</Config>

```

akeyx, 2023-10-25:
@Pukkahq your config looks pretty standard at first glance. I'll check it in details later.
Could you please advise how you populated your MAC address? Did you take the MAC address directly from the back plate of your ISP's ONT or somehow different?
My problem maybe the MAC as suggested by @Anime4000. I simply copy pasted the MAC from the back plate of my ISP's ONT. I can see that on the LAN side the MAC is the correct (cloned) but I guess the WAN side is different. I saw a video where the difference between MAC on the LAN and WAN sides on ONT was just one one value up so eg:
if LAN (the back plate) MAC is: XX:XX:XX:XX:XX:4C the WAS was simply: XX:XX:XX:XX:XX:4D. Could you please advise if that's the case on your ONT as well?

ToTheCLI, 2023-10-25:
> @Pukkahq your config looks pretty standard at first glance. I'll check it in details later. Could you please advise how you populated your MAC address? Did you take the MAC address directly from the back plate of your ISP's ONT or somehow different? My problem maybe the MAC as suggested by @Anime4000. I simply copy pasted the MAC from the back plate of my ISP's ONT. I can see that on the LAN side the MAC is the correct (cloned) but I guess the WAN side is different. I saw a video where the difference between MAC on the LAN and WAN sides on ONT was just one value up so eg: if LAN (the back plate) MAC is: XX:XX:XX:XX:XX:4C the WAN was simply: XX:XX:XX:XX:XX:4D. Could you please advise if that's the case on your ONT as well?

If you change the MAC Address you need to change MAC Key also or it will not work...

akeyx, 2023-10-25:
> If you change the MAC Address you need to change MAC Key also or it will not work...

That's probably the problem then... I'm sure I've seen the algo to generate the key somewhere... I'll dig. Thanks for the tip.

akeyx, 2023-10-25:
Yeah, that's worked. The MAC_KEY set via this: https://gist.github.com/rajkosto/29c513b96ea6262d2fb1f965a52ce16f
I actually used this simple bash:
```mac="xx:xx:xx:xx:xx:xx";MAC=${mac^^};echo -n "hsgq1.9a${MAC//:/}"|md5sum``` 
...which does exactly the same thing.

And now I have O5 but no internet yet and omcicli mib get 171 and 84 only give me:
```
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
```

which seems a bit short in compare to what you had posted earlier... But I have to fix the config as I may have made too many changes to try things out. Will let ya know.
Thanks.

Pukkahq, 2023-10-25:
That's interesting as I did not generate a new MACKEY. I just left the field blank.

akeyx, 2023-10-25:
I didn't try with it empty. Just had it pre-populated with what ever was there out of the box. Can try with the field empty later just for the fun of it.

Anime4000, 2023-10-25:
Currently I working on my custom WebGUI, so far I added to MAC Key Generator build directly into WebGUI, next adding simple command-line with automatically update correct MAC Key

https://github.com/Anime4000/RTL960x/assets/1908715/540a9a4c-72e7-4ccc-8885-3ab20905e29c

akeyx, 2023-10-26:
> Currently I working on my custom WebGUI, so far I added to MAC Key Generator build directly into WebGUI, next adding simple command-line with automatically update correct MAC Key
> 
>  2023-10-07.21-33-24.mp4

Nice!
Since setting up OUI is also pretty much always the case you might want to consider auto generating it from the MAC address with the option of overriding it if required.
I guess the same could apply to Vendor ID based on serial number. 
...just a suggestion.

Anime4000, 2023-10-26:
> > Currently I working on my custom WebGUI, so far I added to MAC Key Generator build directly into WebGUI, next adding simple command-line with automatically update correct MAC Key
> > 
> >  2023-10-07.21-33-24.mp4
> 
> Nice!
> Since setting up OUI is also pretty much always the case you might want to consider auto generating it from the MAC address with the option of overriding it if required.
> I guess the same could apply to Vendor ID based on the serial number. 
> ...just a suggestion.

that in my dev list, simplified the GUI,
of course can override via telnet

Pukkahq, 2023-11-10:
@akeyx how did you get on?

akeyx, 2023-11-10:
Sadly it doesn't work for me.
I do get some mibs back but the vlan ones as shown earlier are empty.
I have two sticks with the same firmware just different brand logo and both behave the same. 
I'm wondering if the issue is that I'm simply basing my settings on the info I got from the back of the original ONT rather than reading from the running ont os ( serial or telnet/ssh ) but I'm a bit hesitant to crack it open for soldering the serial console there. I'm Ok with electronics so it's not that. I'm not sure if there's much point of doing so since the console is most likely password protected and I don't have the credentials.
Any advise welcome.

zentavr, 2023-11-29:
@akeyx had the same issue like yours. My sets look like this:
```
flash set GPON_PLOAM_FORMAT 0
flash set GPON_PLOAM_PASSWD 61343330393488787878
flash set ELAN_MAC_ADDR 383A2129001C
flash set WLAN_MAC_ADDR 249EBAC7B7CA
flash set MAC_KEY eab829c77a33bb2c2eb98928d1f59bbd
flash set PON_MODE 1
# With OLT Mode 1 you cannot set custom SW version :(
#flash set OMCI_OLT_MODE 1
flash set OMCI_OLT_MODE 21
flash set OMCI_SW_VER1 V3R017C00S100
flash set OMCI_SW_VER2 V3R017C00S100
flash set GPON_ONU_MODEL 1GE-XPON-ONU
flash set GPON_SN HWTCBAC7B7CA
flash set PON_VENDOR_ID HWTC
flash set HW_HWVER 6A5.A
flash set HW_CWMP_MANUFACTURER 'SK Technologies Co., Ltd'
flash set HW_CWMP_PRODUCTCLASS HGU

flash set HW_SERIAL_NO 48575443FFEEDD00
flash set OUI 00259e

flash set E8BDUSER_PASSWORD newpassword
flash set SUPER_PASSWORD newpassword
flash set USER_PASSWORD newpassword
flash set OMCI_FAKE_OK 1

# Extra
flash set OMCI_VENDOR_PRODUCT_CODE 106
flash set OMCC_VER 128
```

In my case I had custom  `OMCC_VER` which my provider didn't like (It was `0xB0`). I'd put 128 (the default value) back and everything appeared (I was the VLANs with `omcicli mib get 84`).