Anime4000/RTL960x #120: VLAN_MANU_MODEs  2 and 3 Function

ToTheCLI, 2023-01-13:
How ONU Stick handle VLAN from PON to ETH interface
| Value | Mode | Note | Explaination |
|-------|------|------|--------------|
| `0` | Transparent Mode | VLAN will mapped as is | Map VLAN without changing original id |
| `1` | Tagging Mode | VLAN Stacking / Q-in-Q | Untag first VLAN layer and bridge as is |
| `2` | Remote Access Mode |  |  |
| `3` | Special Case Mode |  |  |

what do VLAN_MANU_MODEs  2 and 3 do?
and when set to 2 or 3 is there a possibility to not be able to access SFP by telnet or WebUI?

Anime4000, 2023-01-14:
these 2 and 3 I no info about it, no one does, it seem might have custom driver? I not really sure

ToTheCLI, 2023-01-14:
> these 2 and 3 I no info about it, no one does, it seem might have custom driver? I not really sure

Does setting to 2 or 3 affect the ability to access SFP by Telnet?

Anime4000, 2023-01-14:
nope, can set as it, it wont bother