Anime4000/RTL960x #121: Problem with VLAN in DFP-34X-2C2

diogosilvan, 2023-01-14:
I'm from Brazil, where I'm trying to use a DFP-34X-2C2 with the company "Oi Fibra" which in my region uses Nokia in the OLT and a Nokia G-140W-H in my residence.
I did all the clone procedure, GPON serial and etc and had O5 successfully.
My provider has two vLANs 210 for internet and 220 for phone/SIP.
However, according to the print below and my tests, only vLAN 220 works, where it is effectively accessed by vLAN 1899, however, vLAN 210 does not work.

Interestingly in "omcicli mib get 171" both vLANs 210 and 220 are displayed but in "omcicli mib get 84" only 1899 which is 220 works.

I've already changed the IGD and SFU firmwire, the type of router or bridge firmwire and I've tried all forms of configuration, however, the only vLAN that works is 1899/220, which is the voice/sip vLAN.

Could anyone help me with something to do?

PS. In other regions Oi Fibra uses Huawei and where it uses the DFP-34X-2C2 it works perfectly.

# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0xe102
FilterTbl[0]: PRI 0,CFI 0, VID 1899
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xe103
FilterTbl[0]: PRI 0,CFI 0, VID 1899
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xe10b
FilterTbl[0]: PRI 0,CFI 0, VID 1899
FwdOp:  0x10
NumOfEntries: 1
=================================

# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0xe01
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 8,VID 220, TPID 5
Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 9,VID 1899, TPID 3
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 220, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 1899, TPID 2
INDEX 2
Filter Outer   : PRI 8,VID 210, TPID 5
Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 9,VID 1899, TPID 3
INDEX 3
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 210, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 1899, TPID 2
INDEX 4
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 0, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 2
INDEX 5
Filter Outer   : PRI 14,VID 4096, TPID 5
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 4096, TPID 3
INDEX 6
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 4096, TPID 2
AssociatedMePoint: 0x601
DscpToPbitMapping:
        0x000000
        0x249249
        0x492492
        0x6db6db
        0x924924
        0xb6db6d
        0xdb6db6
        0xffffff

ToTheCLI, 2023-01-14:
This seems related to [issue #26](https://github.com/Anime4000/RTL960x/issues/26)
It seems that both VLANs 210 and 220 end up as VLAN 1899 with different EntityIDs

Anime4000, 2023-01-14:
this might help if using `fix_vlan_tag` on older firmware
https://github.com/Anime4000/RTL960x/tree/main/Firmware/DFP-34X-2C2#m110_sfp_odi_220817tar

diogosilvan, 2023-01-14:
> this might help if using `fix_vlan_tag` on older firmware https://github.com/Anime4000/RTL960x/tree/main/Firmware/DFP-34X-2C2#m110_sfp_odi_220817tar

I already tried with this firmware too, don't works.
But I will use it again and post here to see if anything has changed on "omcicli" output.

ToTheCLI, 2023-01-14:
Try setting FwdOp to 0x02 for each entityID Alone with the following commands then check for Internet then reboot after no connection
omcicli mib set 84 0xe102 FwdOp 0x02
Then check for connection on VLAN 1899
No connection reboot
then
omcicli mib set 84 0xe103 FwdOp 0x02
Then check for connection on VLAN 1899
No connection reboot
then
omcicli mib set 84 0xe10b FwdOp 0x02

diogosilvan, 2023-01-14:
> Try setting FwdOp to 0x02 for each entityID Alone with the following commands then check for Internet then reboot after no connection omcicli mib set 84 0xe102 FwdOp 0x02 Then check for connection on VLAN 1899 and 210 No connection reboot then omcicli mib set 84 0xe103 FwdOp 0x02 Then check for connection on VLAN 1899 and 210 No connection reboot then omcicli mib set 84 0xe10b FwdOp 0x02

If "omcicli mib set 84 0xe102 FwdOp 0x02" only vLAN 1899 works, but only on 220 voice/SIP.
If "omcicli mib set 84 0xe103 FwdOp 0x02" WORKS! On vlan 1899.
If "omcicli mib set 84 0xe10b FwdOp 0x02" same result of first command, only 1899 on voice/SIP.

Is there a way to make this change permanent, resistant to reboot?

ToTheCLI, 2023-01-14:
#!/bin/sh
# VLAN Tag Fix: Fix VLAN wrong mapping
# By inyourgroove
# Additional VID check if config file is not empty by rajkosto

set -e

while true; do

  for _EntityID in $(omcicli mib get 84 | grep '^EntityID:' | awk '{print $2}'); do

    _FwdOp=$(omcicli mib get 84 ${_EntityID} | grep '^FwdOp:' | awk '{print $2}')

    if [[ ! -z ${_FwdOp} && ${_FwdOp} != '0x02' ]]; then
      if [ -s " 0xe103" ]; then #only change FwdOp VIDs specified in the file
        _VlanId=$(omcicli mib get 84 ${_EntityID} | grep -o " VID [0-9]*" | awk '{print $2}')
	if ! grep -o "\b${_VlanId}\b" "$configFname" > /dev/null; then
          continue
        fi
      fi
      
      omcicli mib set 84 ${_EntityID} FwdOp 0x02
    fi

  done

  sleep 30
done

ToTheCLI, 2023-01-14:
Maybe @Anime4000 or @rajkosto can chime in and make a custom script firmware for you.

ToTheCLI, 2023-01-14:
Or you can DIY see this
https://github.com/Anime4000/RTL960x/tree/main/Tools/emulator
https://github.com/Anime4000/RTL960x/blob/main/Tools/etc/scripts/fix_vlan_tag.sh

S8T8, 2023-01-14:
Hi, may I suggest you to try @rajkosto's mod mod of the firmware M114_sfp_ODI_hybrid_220916 https://github.com/rajkosto/RTL960x/blob/main/Firmware/DFP-34X-2C2/M114_sfp_ODI_hybrid_220916F.tar ?

It's working for me and I have the same issue with other versions.
Also tried M110_sfp_ODI_220923FS with:
```
OMCI_CUSTOM_ME=32832
OMCI_CUSTOM_ME=65792
OMCI_VEIP_SLOT_ID=14
```
without success.

Only M114_sfp_ODI_hybrid_220916 is working with
```
flash set VLAN_CFG_TYPE 1
flash set VLAN_MANU_MODE 0
```
#107

diogosilvan, 2023-01-15:
> omcicli mib set 84 0xe103 FwdOp 0x02

It didn't work for me, right now I'm studying how to make a custom that executes the command that works for my case automatically.
At this moment I have a script that runs on the router that automatically executes the command, but it is not ideal.

diogosilvan, 2023-01-23:
> Or you can DIY see this https://github.com/Anime4000/RTL960x/tree/main/Tools/emulator https://github.com/Anime4000/RTL960x/blob/main/Tools/etc/scripts/fix_vlan_tag.sh

I customized the firmware "M110_sfp_ODI_220817.tar" by editing the script "fix_vlan_tag.sh" to run my customized command instead of the original script.
But unfortunately I don't have access to the voice/telephone vlan and still the DHCP gets the IP of the wrong vlan before the script runs and applies the fix. 
I had to make a script that checks the IP obtained via DHCP, if it starts with 10. it renews to then obtain the correct internet IP.

bassopt, 2023-04-03:
I have exactly the same problem
I can’t for the life of me use the vlans provided by omcicli mib get 171. I can although use the one given by  omcicli mib get 84 by going to the web interface choose vlan - manual, save, then use transportation and save. The issue is that if the ONU reboots I have to go through these steps all over again which is not a practical solution. 

Hope someone cal solve this issue soon