Anime4000/RTL960x #122: Difference between DFP-34X-2C2 and DFP-34G-2C2

paulierco, 2023-01-14:
First of all, I would like to thank you all for the time you invest in this project.

After I found this, I decided to order DFP-34G-2C2 and everything works as expected but even if I have 1000/500 pppoe connection, I cannot get more than ~950Mbit.  Do you think ordering DFP-34X-2C2 might have a change to get a higher speed ?
Currently the Stick is plugged in into a CRS305-1G-4S+ and I had a second port connected to a 2.5/5G USB3.1 Network card into my PC.

I can provide details from the SFP ODI if is necessary.

Best Regards,
Paul

ToTheCLI, 2023-01-14:
Some newer orders of the g version use the exact chipset of the x version.
Check cpu by telnet.

paulierco, 2023-01-14:
```
# cat /proc/cpuinfo
system type             : RTL8672
processor               : 0
cpu model               : 56322
BogoMIPS                : 299.00
tlb_entries             : 64
mips16 implemented      : yes
#
```

ToTheCLI, 2023-01-14:
> ```
> # cat /proc/cpuinfo
> system type             : RTL8672
> processor               : 0
> cpu model               : 56322
> BogoMIPS                : 299.00
> tlb_entries             : 64
> mips16 implemented      : yes
> #
> ```

Same as DFP-34X-2C2

paulierco, 2023-01-14:
So what's the difference?

ToTheCLI, 2023-01-14:
In your case no difference what so ever, older models g had ZTE chipset

Gouster4, 2023-01-19:
DFP-34E-2C2 - EPON
DFP-34G-2C2 - GPON
DFP-34X-2C2 - EPON/GPON Combo.

I guess that epon version has weaker/different chipset.
Not sure about hardware difference between G and X version... But if older ones used different chipset, and new g version use same chipset, then difference on new verisions is most likely only firmware.

Egyras, 2023-01-27:
> DFP-34E-2C2 - EPON DFP-34G-2C2 - GPON DFP-34X-2C2 - EPON/GPON Combo.
> 
> I guess that epon version has weaker/different chipset. Not sure about hardware difference between G and X version... But if older ones used different chipset, and new g version use same chipset, then difference on new verisions is most likely only in firmware (no epon support implemented)

so if I want to replace my providers ZTE F601 GPON converter and use directly in mikrotik RB5009 I can select 
DFP-34G-2C2 - GPON or DFP-34X-2C2 - EPON/GPON (this is XPON ?)   ?

Anime4000, 2023-01-27:
34X is XPON, always choose XPON no matter what, it support EPON and GPON

narapon, 2023-02-09:
I have the DFP-34G-2C2, PON STICK V03, plugged into a ccr1009, auto-nego off and set to 1g as per your guide, im getting no link once in a while, can't even ping 192.168.1.1 from 192.168.1.11, with masquerade for out interface SFP

SFP tab shows:
Vendor Name: OEM
Vendor Part Number: STICK
Vendor Serial: GPON21111317
Manufacturing Date: 21-11-13

without fiber connected, its always no link as expected, with fiber connected i still get link downs
are we supposed to configure the module before connecting fibre?

Anime4000, 2023-02-09:
You need connect the fiber in order to work, only some Mikrotik Router are not required, like RB5009.
This behavior seem different from Router to Router,

RB3011 need PON Stick Fiber connected
RB5009 no need fiber to connect

narapon, 2023-02-09:
Understood, the link down still happens when the fiber is connected even if I do not do anything and just observe. Is this the firmware rebooting, module overheating or just a bad module that I got? It's gotten 806 link downs in the span of only a few hours

I've unplugged it from the SFP+ port and switched to the SFP port and have 0 link downs for the past 10 minutes so I'm getting my hopes up for a bit on this module. 

192.168.1.1 shows up on the ARP table as the module seems to be sending ARP discovery packets to 192.168.1.11. Ping to 192.168.1.1 still fails with timeout though

narapon, 2023-02-09:
looks like its my CCR, will look into getting this working on a CRS

narapon, 2023-02-15:
just an update on what mikrotik hardware works and doesn't

- CCR1009 - Direct to Tilera CPU - SFP module detected on both SFP+ and SFP ports but doesn't work
- CRS309-1G-8S+IN - 98DX8208 - works, used ROS 7.7
- CCR2116-12G-4S+ - 98DX3255 - works, used ROS 7.8rc1
seems to like the Marvell Prestera switch chips?

Anime4000, 2023-02-16:
> just an update on what mikrotik hardware works and doesn't
>
>     * CCR1009 - Direct to Tilera CPU - SFP module detected on both SFP+ and SFP ports but doesn't work
>
>     * CRS309-1G-8S+IN - 98DX8208 - [...]

this default 1G mode ?

narapon, 2023-02-18:
> this default 1G mode ?

not sure what you mean by default 1G mode, but I disabled auto-nego as mentioned in your guide

S8T8, 2023-02-18:
I think @Anime4000 means that those devices works also in 2.5G mode, I can confirm that CCR2116-12G-4S+ works with auto-negotiation=no speed=2.5Gbps and LAN_SDS_MODE 4

Anime4000, 2023-02-19:
@S8T8 
list updated

bsborisov, 2023-12-10:
Somehow for me is not working. :/ When I set "auto-negotiation=no speed=2.5Gbps and LAN_SDS_MODE 4" somehow the router is dropping the link to all sfp+ ports. I can't even log in to change it back. Module is DFP-34G-2C2 and the router is CCR2116-12G-4S+ Any ideas what I am missing?

Anime4000, 2023-12-12:
Yes, use `LAN_SDS_MODE 6` (2500Base-X) for ROS 7.12+

Even Mikrotik `2.5G baseX` tell us to use `base-X`, it also support HiSGMII PHY,
As I tested Huawei MA5671a on RB5009 with ROS7.12.2