Anime4000/RTL960x #123: Unknown RTL9601B EPON converter?

dolphin22, 2023-01-21:
I got this converter secondhand and found out it uses RTL9601B. I successfully connected to on-board UART and run few commands.
Unfortunately, I cannot set PON_MODE to 1 instead of 2. 
`flash set PON_MODE 1
reboot`
I think it is because default configuration lastgood_hs.xml from /dev/mtd3 is always loaded at reboot. I tried to change lastgood_hs.xml PON_MODE Value="1" but unsuccess. Please shine your light to sort this issue out. 
`# cat /var/config/lastgood_hs.xml
<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="adsl"/>
  <Value Name="SUPER_PASSWORD" Value="realtek"/>
  <Value Name="ELAN_MAC_ADDR" Value="00e04c867001"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,                                     00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,                                     00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00                                     ,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00                                     ,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00                                     ,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00                                     ,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00                                     ,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="000000000002"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,                                     72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="PON_MODE" Value="2"/>
  <Value Name="PON_LED_SPEC" Value="0"/>
  <Value Name="GPON_SN" Value="BLKG47010000"/>
 </Dir>
</Config>
# cat /var/config/lastgood.xml
<Config Name="ROOT">
<Config Name="ROOT">
 <Dir Name="MIB_TABLE">
  <Value Name="WAN_MODE" Value="7"/>
  <Value Name="LAN_IP_ADDR" Value="192.168.1.1"/>
  <Value Name="LAN_SUBNET" Value="255.255.255.0"/>
  <Value Name="LAN_ENABLE_IP2" Value="0"/>
  <Value Name="LAN_IP_ADDR2" Value="192.168.100.1"/>
  <Value Name="LAN_SUBNET2" Value="255.255.255.0"/>
  <Value Name="LAN_DHCP" Value="0"/>
  <Value Name="LAN_DHCP_POOLUSE" Value="0"/>
  <Value Name="LAN_DHCP_START" Value="2"/>
  <Value Name="LAN_DHCP_END" Value="100"/>
  <Value Name="LAN_DHCP_POOL_START" Value="192.168.1.100"/>
  <Value Name="LAN_DHCP_POOL_END" Value="192.168.1.199"/>
  <Value Name="LAN_DHCP_DNS_OPT" Value="0"/>
  <Value Name="DHCPS_DNS1" Value=""/>
[... 403 lines cut ...]
  <Value Name="protectionType" Value="0"/>
  <Value Name="numPONIf" Value="1"/>
  <Value Name="numSlot" Value="0"/>
  <Value Name="numInterface" Value="1"/>
  <Value Name="interfaceType" Value="0"/>
  <Value Name="fe_numPorts" Value="0"/>
  <Value Name="ge_numPorts" Value="1"/>
  <Value Name="numPOTSPorts" Value="0"/>
  <Value Name="numE1Ports" Value="0"/>
  <Value Name="numUSQueues" Value="8"/>
  <Value Name="maxQueueUSPort" Value="8"/>
  <Value Name="numDSQueues" Value="8"/>
  <Value Name="maxQueueDSPort" Value="8"/>
  <Value Name="BatteryBackup" Value="0"/>
  <Value Name="fecSupport" Value="1"/>
  <Value Name="fecMode" Value="2"/>
 </Dir>
 <Dir Name="CHIPSET_ID_TBL"> 
  <Value Name="vendorid" Value="1001"/>
  <Value Name="chip_model" Value="38401"/>
  <Value Name="revision" Value="2"/>
  <Value Name="date" Value="787969"/>
 </Dir>
</Config>
`

dolphin22, 2023-01-21:
[https://i.imgur.com/gZEVTE9.jpeg](url)
[https://i.imgur.com/xIrWoz3.jpeg](url)

dolphin22, 2023-01-21:
Video of WebUI

https://user-images.githubusercontent.com/1845502/213858423-e130b64b-5978-4e66-a47b-3ce1def3b85c.mp4

Anime4000, 2023-01-21:
If keep reverting to PON_MODE 1, mean the kernel and driver having missing EPON support,
this might need to modify the firmware it self, dump the SPI Flash and extract the mtd

dolphin22, 2023-01-21:
> If keep reverting to PON_MODE 1, mean the kernel and driver having missing EPON support, this might need to modify the firmware it self, dump the SPI Flash and extract the mtd
Seems like a complicated process, any tutorial is much appreciated. Thanks!

dolphin22, 2023-01-21:
Got it thanks https://github.com/Anime4000/RTL960x/blob/main/Docs/Modify_Firmware.md

Ulkizz, 2023-02-12:
> I got this converter secondhand and found out it uses RTL9601B. I successfully connected to on-board UART and run few commands. Unfortunately, I cannot set PON_MODE to 1 instead of 2. `flash set PON_MODE 1 reboot` I think it is because default configuration lastgood_hs.xml from /dev/mtd3 is always loaded at reboot. I tried to change lastgood_hs.xml PON_MODE Value="1" but unsuccess. Please shine your light to sort this issue out.
> 
> `# cat /var/config/lastgood_hs.xml
> 
> `
> [rtl9601b_config_files.txt](https://github.com/Anime4000/RTL960x/files/10471758/rtl9601b_config_files.txt)

Keep in mind that some XPON adapters automatically detects type of PON. Thus sets PON MODE.

dolphin22, 2023-02-18:
> Keep in mind that some XPON adapters automatically detects type of PON. Thus sets PON MODE.

Thanks for your suggestion! I found out that firmware sw_commit=1 will let you set PON MODE but I failed to get PON working so I gave up.