Anime4000/RTL960x #125: Connection keep dropped for DFP-34X-2C2 and IPTV not working

zhijianglee, 2023-01-30:
Hi, 

I have followed the guide and have successfully setup the stick. 

However, I noticed that 

- IPTV is not working although VLAN is set. Able to watch only single channel and the stream will get frozen after some time
- Internet get disconnected. and the stick unable to be ping

Anime4000, 2023-01-30:
what firmware are you using?

can show us your network topology? the WAN ingress, WAN VLAN

zhijianglee, 2023-01-30:
Diagram to show my current network setup. 

[image]

This setup is working fine currently. 

Below are the configuration. 

IPTV configuration for Asus AC86U. WAN IP is obtained via DHCP. 

[image]

Configuration for Mikrotik 

[image]
[image]
[image]
[image]
[image]

[image]
[image]

For this, my plan is to disable ether1 and enable sfp plus1. I did that before. Internet connection and connection to the stick is short lived. 

I did enable IGMP Snooping in the stick, and I able to watch On demand videos using Unifi TV Box but not live videos. Channel guide and navigation works fine. 

Given that IGMP Snooping is disabled in ONU stick, Some attempts of enabling and disabing IGMP in Mikrotik router and Asus router will cause the stick to hang and not responding.  It will be OK after 10 minutes and it repeats again. 

My intended network setup is to replace the ONU with that GPON stick with IPTV working as well with stable connection.

zhijianglee, 2023-01-31:
[image]

zhijianglee, 2023-01-31:
`<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="adsl"/>
  <Value Name="SUPER_PASSWORD" Value="realtek"/>
  <Value Name="TEST_MAC_ADDR" Value="realtek"/>
  <Value Name="ELAN_MAC_ADDR" Value="383a21280508"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="MAC_KEY" Value="e66a4b462f2bddca33864781888c97df"/>
  <Value Name="RESTORE_DEFAULT_EN" Value="1"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="LASER_POLARITY_TYPE" Value="0"/>
  <Value Name="PON_DETECT_ENABLE" Value="1"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="48575443B1313591"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="HW_WLAN0_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN1_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN0_COUNTRYSTR" Value="US"/>
  <Value Name="HW_WLAN1_COUNTRYSTR" Value="US"/>
  <Value Name="HW_CWMP_MANUFACTURER" Value="Huawei Technologies Co., Ltd"/>
  <Value Name="HW_CWMP_PRODUCTCLASS" Value="HGU"/>
  <Value Name="HW_HWVER" Value="294.A"/>
  <Value Name="GPON_ONU_MODEL" Value="HG8240w"/>
  <Value Name="LAN_SDS_MODE" Value="0"/>
  <Value Name="PON_MODE" Value="1"/>
  <Value Name="PON_LED_SPEC" Value="0"/>
  <Value Name="OUI" Value="00,25,9e"/>
  <Value Name="PORT_REMAPPING" Value="00,01,00,00"/>
  <Value Name="PON_VENDOR_ID" Value="HWTC"/>
  <Value Name="GPON_SN" Value="HWTCB1313591"/>
  <Value Name="EPON_ONU_MODEL" Value="ODI"/>
  <Value Name="EPON_EXTONU_MODEL" Value="DFP-34X-2C2"/>
  <Value Name="EPON_SILENT_MODE" Value="1"/>
 </Dir>
</Config>
<Config Name="ROOT">
 <Dir Name="MIB_TABLE">
  <Value Name="WAN_MODE" Value="7"/>
  <Value Name="LAN_IP_ADDR" Value="192.168.1.1"/>
[... 129 lines cut ...]
 <Dir Name="AUTO_PVC_SEARCH_TBL"> 
  <Value Name="VPI" Value="0"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> 
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="43"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> 
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="51"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> 
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="PPPOE_SESSION_TBL">
 </Dir>
 <Dir Name="RIP_TBL">
 </Dir>
 <Dir Name="DHCP_RESERVED_IPADDR_TBL">
 </Dir>
 <Dir Name="EPON_LLID_TBL">
 </Dir>
</Config>`

zhijianglee, 2023-01-31:
This is what happened when I disabled IGMP Snooping in the stick 

[image]

[image]

Anime4000, 2023-02-01:
I see, you are on Unifi,
There are know issue with IPTV, but PON Stick should bridge it and untouched,
which firmware are you using? SFU or IGD type?

zhijianglee, 2023-02-01:
should be SFU type.

zhijianglee, 2023-02-01:
I believe it is due to IGMP settings is not properly configured in PON stick

This is what wa sobtained rom Huawei ONU

WAP>igmp get global cfg
[igmp]Display global config
```
-------------------------------------------------
 OLTVendor           : 0     MaxUserPerPgmPerPort : 32
 ProcessMode         : 1     McWorkMode           : 2
 IsIgmpSnooping      : 1     IgmpCtlMode          : 0
 MultiGemportID      : 4095  MvlanAction          : 3
 ReplaceMvlan        : 0     ProVlanAction        : 3
 ucIsVeipMode        : 0
 ProUpResVlan        : 4095  pri                  : 7
 ProUserVlan         : 4095  pri                  : 7
 UsrNormalTime       : 300   UsrTempTime          : 30
 HashSearchIndex     : 75    SD5113L2Mode         : 1
 VendorFlag          : 0     NonFastLeave         : 0
 FastLeave           : 0     CtcNonFastLeave      : 0
 CtcFastLeave        : 0     MultiToUni           : 0
 Last5113Mod         : 2     IsPubCtc             : 0
 BtvQueryCvlan       : 0     BtvQueryMvlan        : 0
 PonIgmpAct          : to_cpu
-------------------------------------------------
```
```
IPTV global info:
    IGMPEnable      =  0
    ProxyEnable     =  0
    SnoopingEnable  =  1
    Robustness      =  2
    GenQueryInterval=  125
    GenResponseTime =  100
    SpQueryNumber   =  2
    SpQueryInterval =  10
    SpResponseTime  =  10
    RemarkIPPrecedence =  0
    RemarkPri          =  0
    IGMPVersion        =  2
    ActiveProgramNum   =  1
    STBNumber          =  1
    STBMAC             =
```

zhijianglee, 2023-02-02:
Finally, someone is in the same page with me.

zhijianglee, 2023-02-03:
@diogosilvan  I believe there' something related to igmp and multicast need to be configured on that stick 

OR 

Disable IGMP Snooping but need to set firewall results to drop repeated packets

diogosilvan, 2023-02-03:
> @diogosilvan I believe there' something related to igmp and multicast need to be configured on that stick
> 
> OR
> 
> Disable IGMP Snooping but need to set firewall results to drop repeated packets

Yes, it could be a stick setting, but I have no idea how to do that.
If you change the IGMP configuration, the multicast traffic stops all at once. It doesn't work at all.

I bought other equipment with SFP to test, I'm suspicious that it may be some incompatibility with the Mikrotik SFP in the interface management.
I even got a media converter that can once and for all solve this question of being the Mikrotik the cause of the problem.

zhijianglee, 2023-02-04:
I can affirm that it is due to the stick setting where you need to enable Snooping but disable IGMP for IPTV Vlan. 

I try to figure out the command one by one by using help and ? 

I think it might be possible to do so via diag

I hope with adequate feedback the vendor can imporve the firmware

yuri1313, 2023-02-26:
> > @diogosilvan I believe there' something related to igmp and multicast need to be configured on that stick
> > OR
> > Disable IGMP Snooping but need to set firewall results to drop repeated packets
> 
> Yes, it could be a stick setting, but I have no idea how to do that. If you change the IGMP configuration, the multicast traffic stops all at once. It doesn't work at all.
> 
> I bought other equipment with SFP to test, I'm suspicious that it may be some incompatibility with the Mikrotik SFP in the interface management. I even buy a media converter that can once and for all solve this question of being the Mikrotik the cause of the problem, after delivery i will do a test.

Hi,

I've been trying to set it up here with Vodafone PT, same issue as you guys - I either get internet + ip phone + web UI access to the ODI stick (with IGMP Snooping on) - no iptv, or iptv, but the rest stops working (the stick's web ui stops giving access and I'm unable to connect to the internet - but the iptv multicast traffic works ok). So your guess that this is probably overwhelming multicast traffic I'd say is spot on (I see a boost of about 30Mbps of traffic hitting my firewall on the IPTV vlan - normally I only see about 4 Mbps on that vlan). if I kill the IGMP Proxy running on the firewall I lose iptv, but eveerything else operates as expected.
I have an ubiquiti ONU that runs without any problems, but uses a different IGMP snooping process (the DFP stick uses igmpd while the ubiquti ONU uses mcpd with the following config):

igmp-default-version 3
igmp-query-interval 125
igmp-query-response-interval 10
igmp-last-member-query-interval 10
igmp-robustness-value 2
igmp-max-groups 25
igmp-max-sources 10
igmp-max-members 25
igmp-fast-leave 1
igmp-admission-required 1
igmp-admission-bridging-filter 1
igmp-proxy-interfaces
igmp-snooping-interfaces br0 br1 br2 br3
igmp-mcast-interfaces epon0.1  gpon0.0

My guess is that the ubiquiti ONU is operating as an IGMP proxy (similar to the IGMP Proxy you need to run in, for example, an Opnsense/PFSense firewall to get your STB to operate in multicast mode.

As for the PON stick, the only difference I've seen when setting up IGMP snooping On or off is this setting in the device's flash:
<Value Name="SNOOPING_ENABLED" Value="1"/>, which seems to trigger something in the igmpd daemon running in the stick by default (if I kill that process in the gpon stick, it starts operating as if the Snooping enabled variable is set to 0).

iptables, ebtables and other configs (including the IGMP config) remain the same...

ToTheCLI, 2023-03-08:
I would suggest trying the Huawei MA5671A SFP with latest right.cn.com firmware.

zhijianglee, 2023-04-15:
ToTheCLI,

I wonder if you are commenting for the sake of commenting. The SFP Module
this thread referring does't belongs to Huawei

*"A joke is a very serious thing." Winston Churchill*
*LEE ZHI JIANG*
*017-2691892*

diogosilvan, 2023-04-15:
> I would suggest trying the Huawei MA5671A SFP with latest right.cn.com firmware.

Is there a tutorial in English on how to do this? I don't understand Chinese at all. I know that here is another SFP model, but I have one of those and would like to test this firmware.

ToTheCLI, 2023-04-15:
> > I would suggest trying the Huawei MA5671A SFP with latest right.cn.com firmware.
> 
> Is there a tutorial in English on how to do this? I don't understand Chinese at all. I know that here is another SFP model, but I have one of those and would like to test this firmware.

https://hack-gpon.github.io/ont-huawei-ma5671a/

I have a Huawei OLT and with the Chinese custom frimware on MA5671A I got PVID and IPTV working.