Anime4000/RTL960x #126: Using programmator to reflash firmware

Braidon, 2023-01-30:
Hi!
I faced with the same problem was described at https://github.com/Anime4000/RTL960x/issues/46
I tried to erase mtd3 through UART but without luck - after power on, stick starts to send infinite text junk.
So the question is can i use a programmer (like this https://aliexpress.ru/item/1005002310286412.html?spm=a2g2w.favourites.mywishlist.2.4a234aa6QPPvIo&sku_id=12000029976241404)
to reflash wirmware or modify file system (remove mtd3 for example).

Thanks!

Anime4000, 2023-01-31:
yes you can, SPI Flash used on PON Stick is very generic

Sergey-SRG, 2023-02-03:
@Anime4000 Good afternoon, how can I delete the mtd3 partition from fullflash? (What address is it located at?) For further firmware by the programmer.

Sergey-SRG, 2023-02-03:
I understand 0x000000044000-0x000000080000 : "config" is it?

tdmadam, 2023-02-03:
> I understand 0x000000044000-0x000000080000 : "config" is it?

Yes, that's where the _mtd3 config_ is stored.
The 8MiB flash for the Realtek GPON SoC has the following layout.

| Offset | Length | Description |
|--------------|---------------|-----------|
| 0x000000 | 0x040000 | boot |
| 0x040000 | 0x002000 |  env |
| 0x042000| 0x002000 |  env2 |
| 0x044000 | 0x03c000  | config |
| 0x080000 | 0x3C0000 | image0 |
| 0x440000 | 0x3C0000  | image1 |