Anime4000/RTL960x #127: Indian ISP Airtel: Working Configurations

aki263, 2023-02-20:
ISP: Airtel
Country: India
CPE: Sercomm
Model: AOT-4221SR
GPON SFP: ODI DFP-34X-2C2 (XPON)

Parameters changed/updated
```
flash set OMCI_FAKE_OK 1
flash set PON_VENDOR_ID  SCOM
flash set GPON_SN <from device>
flash set ELAN_MAC_ADDR <from deice>
flash set OMCI_SW_VER1 <from device>
flash set OMCI_SW_VER2 <from device>
flash set HW_HWVER <from device>
flash set GPON_ONU_MODEL <from device>
flash set VLAN_CFG_TYPE 1
flash set VLAN_MANU_MODE 0
flash set VLAN_MANU_TAG_VID 100
```

reboot after making changes

[image]

[image]

Shaan-hbli, 2023-04-26:
Nice.....
Did you get VOIP working on the ODI DFP-34X-2C2 (XPON)...?

aki263, 2023-04-26:
@Shaan-hbli I didn't try it as I don't need that. But I think its doable as ODI support vlan which is main thing for VOIP as AIRTEL uses Vlan ID 660.

Shaan-hbli, 2023-04-27:
Nice.....
Did you get VOIP working on the ODI DFP-34X-2C2 (XPON)...?

> @Shaan-hbli I didn't try it as I don't need that. But I think its doable as ODI support vlan which is main thing for VOIP as AIRTEL uses Vlan ID 660.

Thanks for your quick reply........

xbipin, 2023-05-01:
they gave me a nokia router, other than mac id i have configured the rest, few days back it was working, today i got a mikrotik router with sfp cage and tried again and its stuck in 03 state, any clue whats wrong?

Shaan-hbli, 2023-05-01:
> they gave me a nokia router, other than mac id i have configured the rest, few days back it was working, today i got a mikrotik router with sfp cage and tried again and its stuck in 03 state, any clue whats wrong?

Can you post your configuration here...

physx2494, 2023-05-02:
From where did you buy the xPON, Aliexpress seller is quite expensive to ship here

aki263, 2023-05-02:
> From where did you buy the xPON, Aliexpress seller is quite expensive to ship here

This whole endeavour is expensive, getting gpon sfp, sfp+ card and then a dedicated PC to run it.

aki263, 2023-05-02:
Added link for video tutorial for same in original post.

xbipin, 2023-05-02:
> > they gave me a nokia router, other than mac id i have configured the rest, few days back it was working, today i got a mikrotik router with sfp cage and tried again and its stuck in 03 state, any clue whats wrong?
> 
> Did you enabled OMCI_FAKE_OK ?

yes set to 1

xbipin, 2023-05-02:
btw i added the module to a mikrotik hex s, the vlan mentioned in your config doesnt set vlan 100, i checked in the module it seems airtel is using vlan 2271 here but in airtel stock router its set to 100, no idea how to map it to 2271 to 100. i got pppoe with ipv4 and ipv6 working, next step is to make voip work, any clue on that. The modules does get extremely hot though

physx2494, 2023-05-03:
Is the ELAN_MAC_ADDR GPON's MAC address as shown on the Dragonpath 707GR1 router webui or the router's MAC address printed on the back of the router?

aki263, 2023-05-03:
> Is the ELAN_MAC_ADDR GPON's MAC address as shown on the Dragonpath 707GR1 router webui or the router's MAC address printed on the back of the router?
> 
> Also, I couldn't find anything related to HW_HWVER in 707GR1.

You can find mac address written on gpon itself. If thats not an option, call airtel and tell them that your cpe is unstable, they will replace it.

aki263, 2023-05-03:
> > > From where did you buy the xPON, Aliexpress seller is quite expensive to ship here
> > 
> > 
> > This whole endeavour is expensive, getting gpon sfp, sfp+ card and then a dedicated PC to run it.
> 
> do you want the module, im planning to order a few for commercial use to india

I already have like 15 of them running, thanks for the offer though.

physx2494, 2023-05-03:
@aki263 Can you please confirm the values for `flash get OMCI_OLT_MODE` per your configurations?
I ask this because the values `OMCI_SW_VER1` and `OMCI_SW_VER2` won't stick for me when OMCI_OLT_MODE is 0. Every time they revert to the firmware defaults as discussed here #77

https://github.com/rajkosto/RTL960x/blob/main/Docs/Setup_Stick.md#omci-olt-mode

chinmaythosar, 2023-05-05:
> ISP: Airtel Country: India CPE: Sercomm Model: AOT-4221SR GPON SFP: ODI DFP-34X-2C2 (XPON)
> 
> Video Tutorial for same https://youtu.be/MrVvszeQeuM
> 
> Parameters changed/updated
> 
> ```
> flash set OMCI_FAKE_OK 1
> flash set PON_VENDOR_ID  SCOM
> flash set GPON_SN <from device>
> flash set ELAN_MAC_ADDR <from deice>
> flash set OMCI_SW_VER1 <from device>
> flash set OMCI_SW_VER2 <from device>
> flash set HW_HWVER <from device>
> flash set GPON_ONU_MODEL <from device>
> flash set VLAN_CFG_TYPE 1
> flash set VLAN_MANU_MODE 0
> flash set VLAN_MANU_TAG_VID 100
> ```
> 
> reboot after making changes
> 
> [image]
> 
> [image]

where did you find this module in India ? I have TataPlay fiber and recently managed to get a Nokia G-010S-A which gets O5 but I lose connectivity because of some weird OMCI stuff. This one might work with fake omci which is not present in the Nokia module. But I am unable to find this for a reasonable price anywhere. Shipping from outside India costs more than the module itself.

xbipin, 2023-05-05:
> Is the ELAN_MAC_ADDR GPON's MAC address as shown on the Dragonpath 707GR1 router webui or the router's MAC address printed on the back of the router?
> 
> Also, I couldn't find anything related to HW_HWVER in 707GR1.

the one printed on back of router is fine or if you dont clone still it should work, i decrypted the nokia router config and got the mac id for pppoe connection from that and cloned that

xbipin, 2023-05-05:
> OMCI_OLT_MODE=0

> @aki263 Can you please confirm the values for `flash get OMCI_OLT_MODE` and the firmware version per your configurations? I ask this because the values `OMCI_SW_VER1` and `OMCI_SW_VER2` won't stick for me when OMCI_OLT_MODE is 0. Every time they revert to the firmware defaults as discussed here #30
> 
> https://github.com/rajkosto/RTL960x/blob/main/Docs/Setup_Stick.md#omci-olt-mode

first set OMCI_OLT_MODE=0 then set sw1 and sw2 and then reboot and it should stick

xbipin, 2023-05-05:
> I already have like 15 of them running, thanks for the offer though.

are you selling them or know any1 selling them in india?

xbipin, 2023-05-05:
> > btw i added the module to a mikrotik hex s, the vlan mentioned in your config doesnt set vlan 100, i checked in the module it seems airtel is using vlan 2271 here but in airtel stock router its set to 100, no idea how to map it to 2271 to 100. i got pppoe with ipv4 and ipv6 working, next step is to make voip work, any clue on that. The modules does get extremely hot though
> 
> Not sure, airtel uses 100 at every cpe that I have looked at. 2271 is super odd, maybe its due to location? Which city you live in?

in the nokia router it shows vlan 100 only but the gpon module gets 2271 on which pppoe and ipv6 works, i guess the nokia router does some vlan mapping

smnrock, 2023-06-27:
> btw i added the module to a mikrotik hex s, the vlan mentioned in your config doesnt set vlan 100, i checked in the module it seems airtel is using vlan 2271 here but in airtel stock router its set [...]

Is it 2271 or 100 which one was finally working? Did you find 2271 using omcicli mib 84 command?

Any specific settings needed in HEX S?

judge2112, 2023-06-28:
Can this SFP work with sophos firewall ?

smnrock, 2023-06-29:
> The aim of this complete testing was to increase the speed of the internet connection at our home. without paying more. ?

I dont think this will increase your plan speed.

Anime4000, 2023-06-29:
I don't think Sophos Firewall support HiSGMII, most of their equipment are 1/10G only, no 2.5G and 5G which very stupid and dumb.

even they support 2.5G, only NBase-T and not HiSGMII or 2500Base-X

which is make Sophos Firewall very stupid.

if you have 1Gbps 2Gbps internet plan, don't use Sophos Firewall, Fortinet, Aruba or similar as they don't support GPON 2.5Gb

plus with firewall and IDS/IPS will slow your internet as 200MB/s of data need to process which overkill the Router CPU.

I ranting here because people keep messaging me their Fortinet can't connect with any GPON ONU SFP.

for me, Equipment don't support HiSGMII or 2500Base-X is stupid, please don't buy it and boycott it, this will force manufacturers to support HiSGMII and 2500Base-X!

rajil, 2023-08-18:
Was anybody successful in mimicking a Huawei HG8145V5 router?

Unfortunately, with following settings i am stuck on O0 state

```
OMCI_OLT_MODE=0
GPON_ONU_MODEL=HG8145V5
OMCI_FAKE_OK=1
PON_VENDOR_ID=HWTC
GPON_SN=HWTCXXXXXX
ELAN_MAC_ADDR=e0xxxxx
OMCI_SW_VER1=V5R019C00S107
OMCI_SW_VER2=V5R019C00S107
HW_HWVER=15AD.A
GPON_ONU_MODEL=HG8145V5
VLAN_CFG_TYPE=1
VLAN_MANU_MODE=0
VLAN_MANU_TAG_VID=100
```
The mac address on the bottom sticker of HG8145V5 was used for ELAN_MAC_ADDR.

Anime4000, 2023-08-19:
> Unfortunately, with following settings i am stuck on O0 state

when change MAC Address, need generate new MAC_KEY

rajil, 2023-08-26:
Thanks with a new MAC_KEY i was able to connect.

Airtel also provides VOIP on VLAN 660 and they show up as follows:

```

# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x01
FilterTbl[0]: PRI 0,CFI 0, VID 100
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x02
FilterTbl[0]: PRI 0,CFI 0, VID 660
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 100
FwdOp:  0x04
NumOfEntries: 1
=================================
```

However, I have not been successful in having traffic pass through VLAN 660. It seems there is some filter on the ISP side which is blocking access.

KillerRoi, 2023-09-13:
> ISP: Airtel Country: India CPE: Sercomm Model: AOT-4221SR GPON SFP: ODI DFP-34X-2C2 (XPON)
>
> Video Tutorial for same https://youtu.be/MrVvszeQeuM
>
> Parameters changed/updated
>
> ```
> flash set [...]

My hardware version is 10, will this work? I'm in bangalore.

rajil, 2023-09-23:
> Thanks with a new MAC_KEY i was able to connect.
>
> Airtel also provides VOIP on VLAN 660 and they show up as follows:
>
> ```
>
> # omcicli mib get 84
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> VlanTagFilterData [...]

Was able to get voip working working after changing FwdOp of VID 660 from 0x04 to 0x02,
```
# omcicli mib set 84 0x02 FwdOp 0x02
```

iprateekjoshi, 2023-11-10:
@aki263, Do we have any module ([TP-Link-MC110CS-100Mbps-Single-Mode-Converter](https://www.amazon.in/TP-Link-MC110CS-100Mbps-Single-Mode-Converter/dp/B0034CSUD0/ref=sr_1_3?keywords=GPON+fiber+to+lan&s=computers&sr=1-3))which can be connected directly to Fiber cable and then to TP Link router ? Will the SFP media converters work ? Going through the post, pls link if these details are available in post.

DroidDotThen, 2023-12-01:
> ISP: Airtel Country: India CPE: Sercomm Model: AOT-4221SR GPON SFP: ODI DFP-34X-2C2 (XPON)
> 

hi @aki263 kinda off topic but I have got this ont today. Version 3.6. settings are unlocked but once I connect to internet it will get forced fw update and all settings will be locked. Can you pls share any way to get rid of airtel forced updates? to disable the tr069 (which is now locked). Any help will really be appreciated thx

Strykar, 2023-12-05:
> @aki263, Do we have any module [...]

What are you trying to achieve by removing the ISP optics?
Forget single multi mode fiber, look up PON, you need something like one of the routers from https://www.fs.com/c/gpon-epon-ont-onu-4099 or your ISP supplied xPON router.

Issue is none of these routers have enough oomph to SQM at Gigabit speeds, you can test your fancy fiber connections bufferbloat at https://www.waveform.com/tools/bufferbloat.

Strykar, 2023-12-05:
> hi @aki263 kinda off topic but I have got this ont today. Version 3.6. settings are unlocked but once I connect to internet it will get forced fw update and all settings will be locked. Can you pls [...]

Can you SSH / telnet in to it?
Do you have root?
Does the web  interface allow you to run terminal commands?

Disabling TR-069 can usually be as simple as killing the tr69 daemon running on the router, see https://github.com/Strykar/H680GM-A

Strykar, 2023-12-08:
Anyone interested in a group buy of the ODI DFP-34X-2C2 (XPON) ?

Rinkusolanki9827, 2024-01-02:
> ISP: Airtel Country: India CPE: Sercomm Model: AOT-4221SR GPON SFP: ODI DFP-34X-2C2 (XPON)
>
> Video Tutorial for same https://youtu.be/MrVvszeQeuM
>
> Parameters changed/updated
>
> ```
> flash set [...]

Kya is ka video mil sakta he

smnrock, 2024-03-01:
> my config if any1 wants to make it work in place of nokia stock router
> 
> > OMCI_OLT_MODE=0
> > OMCI_FAKE_OK=1
> > GPON_ONU_MODEL=G-2425G-A
> > GPON_SN=ALCLXXXXXXXX
> > PON_VENDOR_ID=ALCL
> > HW_HWVER=3FE48299DEAA
> > OMCI_SW_VER1=3FE49362JJIJ50
> > OMCI_SW_VER2=3FE49362JJIJ50
> > OUI=XX,XX,XX
> > HW_SERIAL_NO=XXXXXXXXXXXXXXXXXX
> > HW_CWMP_MANUFACTURER=ALCL
> > HW_CWMP_PRODUCTCLASS=HGU_EPON
> > OMCI_VEIP_SLOT_ID=255
> > GPON_PLOAM_FORMAT=0
> > GPON_PLOAM_PASSWD=44454641554C54303132
> > ELAN_MAC_ADDR=XXXXXXXXXXXX
> > MAC_KEY=XXXX
> > VLAN_CFG_TYPE=1
> > VLAN_MANU_MODE=0
> > VLAN_MANU_TAG_VID=100

After setting ELAN_MAC_ADDR and MAC_KEY, are you able to use VLAN as 100 or still you need to use 2271?  Also is it your OLT is from ALCATEL?

xbipin, 2024-03-02:
> After setting ELAN_MAC_ADDR and MAC_KEY, are you able to use VLAN as 100 or still you need to use 2271? Also is it your OLT is from ALCATEL?

Vlan 100 doesn't work because I can't get the gpon module to translate so I simply use 2271, my olt is nokia. It's not mandatory to set Mac id because this is gpon so Mac id isn't important

risingphoenix23, 2024-03-06:
@xbipin do you still need the sfp I have a spare can sell to you

navaneeth-dev, 2024-03-13:
@risingphoenix23 I am interested.

risingphoenix23, 2024-03-13:
Ok Please Ping me on WhatsApp
+919006431737

Ry-9n, 2024-05-16:
Bro upload the video elsewhere @aki263 
Yt one doesn't Last long

risingphoenix23, 2024-05-16:
Obviously 😂 airt#_'el is behind that

Strykar, 2024-09-02:
> A little off topic but really needed. The same Sercom device 4221SR is locked with no firmware available on internet. Can someone help me with the firmware or unlocking of the device. I see there are other Sercom devices which can be flashed through a CLI tool but not 4221SR. Please contact me at [vikasji1234@g-m-a-i-l.com](mailto:vikasji1234@g-m-a-i-l.com) if anyone can help. (remove dash from email)

What gives you the impression its the same device?

shubhamkastwar1, 2024-11-10:
Hey, Please help!

I bought xpon and put down below details on it however i am not able to authenticate via PPPoE. I am connecting it through OPNsense box so do not have access to GUI for now but can access xpon through telnet. 

Vendor Name: ODI
Part Number: DFP-34X-2C2

flash set OMCI_FAKE_OK 1
flash set PON_VENDOR_ID SCOM
flash set GPON_SN xxx
flash set ELAN_MAC_ADDR xxx
flash set OMCI_SW_VER1 xxx
flash set OMCI_SW_VER2 xxx
flash set HW_HWVER 8.0
flash set GPON_ONU_MODEL AOT-4221SR
flash set VLAN_CFG_TYPE 1
flash set VLAN_MANU_MODE 0
flash set VLAN_MANU_TAG_VID 100
flash set OMCI_OLT_MODE 21 

Below are some logs 

pon get transceiver bias-current
Bias Current: 0.000000 mA

pon get transceiver part-number
Part Number: DFP-34X-2C2

pon get transceiver temperature
Temperature: 42.019531 C

pon get transceiver vendor-name
Vendor Name: ODI

pon get transceiver voltage
Voltage: 3.217100 V

gpon get onu-state
**Error (0x120007): GPON Device has not been activeted**

shubhamkastwar1, 2024-11-10:
When fiber is plugged in 
omcicpon get transceiver rx-power
Rx Power: -21.487417  dBm

omcicli gepon get transceiver temperature
Temperature: 42.019531 C

pon get transceiver tx-power
Tx Power: -inf  dBm

Strykar, 2024-11-10:
> Hey, Please help!
> Vendor Name: ODI Part Number: DFP-34X-2C2
> gpon get onu-state **Error (0x120007): GPON Device has not been activeted**

Follow the [instructions](https://github.com/Anime4000/RTL960x/blob/main/Docs/Setup_Stick.md) carefully, in order, looks like you need to update the [MAC_KEY](https://github.com/Anime4000/RTL960x/blob/main/Docs/Setup_Stick.md#dfp-34x-2c2-1).

shubhamkastwar1, 2024-11-10:
Yes but i do not know how to generate MAC_KEY do you know the syntax to convert it to MD5

shubhamkastwar1, 2024-11-10:
I got the mistake i was using hsgq1.9a[xx:xx:xx:xx:xx] instead of hsgq1.9axx:xx:xx:xx:xx Let me give this a try

shubhamkastwar1, 2024-11-10:
> Follow the [instructions](https://github.com/Anime4000/RTL960x/blob/main/Docs/Setup_Stick.md) carefully, in order, looks like you need to update the [...]

Hi Strykar  i Tried the MAC_KEY bit no luck, I am working with airtel verified all details 3 times now. any idea what i might be doing wrong please let me know if any logs are needed :(

Anime4000, 2024-11-11:
GPON doesn't care ONT Mac Address, so, no need to change it

second, if you still need it, try run this python code:
https://gist.github.com/rajkosto/29c513b96ea6262d2fb1f965a52ce16f

shubhamkastwar1, 2024-11-11:
> Hey, Please help!
>
> I bought xpon and put down below details on it however i am not able to authenticate via PPPoE. I am connecting it through OPNsense box so do not have access to GUI for now but [...]

UPDATE!!!

I am not sure why but the VLAN setting was not saving when using telnet i bought a SFP to ethernet converter and connected it to my PC via webGUI I was able to change the VLAN settings to 100 then commit save the config and now it works for me.

Now i think of it i would've installed windows on the machine and used the inbuilt SFP port to configure it via GUI and might have saved 1600 buks. 😢 . 

however thanks [Strykar](https://github.com/Strykar) and [Anime4000](https://github.com/Anime4000) for quick response 😄

aki263, 2024-11-12:
I have reuploaded the video, https://youtu.be/6swUsegUThE. I have been using same GPON stick from last 2 years without any issue.

shubhamkastwar1, 2024-11-12:
> I have reuploaded the video, https://youtu.be/6swUsegUThE. I have been using same GPON stick from last 2 years without any issue.

are you using GPON or XPON SFP ?

aki263, 2024-11-12:
> > I have reuploaded the video, https://youtu.be/6swUsegUThE. I have been using same GPON stick from last 2 years without any issue.
> 
> are you using GPON or XPON SFP ?

XPON---xpon has both epon and gpon support.

shubhamkastwar1, 2024-11-17:
@aki263 

To clarify, I purchased two XPON modules from AliExpress. They have UPC SC connectors, used a patch cord APC SC to UPC SC. I connected them the same way I did with my second SFP module.

I was able to configure everything except the VLAN settings using the same commands. To get VLAN working, I had to connect the XPON to a Gigabit converter and configure the VLAN settings through the GUI.

I'm unsure if this is a specific issue with my device or if it's related to using XPON instead of GPON. I've attached a photo of the XPON module for reference. I'm aware of the differences between XPON and GPON, and I'm wondering if you've used GPON modules and experienced different behavior."
[image]

aki263, 2024-11-18:
> @aki263
> 
> To clarify, I purchased two XPON modules from AliExpress. They have UPC SC connectors, used a patch cord APC SC to UPC SC. I connected them the same way I did with my second SFP module.
> 
> I was able to configure everything except the VLAN settings using the same commands. To get VLAN working, I had to connect the XPON to a Gigabit converter and configure the VLAN settings through the GUI.
> 
> I'm unsure if this is a specific issue with my device or if it's related to using XPON instead of GPON. I've attached a photo of the XPON module for reference. I'm aware of the differences between XPON and GPON, and I'm wondering if you've used GPON modules and experienced different behavior." ![20241110_205423](https://private-user-images.githubusercontent.com/85588170/386999922-0bc9b554-aa0a-4c00-8e60-f5b78ba17563.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5MjU5MDYsIm5iZiI6MTczMTkyNTYwNiwicGF0aCI6Ii84NTU4ODE3MC8zODY5OTk5MjItMGJjOWI1NTQtYWEwYS00YzAwLThlNjAtZjViNzhiYTE3NTYzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE4VDEwMjY0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2YTdhMGY0MGRlODg0Yjg1OTExZDI1Zjk5ODFiZjRmZWMzY2Y1NDcxODE0YTU0YmQ4ZGUzYjZhZWU4M2YwOWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0bi4Xa5O4yzlaYeQE1oFHD85wfTwK7a3GOQT-pXILRc)

Can you share link of aliepxress where you it from? I dont see ODI branding on this one so I am not sure is this is related to module or firmware.

ianautiyal, 2024-12-02:
Anyone need SFP (SC APC) modules?
I have a bunch of SFP (SC APC) modules. If anyone's interested, let me know!

humanglitch191, 2025-03-17:
> Anyone need SFP (SC APC) modules? I have a bunch of SFP (SC APC) modules. If anyone's interested, let me know!

Hi @ianautiyal I would like to purchase a module. How do I contact you for further steps? Can you share discord / WhatsApp?

shubhamkastwar1, 2025-06-19:
@aki263 
A month ago i had a issue when their was a upgrade of Airtel and since then my ONU stick is not working, Just wanted to check if your stopped working too or it is something changed in my infra 😢 .
I have 2 ONU stick both not working. 😟

Vishalrajput2, 2025-09-13:
> ISP: Airtel Country: India CPE: Sercomm Model: AOT-4221SR GPON SFP: ODI DFP-34X-2C2 (XPON)
>
> Video Tutorial for same https://youtu.be/6swUsegUThE
>
> Parameters changed/updated
>
> ```
> flash set [...]
Could you help for BSNL FTTH if its work?