Anime4000/RTL960x #128: Question about which ONU is better in my case

und3rd06012, 2023-02-25:
Hello,

I have an ftth connection and I am trying to replace the ont. The ISP provides us with the Nokia G-010G-Q. I tried myself several configuration with a carlitoxx v2 module but it didn't work. Anime4000 suggested to by an ODI. So I bought a DFP-34X-2C2 from aliexpress. I tried the default firmware with several configs but nothing worked. I tried several configs with the M114_sfp_ODI_hybrid_220916 but again no luck. I tried the V2801F_V1.9.0-220425 with many different configs and not only nothing worked but the module hangs. I mean I am trying to access it and briefly it gives me access from the webui and then it doesn't respond at all. Neither http requests nor telnet requests are going through. It is like it works for a few seconds and then it dies. The ISP OLT is probably the demonic ALCATEL/LUCENT. I have read everything here I have tried everything:

I have set just the sn and ploam pass and nothing worked.
I have set just the sn and ploam pass and nothing worked with the zte olt config and again nothing.
I tried to spoof everything with these commands:

flash set PON_VENDOR_ID ALCL
flash set HW_CWMP_MANUFACTURER Nokia
flash set HW_CWMP_PRODUCTCLASS G-010G-Q
flash set HW_HWVER  {{ HW_VERSION }}
flash set ELAN_MAC_ADDR  {{ NEW_MAC_ADDRESS }}
flash set VS_AUTH_KEY {{ PON_ENCRYPTION_KEY }}

flash set OMCI_FAKE_OK 1
flash set OMCI_OLT_MODE 3
flash set OMCI_SW_VER1  {{ PRIMARY_SW_VERSION }}
flash set OMCI_SW_VER2  {{ BACKUP_SW_VERSION }}
flash set GPON_ONU_MODEL  G-010G-Q
flash set GPON_SN  {{ GPON_SN }}
flash set GPON_PLOAM_PASSWD  {{ SLID_OR_PLOAM_PASSWORD }}

reboot

You guesses it didn't work. (disclaimer: I generated a new vs_auth_key with the generator)

My ISP is Cosmote in Greece. 
Right now the ODI onu is pretty much inaccessible. I do not know if it died due to malfunction or somehow it got locked...
To give you a brief history of the states:

Carlitoxx with all the configs: O5 (fake operational I couldn't do a pppoe connection and receive an IP)
ODI with default firmware with all the configs: O5 (fake operational I couldn't do a pppoe connection and receive an IP)
ODI with M114_sfp_ODI_hybrid_220916 and all the configs: O3 & O5 but it stated LOID (WRONG)
ODI with V2801F_V1.9.0-220425 and all the configs: O0 

Now the ODI is pretty much inaccessible.

Could anyone please help? I am quite desperate. I mean I have spent countless hours. I think the main problem is that the onu sticks are unable to do the VLAN mapping correctly as I am executing omcicli mib get 84 it outputs nothing and when i execute diag + l2-table get entry address valid and I observe that the VLAN is 0.
If I need to buy a new stick please tell me which one and I will buy it. Somewhere here it was stated that in case of the ALCATEL/LUCENT it is better to buy a Nokia ONU Stick. Is this true? 

P.S.1: Carlitoxx module is also dead. I flashed it with V2801F_V1.9.0-220425 and I forgot to set LAN_SDS_MODE to 1 and I lost access.
PS.2: My provider uses only 1 vlan for everything. It doesn't seperate voip, iptv, and internet.

Anime4000, 2023-02-25:
since you change the SW Version, you need set `OMCI_OLT_MODE 21` on ODI Stick ?
because SW Version will revert back

und3rd06012, 2023-02-25:
Thank you for your response.OK. I didn't know that. 
Since now the stick is inaccessible do I need to buy a UART device to recover?
If yes, something like this will do the work: https://topelectronics.gr/electronics/converters/usb-2.0-to-ttl-uart-5pin-module-serial-converter-cp2102-stc-prgmr/

Anime4000, 2023-02-25:
if ODI can't be access, yes, you need UART for that

und3rd06012, 2023-02-25:
I am going to buy the one on the link first thing in the morning. Do you have a diagram on how to connect/probe the uart on the onu network chip? Also, are there steps on what should I do on the PC to recover it?

und3rd06012, 2023-02-25:
Anime do you have an email I could reach you?

Anime4000, 2023-02-25:
> I am going to buy the one on the link first thing in the morning. Do you have a diagram on how to connect/probe the uart on the onu network chip? Also, are there steps on what should I do on the PC to recover it?

diagram to UART
https://github.com/Anime4000/RTL960x/blob/main/Docs/UART.md

also, you just need erase mtd3 (config) to factory reset:
https://github.com/Anime4000/RTL960x/blob/main/Docs/factory_reset.md#reset

Anime4000, 2023-02-25:
or you can join here https://discord.gg/hZ5eAHzE by stich86