Anime4000/RTL960x #131: Fiberhome

john01031, 2023-03-08:
Hi, I'm not sure what I'm doing wrong, I added all the details and I got in the following loop o2 to o5 initial status, my ISP uses Fiberhome, I wonder if it is a no-go on this platform since there is no Ploam password:

[image]

Anime4000, 2023-03-09:
fiberhome OLT have many restriction, have you set OMCI_OLT_MODE to 21 ?

john01031, 2023-03-09:
Yes, I've tried that, with no luck, I have root access to the fiberhome ONT and I can't find the ploam information anywhere on it, loid and loid pass is found and other information are available but that

Anime4000, 2023-03-09:
MAC Address also change? together with generate MAC_KEY

john01031, 2023-03-09:
Yes, that too

Anime4000, 2023-03-09:
can show share me the flash get you set?

john01031, 2023-03-10:
[image]

john01031, 2023-03-10:
I've got to see some OMCI messages on my original onu and it reports this:
[image]
However, even with this information, it is a no go :( 
It also converts the serial to hex:
[image]
then the only thing mentioning PLOAM appears, and then it synchronizes:
[image]

nguyenthanhhong-tg, 2023-03-10:
Seems it must authenticate with PLOAM. Make sure your PLOAM correct.

john01031, 2023-03-10:
I see... lets's hope to find someone who had luck with Fiberhome, probably something on the firmware maybe?

Seeing on the Fiberhome onu firmware I only can find the default values
[image]

john01031, 2023-03-10:
I was able to force the telnet server to start before fiber sync, then I made the change permanent on /etc/rc.d/ I had the admin password before in my case, if you do not have that you will need to read the chipboard

[image]

john01031, 2023-03-10:
It seems your provider uses some custom stuff on the firmware, maybe you're going to need to use the serial interface, have you tried root / root?

john01031, 2023-03-10:
I'm starting to think the same over here... this platform sucks big time

john01031, 2023-03-10:
I just found this schematic on **"Van Tech Corner"** youtube channel, I'll try it later with the serial number as the PLOAM password

[image]

john01031, 2023-03-10:
Not results from my side with the SN as ploam, haha

I wonder if someone ever made this work with fiberhome
I'm running the firmware version:
[image]

nguyenthanhhong-tg, 2023-08-04:
Router is FiberHome, but what is your ISP name?

sweetpotato133, 2023-10-11:
> Not results from my side with the SN as ploam, haha
> 
> I wonder if someone ever made this work with fiberhome I'm running the following firmware version: [image]

We are on the same boat, did you manage to fix this?

john01031, 2023-10-11:
No =/

Antex, 2023-10-16:
> Yes, I've tried that, with no luck, I have root access to the fiberhome ONT and I can't find the ploam information anywhere on it, loid and loid pass is found and other information are available but that
Can you please tell me where to find loid and loid pass, and if you know it, ploam password?

kevindd992002, 2023-10-16:
So is this for PLDT?

kevindd992002, 2023-10-16:
@sweetpotato133 made it work for him with PLDT by using the latest HGU firmware instead of the SFU. Mind sharing more details here for the benefit of everyone?

jerome997, 2024-01-03:
any luck? I have the same issue with PH ISP:Globe Telecom

john01031, 2024-01-03:
Man, I'm still out of luck, looks like Fiberhome is a no-go for this SFP

Anime4000, 2024-01-04:
My ISP use Fiberhome, but OLT is configured to accept any ONT as ISP told me

nguyenthanhhong-tg, 2024-01-05:
You meant OLT accepts any ONU?

Anime4000, 2024-01-05:
> You meant OLT accepts any ONU?
> […](#)

Yes, OLT model is AN5116-06B+

ISP didn't tell me how they configure it to accept any ONT

jerome997, 2024-01-10:
@john01031 
I've finally figured it out and made it work
need to set the RTK_* infos

in my case
`  
<Value Name="RTK_DEVID_MANUFACTURER" Value="fiberhome"/>
  <Value Name="RTK_DEVID_OUI" Value="000AC2"/>
  <Value Name="RTK_DEVID_PRODUCTCLASS" Value="AN5506_04F1A"/>
  <Value Name="RTK_DEVINFO_SPECVER" Value="1.0"/>
  <Value Name="RTK_DEVINFO_SWVER" Value="RP2615"/>
  <Value Name="RTK_DEVINFO_HWVER" Value="WKE2.134.285F1A"/>
  <Value Name="DEVICE_NAME" Value="GPON"/>
`

jerome997, 2024-01-11:
> What does RTK_DEVID_OUI mean?
> […](#)

https://github.com/Anime4000/RTL960x/blob/main/Docs/FLASH_GETSET_INFO.md#oui

kevindd992002, 2024-01-11:
> @john01031 I've finally figured it out and made it work need to set the RTK_* infos
>
> in my case
>
> ```
> <Value Name="RTK_DEVID_MANUFACTURER" Value="fiberhome"/>
> <Value Name="RTK_DEVID_OUI" [...]

So aside from modifying the usual parameters like GPON_SN, OMCI_SW_VER1, OMCI_SW_VER2, HW_HWVER, PON_VENDOR_ID, GPON_ONU_MODEL, OMCI_OLT_MODE, and OMCI_FAKE_OK, what made it work for you is to modify the RTK_* properties?

Which ISP do you have? I have this working for Converge and Globe but was having issues with PLDT (Fiberhome) when I tried it last year. I'm planning to make it work with PLDT real soon.

Also, what fw are you using? 220923 SFU? Or are you using HGU?

jerome997, 2024-01-11:
> So aside from modifying the usual parameters like GPON_SN, OMCI_SW_VER1, OMCI_SW_VER2, HW_HWVER, PON_VENDOR_ID, GPON_ONU_MODEL, OMCI_OLT_MODE, and OMCI_FAKE_OK, what made it work for you is to [...]

yup that's pretty much it i just supplied the RTK_* values and it fixed my issue for the looping of O2-O5 ONU State

Im using this firmware [https://github.com/rajkosto/RTL960x/blob/main/Firmware/DFP-34X-2C2/M110_sfp_ODI_220923FS.tar](url) SFU mode

kevindd992002, 2024-01-11:
Perfect! Thanks for the tip. I'll try it out with PLDT.

@rajkosto do you have any ideas as to why the RTK parameters were needed in this case?

Anime4000, 2024-01-12:
> Perfect! Thanks for the tip. I'll try it out with PLDT.
> 
> @rajkosto do you have any ideas as to why the RTK parameters were needed in this case?

I have checked with my ISP that using Fiberhome, it seem here doesn't read any RTK info, only standard OMCI

Anime4000, 2024-01-12:
Mark as solved, added [this](https://github.com/Anime4000/RTL960x/issues/131#issuecomment-1885032543) as part of Troubleshooting Readme

Feel free to re-open if anyone encountered

closed 2024-01-12 by Anime4000

rodfai, 2024-03-31:
My ONU is fiberhome as well. After setting RTK thing, I got O5 but that's it.  What should I look at after this ?  
Checking VLAN and nothing show up.   firmware 220923
 

[image]