Anime4000/RTL960x #133: O5 no Internet connection problem

Antonov225, 2023-03-14:
Hi, I am trying to replace my stock ISP provided GPON/router device as described in the guide, but no matter what I try, I cannot get an Internet connection. 

Hardware I am using:
- ODI DFP-34X-2C2 SFP module
- [Realtek 2.5Gb Media Converter](https://www.aliexpress.com/item/1005004340369253.html) - same one described as plug and play in the 2.5Gb Compatibility page
- Asus GT-AX6000 router
- SC/UPC male to SC/APC female adapter - needed because ISP provided cable has an APC connector and the ODI stick has an UPC connector

My ISP is a Polish company Fiberlink/Toya and they provided me with a GPON/router device - Halny HL-4GMV3 I want to replace.  Previously one of its LAN ports was connected to the ASUS router WAN port, creating a double NAT problem. This is what I want to avoid. It seems my ISP uses authentication by SN or MAC, because there is no PLOAM/LOID information anywhere in the Halny router Web GUI.
I replaced all stock ONU information I could (including the MAC address with a generated MAC_KEY) and I get an O5 state on the SFP stick - although I think O5 state was there even with stock information, before changing anything. I also tried setting OMCI_OLT_MODE to 21 but it didn't change anything. 
Additionally the ODI stick seems to be unstable - it reboots/crashes? constantly dropping the Telnet or Web GUI connection. After a minute or two it recovers and after next few minutes rebooting repeats - is this normal or is the stick defective?

Here is what the configuration looks like:

[image]

[image]

[image]

The PLOAM pasword field displays broken characters after I tried to change it via Telnet.

I don't have much experience with more advanced networking such as VLANs but I am suspecting the issue may lie in the Asus router WAN configuration.  If I am supposed to use PPPoE, I don't know what information to use in Asus settings panel. For now I set the WAN IP to static and in the same address range as the ODI stick and can acces it via telnet in this configuration:

[image]

I am attatching my current config file:

[config.zip](https://github.com/Anime4000/RTL960x/files/10970923/config.zip)

I would greatly appreciate any help. Thank you for reading.

ToTheCLI, 2023-03-14:
omcicli mib get 84
omcicli mib get 171
check output of these 2 commands in telnet

Antonov225, 2023-03-14:
@ToTheCLI Hi, here is the result:

[image]

ToTheCLI, 2023-03-14:
> 

Means you have probably fake O5

Try  
flash set OMCI_OLT_MODE 1
reboot

and what was the VLAN in Halny HL-4GMV3?

Antonov225, 2023-03-14:
After flashing OLT_MODE 1 and FAKE_OK 1 commands, there is no difference. Halny router/gpon is locked down hard by my ISP and most of the settings in its Web GUI are missing - I cannot see anything regarding VLANs there. Is there a different way to find this information?

Antonov225, 2023-03-14:
After quick googling I found this documentation:

https://halny.com/knowledge-base/halny-hl-4gmv-configuration-huawei/
https://halny.com/knowledge-base/halny-hl-4gmv2-configuration-nokia-7360/
https://halny.com/knowledge-base/kategoria/olt-configuration/page/5/

There is a knowledgebase on the Halny website describing how to configure different OLTs for the use with their devices. And there is information about VLANs there. I don't really have a way to know what type of OLT my ISP uses unfortunately.

keekkenen, 2023-03-18:
> The PLOAM pasword field displays broken characters after I tried to change it via Telnet.

I see on your screen ploam pass looks like incorrect. On the version 220923 in telnet for ploam pass you need using hex format (I think you set ASCII value).
flash get GPON_PLOAM_FORMAT must return 0
you need convert ploam pass to hex format and set it value

Antonov225, 2023-03-18:
Hi, yes I know I changed it later but still no success. I know my ISP uses a combination of serial number and vendor id for authentication, but no matter what combination of settings I try I cannot get any VLAN information and no Internet access.

keekkenen, 2023-03-18:
as usual stick have 192.168.1.1, your Asus home network must be in another subnetwork, for example, 192.168.0.1, after it need to change ethernet connection (means SFP connection) set in DHCP mode..

> I try I cannot get any VLAN information and no Internet access.
as usual VLAN ID (also like a PLOAM password and GPON serial number, MAC key) getting form device in use (means router which rent form your provider)

Antonov225, 2023-03-18:
Well, right now my Asus router (gateway) has the 192.168.50.1 address so it is in another subnetwork. If I change the connection type on WAN to Automatic IP, the router displays the message that my ISP DHCP does not work correctly.
Regarding the VLAN, I cannot check it from the ISP provided device because it is locked down in a way that most of its settings are hidden and inaccessable. What is meant is that VLANs accessed from the OLT should be visible via the diag l2-table or omcicli commands on the ODI stick. In my situation however there is nothing there.

keekkenen, 2023-03-18:
> Regarding the VLAN, I cannot check it from the ISP provided device because it is locked down 

In [docs](https://www.tesshop.sk/img.asp?attid=445856) looks like VLANID accessable in web-interface

Antonov225, 2023-03-18:
Yes it should be there by default, but my ISP modified the firmware on the Halny device and its web interface looks more like this:

[image]

Some menus/sections are missing. I'm wondering if there is some different way to find the VLAN numbers other than trial and error.

d5aqoep, 2023-05-28:
Just to help, You need PPPoE username and password from your ISP to do anything further. 
There is a cheap TP-Link XZ000-G3 ONU in the market which scans and tells you your actual VLAN ID for connection.

zentavr, 2024-03-16:
Had you any chance to dump the configuration?

zentavr, 2024-03-18:
@Antonov225 @d5aqoep - I have HL-4GMV4 device here and seems like it's more modified than @Antonov225 has in his post.

zentavr, 2024-03-18:
Found a similar thread here: https://trzepak.pl/viewtopic.php?t=69695

zentavr, 2024-05-30:
I was able to connect using serial interface and here are the dumps:
- Boot without PON connected: https://pastebin.com/uGmTne45
- Boot with PON connected: https://pastebin.com/bMmcDX6c
- Boot into the bootloader: https://pastebin.com/GNS9WCpZ

...if I have the bootloader, can I bypass the standard boot process and log into /bin/sh directly?

ajania, 2024-05-30:
Hi there, sorry wasn't following the topic for quite some time. Originally I decided that poking around in the isp router firmware was just not worth the risk and gave up on the whole idea. They have a literal monopoly on fiber internet in my region and I just cannot afford to get banned for violating their contract like that. Maybe I'll try some other time when I feel more brave. They also updated the firmware on my router remotely in the meantime and now the gui looks completely different. Not sure what changes were made under the hood. The post on trzepak.pl was made by me :), it's a polish networking forum focused on isp infrastructure. As you can see I didn't get much help there.
It is very interesting that you managed to connect to the router via serial. I haven't done this but I am pretty sure in my case access to the shell would be password protected. If you are looking for the pppoe credentials like me, the most obvious way to get them would be to extract the spi firmware from the device in its entitreity. Two methods come to mind: 
1. Using an external spi programmer/reader, for example the popular ch341a. You would need to find the physical spi rom chip in the router and connect it to the programmer, either by desoldering it or with included clips. From what I head the clips are often unreliable.
2. Interrupting the boot process and extracting the firmware via serial in plain text form. Here is the general idea: https://www.youtube.com/watch?v=I1w_HQ7soSE 
In your dump I can see several functions for reading and writing to memory locations, including the dump <addr> <len> command. You could use these.
I am interested if you succeed, good luck!

Antonov225, 2024-05-30:
Hi there, sorry wasn't following the topic for quite some time. Originally I decided that poking around in the isp router firmware was just not worth the risk and gave up on the whole idea. They have a literal monopoly on fiber internet in my region and I just cannot afford to get banned for violating their contract like that. Maybe I'll try some other time when I feel more brave. They also updated the firmware on my router remotely in the meantime and now the gui looks completely different. Not sure what changes were made under the hood. The post on trzepak.pl was made by me :), it's a polish networking forum focused on isp infrastructure. As you can see I didn't get much help there.
It is very interesting that you managed to connect to the router via serial. I haven't done this but I am pretty sure in my case access to the shell would be password protected. If you are looking for the pppoe credentials like me, the most obvious way to get them would be to extract the spi firmware from the device in its entitreity. Two methods come to mind:

Using an external spi programmer/reader, for example the popular ch341a. You would need to find the physical spi rom chip in the router and connect it to the programmer, either by desoldering it or with included clips. From what I head the clips are often unreliable.
Interrupting the boot process and extracting the firmware via serial in plain text form. Here is the general idea: https://www.youtube.com/watch?v=I1w_HQ7soSE
In your dump I can see several functions for reading and writing to memory locations, including the dump command. You could use these.
I am interested if you succeed, good luck!

zentavr, 2024-05-30:
[image]

There is a chip like this at the board, so in the theory it can contain the firmware.

zentavr, 2024-05-30:
@Antonov225 seems like there are no commands (or they are hidden?) in the uBoot. I was able to run only `go` which boots up the kernel and a couple of others:
```
bldr> mtd
0x00000000-0x00040000 : "bootloader"
0x00040000-0x00080000 : "romfile"
0x00080000-0x0041c9ea : "kernel"
0x0041c9ea-0x0109c9ea : "rootfs"
0x00080000-0x02080000 : "tclinux"
0x02080000-0x02414e2f : "kernel_slave"
0x02414e2f-0x03214e2f : "rootfs_slave"
0x02080000-0x04080000 : "tclinux_slave"
0x04080000-0x04880000 : "opt0"
0x04880000-0x05c80000 : "ubifs"
0x06dc0000-0x07000000 : "reservearea"

bldr> imginfo
os1:V3.1.28p2
os2:V3.2.40p2
```

zentavr, 2024-05-30:
Also I had noticed that PPPoE sessions is being created. Like:
```
Got connection: 8353
Connecting PPPoE socket: 6c:03:b5:b4:96:49 5383 nas1_0 0x7e8088
Using interface ppp8
Connect: ppp8 <--> nas1_0
Couldn't increase MTU to 1500.
Couldn't increase MRU to 1500
7613 sw rps
local  IP address 10.0.99.143
remote IP address 172.16.220.8
primary   DNS address 91.227.2.157
secondary DNS address 91.227.3.190
```

zentavr, 2024-05-31:
@Antonov225 BTW how did you get the PLOAM password (saw that in your early messages).

zentavr, 2024-06-14:
@Antonov225 got my CH341A today...
> Using an external spi programmer/reader, for example the popular ch341a. You would need to find the physical spi rom chip in the router and connect it to the programmer, either by desoldering it or with included clips. From what I head the clips are often unreliable.

The clips which came with the device cannot catch the legs of the chip... and I do not have any equipment to desolder the chip from the mainboard.

zentavr, 2024-06-14:
[image]
[image]
Regarding the [datasheet](https://www.esmt.com.tw/upload/pdf/ESMT/datasheets/F50L1G41LB(2M).pdf) seems like there could be 2 versions of such chip:

| Product ID | Speed    | Package |
|------------|---------|----------|
| F50L1G41LB-104YG2M | 104MHz | 8-contact WSON 8x6mm Pb-free |
| F50L1G41LB-104YG2ME | 104MHz | 8-contact WSON (without expose metal pad) 8x6mm Pb-free |

At my router I can see the pads, so it's **F50L1G41LB-104YG2M**, correct?

zentavr, 2024-06-30:
Bought [these clips](https://www.aliexpress.com/item/1005005452043600.html) but seems like contacts are too small to hold the chip.

Also, when I'd put the clips at the chip, I saw that the light diods at the router mainboard started to blink (for a second or something) and then the diods at the ch341a extinguished.
What I had noticed is that the firmware had changed...

Last time it was:
```
bldr> mtd
0x00000000-0x00040000 : "bootloader"
0x00040000-0x00080000 : "romfile"
0x00080000-0x0041c9ea : "kernel"
0x0041c9ea-0x0109c9ea : "rootfs"
0x00080000-0x02080000 : "tclinux"
0x02080000-0x02414e2f : "kernel_slave"
0x02414e2f-0x03214e2f : "rootfs_slave"
0x02080000-0x04080000 : "tclinux_slave"
0x04080000-0x04880000 : "opt0"
0x04880000-0x05c80000 : "ubifs"
0x06dc0000-0x07000000 : "reservearea"

bldr> imginfo
os1:V3.1.28p2
os2:V3.2.40p2
```

Today I have:
```
bldr> mtd
0x00000000-0x00040000 : "bootloader"
0x00040000-0x00080000 : "romfile"
0x00080000-0x00414e2f : "kernel"
0x00414e2f-0x01214e2f : "rootfs"
0x00080000-0x02080000 : "tclinux"
0x02080000-0x02414be9 : "kernel_slave"
0x02414be9-0x03234be9 : "rootfs_slave"
0x02080000-0x04080000 : "tclinux_slave"
0x04080000-0x04880000 : "opt0"
0x04880000-0x05c80000 : "ubifs"
0x06dc0000-0x07000000 : "reservearea"

bldr> imginfo
os1:V3.2.40p2
os2:V3.2.42
```

I saw that the equipment supports the update via DHCP (it requests special vendor's id). So maybe we can somehow emulate PON sessions and try to fetch the firmware from ISPs servers? I did not find it in the public