Anime4000/RTL960x #134: DFP-34X-2C2 show "No link" & web interface not accessible by ip after reboot on Mikrotik RB4011iGS+

rndm2, 2023-03-15:
Hi, 

I noticed strange behaviour of this module: sometimes after Mikrotik reboot it shows "No link" and not recovering itself even after some time, I need to reboot router 2-3 times (or power cycle) to get it working. At the moment when it shows "No link" SFP web interface is also unavailable so I even can't check what is going on with SFP at this time. 

Is it problem with my Mikrotik? With module? With hardware on other side? Bug? Any ideas how to fix? 

Thanks

rndm2, 2023-03-27:
Done. 

Fixed. Many thanks @akw28888

rndm2, 2023-05-05:
Got this issue again during firmware update. 
LAN_SDS_MODE set well. 

Any ideas?

Anime4000, 2023-05-06:
I have tried on RB4011, the SFP+ kind very picky sometime

rndm2, 2023-05-31:
They released new firmware today 
```
What's new in 7.9.2 (2023-May-30 16:49):

*) routerboard - improved RouterBOOT stability for Alpine CPUs ("/system routerboard upgrade" required);
*) sfp - improved SFP interface handling for RB4011 device;
```

But it didn't help :( 
Still need to reboot 2-3 times to get it working :(

ToTheCLI, 2023-06-03:
> if I reboot the router, should it also power-cycle SFP?

No, to power-cycle SFP you have to shutdown and disconnect power or remove and insert SFP.

rndm2, 2023-11-16:
I changed provider to Movistar. They switched me to another port on CTO, and now, when I try to do 
`cat /proc/lan_sds/lan_sds_cfg`
`flash set LAN_SDS_MODE X`

It works for a small period of time (10-15 minutes), and SFP becomes unresponsible.

bykidi, 2024-08-20:
> > if I reboot the router, should it also power-cycle SFP?
> 
> No, to power-cycle SFP you have to shutdown and disconnect power or remove and insert SFP.

while not mentioned in the changelogs, this problem is fixed with the latest 7.16rc2 (release-candiate) routeros.
you no longer have to power cycle SFP module after your mikrotik reboot.