Anime4000/RTL960x #135: Fake O5 Issue

sudozid, 2023-03-18:
Hi,

I have a Zyxel PMG5617-R20B on Airtel (India). I'm having fake O5 issue and have tried many combinations on Realtek but none seem to work. 

PON Status

Temperature | 59.820313 C
Voltage | 3.372800 V
Tx Power | 2.255692  dBm
Rx Power | -30.000000  dBm
Bias Current | 13.110000 mA

GPON Status
ONU State | O5
ONU ID | 0
LOID Status | Initial Status

This is the mib info from Zyxel:

https://pastebin.com/mdUFaJpX

And this is the mib info I have on Realtek:

https://pastebin.com/6m9aEdd5

sudozid, 2023-03-19:
Ok so I finally got it working after a lot of trial and error with the config.

Here is the mib info now:

https://pastebin.com/raw/YT11giWj

But there is a problem my speed is stuck at 16-20Mbps

https://beta.speedtest.net/result/14499126843.png

Shaan-hbli, 2023-04-26:
> Ok so I finally got it working after a lot of trial and error with the config.
> 
> Here is the mib info now:
> 
> https://pastebin.com/raw/YT11giWj
> 
> But there is a problem my speed is stuck at 16-20Mbps
> 
> https://beta.speedtest.net/result/14499126843.png
> 
> This is my ONU, it is Realtek but not in the sfp stick form
> 
> http://www.hsgq.com.cn/products_34/67.html

Can you please share your config........

BittorB, 2023-06-08:
i have the same issue with alcl olt but its some strange because the voip lan stablishes link and work but internet isnt working