Anime4000/RTL960x #136: ODI DFP-34X-2C2 not working

galvesribeiro, 2023-03-20:
Hey!

I bought and received the ODI DFP-34X-2C2 module which you suggested as the best one for sale from the link on AliExpress on our documents. Connected it on either a regular SFP media converter or my Ubiquiti Unifi Switch. 

I see the link (auto-negotiate 1G) but it doesn't respond to ping nor SSH/web on 192.168.1.1.

Do you have any idea how to get into this module? Is it actually a real thing or is it a scam?

Thank you!

Anime4000, 2023-03-20:
the SFP is like mini modem, depend on SFP host if support SFP MAC

ToTheCLI, 2023-03-20:
> Hey!
>
> I bought and received the ODI DFP-34X-2C2 module which you suggested as the best one for sale from the link on AliExpress on our documents. Connected it on either a regular SFP media converter [...]

The media converter you used probably requires the fiber to be attached with signal while TP-Link media converter does not need so, what is your media converter?

galvesribeiro, 2023-03-20:
Mine is TP-Link MC220L @ToTheCLI.

Thanks for all the replies. It is jut that the boot is very slow, but it worked. I'm able to use it with my ISP now.

Now the challenge is to figure out how to make it work with 2.5G with the Ubiquiti UDM-Pro.

galvesribeiro, 2023-05-10:
@Anime4000 / @ToTheCLI I finally got it to sync to my ISP. It worked just fine that I bought a second one to use in my other apartment.

The thing is, it works pretty well when I stick on the UDM-PRO or UDW and set them to 1GbE.

However, I have 1.25Gbps link and the way the stick is set with 1GbE, `LAN_SDS_MODE=0`, the default, which will start in mode 1, Fiber 1G.

Now I bought a Mikrotik CRS305-1G-4S+ switch. Then I made a bridge where I have 2 of the SFP+ ports bridged. In one of them, I have the ODI GPON stick, and another I have a 10GbE SFP+ module which is wired with fiber to my UDM-PRO or UDW. This allow me to set the WAN port on the UDM-PRO/UDW to 10GbE. 

However, the port where I have the GPON stick still in 1GbE. So far so good. I still have internet connection, PPPoE sync, everything is wonderful but, it still 1GbE. 

I then made the ODI GPON stick to use the `LAN_SDS_MODE=6`, which is "2500BaseX Fiber with auto-negotiation on". As soon as I set it and I put it on the Switch again, my internet connection never get back to work. I see the ODI stick SFP port is synced on 2.5Gbps, but for whatever reason it don't get the PPPoE to connect or anything :(

[image]

[image]

Do you have any idea if changing the `LAN_SDS_MODE` would affect the way the ODI stick syncs with my ISP that would prevent it to work? Or is that something else?

If I go to the telnet using Mikrotik/WinBox tools, it stands on O2:

```bash
# diag gpon get onu-state
gpon get onu-state 
ONU state: Standby State(O2) 
```

Any ideas?

Anime4000, 2023-05-10:
many user on V06 stick having issue 2.5Gb and upload issue

my test was on V05 stick, so, hardware change might cause issue too, I not too sure.

so far 2.5Gb was works on RB5009 via HSGMII PHY

galvesribeiro, 2023-05-10:
Do you suggest any other module? I bought this one because I saw you posting somewhere that this was the one which works correctly.

ToTheCLI, 2023-05-10:
> Do you suggest any other module? I bought this one because I saw you posting somewhere that this was the one which works correctly.

What OLT and ONT do you have? (e.g Huawei)
I would suggest you try Lantiq based GPON SFPs (MA5671A if you can find one that is unlocked or [FS.com](https://www.fs.com/products/133619.html) SFP) with Custom Frimware.

galvesribeiro, 2023-05-10:
@ToTheCLI it is Huawei on the ISP side.

I have one from FS.com GPON-ONU-34-20BI which I tried a while ago but I think it got bricked somehow as I can't get into it anymore and I have no tools or skills to wire/solder it to recover it.

Which one do you recommend?

- https://www.fs.com/products/133619.html (the one I have -  I can get a new one) or
- https://www.fs.com/products/141198.html

None of those on their site mention the it supports 2.5Gb tho...

> Try LAN_SDS_MODE=4

I have that now. I was looking into other post here and a guy got similar setup got it to work. 

I was able to get it to connect but yesterday it was unstable. After a while it disconnects. Since the module is inside the Mikrotik bridge, I can't access it to know why it is down. My suspicious is temps, but I can't confirm it from the terminal unless I remove it from the bridge.

ToTheCLI, 2023-05-10:
What is the Minimum amount of OMCI info you need to put in ODI SFP to get it working (SN, PON Vendor ID, OLT Mode,etc)?

Anime4000, 2023-05-10:
> What is the Minimum amount of OMCI info you need to put in ODI SFP to get it working (SN, PON Vendor ID, OLT Mode,etc)?

you can read here:
https://github.com/Anime4000/RTL960x/blob/main/Docs/Stock_ONU.md

since GPON not required MAC Address to set, this absolute minimum OMCI to work

ToTheCLI, 2023-05-10:
> @ToTheCLI it is Huawei on the ISP side.
>
> I have one from FS.com GPON-ONU-34-20BI which I tried a while ago but I think it got bricked somehow as I can't get into it anymore and I have no tools or [...]
The one you have
> 
> None of those on their site mention the it supports 2.5Gb tho...
2.5Gbps is supported with command (fw_setenv sgmii_mode 5)

galvesribeiro, 2023-05-10:
@ToTheCLI 

OMCI_FAKE_OK / OMCI_OLT_MODE / PON_VENDOR_ID / GPON_PLOAM_PASSWD / GPON_SN / LAN_SDS_MODE

Those are the ones I had to change to get it to work. That is kinda general for me. All the GPON SFP modules I've tested I had to set this one way or another to get O5 and the PPPoE connection to work.

Anime4000, 2023-05-10:
> 2.5Gbps is supported with command (fw_setenv sgmii_mode 5)

I tried that on my RB5009, didn't work 😆 revert to 1Gbps and stop working

ToTheCLI, 2023-05-10:
> > 2.5Gbps is supported with command (fw_setenv sgmii_mode 5)
> 
> I tried that on my RB5009, didn't work 😆 revert to 1Gbps and stop working

Disable auto-neg and set 2.5Gbps on RB5009 it is working for me

ToTheCLI, 2023-05-10:
> @ToTheCLI
>
> OMCI_FAKE_OK / OMCI_OLT_MODE / PON_VENDOR_ID / GPON_PLOAM_PASSWD / GPON_SN / LAN_SDS_MODE
>
> Those are the ones I had to change to get it to work. That is kinda general for me. All the GPON [...]

Can you check if it works without OMCI_FAKE_OK because there is no equivalent in Lantiq SFPs.

galvesribeiro, 2023-05-10:
Yeah, to "work" I had to disable it as well and have the  LAN_SDS_MODE 4. But again, I'm not sure how stable it is. It is working now, but not sure until when :)

[image]

ToTheCLI, 2023-05-10:
> @ToTheCLI
>
> OMCI_FAKE_OK / OMCI_OLT_MODE / PON_VENDOR_ID / GPON_PLOAM_PASSWD / GPON_SN / LAN_SDS_MODE
>
> Those are the ones I had to change to get it to work. That is kinda general for me. All the GPON [...]

flash set OMCI_FAKE_OK 0
reboot

Does it still work?

galvesribeiro, 2023-05-10:
```
# diag gpon get onu-state
gpon get onu-state 
ONU state: Operation State(O5) 
RTK.0> command:# diag gpon get onu-state
gpon get onu-state 
ONU state: Operation State(O5) 
RTK.0> command:# flash get LAN_SDS_MODE
LAN_SDS_MODE=4
# flash get OMCI_FAKE_OK
OMCI_FAKE_OK=0
```

Although it return O5, the PPPoE connection doesn't work.

ToTheCLI, 2023-05-10:
When OMCI_FAKE_OK 0
put
omcicli mib get 84
omcicli mib get 171

ToTheCLI, 2023-05-10:
Post output with OMCI_FAKE_OK 0 and OMCI_FAKE_OK 1

ToTheCLI, 2023-05-10:
> With OMCI_FAKE_OF 1:
> 
> ```shell
> # omcicli mib get 84
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> VlanTagFilterData
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> =================================
> EntityID: 0x03
> FilterTbl[0]: PRI 0,CFI 0, VID 10
> FwdOp:  0x10
> NumOfEntries: 1
> =================================
> # omcicli mib get 171
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> ExtVlanTagOperCfgData
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> =================================
> EntityId: 0x01
> AssociationType: 2
> ReceivedFrameVlanTagOperTableMaxSize: 0
> InputTPID: 0x8100
> OutputTPID: 0x8100
> DsMode: 0
> ReceivedFrameVlanTaggingOperTable
> INDEX 0
> Filter Outer   : PRI 15,VID 4096, TPID 0
> Filter Inner   : PRI 8,VID 10, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
> Treatment Inner   : PRI 8,VID 10, TPID 0
> INDEX 1
> Filter Outer   : PRI 14,VID 4096, TPID 0
> Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
> Treatment Inner   : PRI 15,VID 0, TPID 0
> INDEX 2
> Filter Outer   : PRI 15,VID 4096, TPID 0
> Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
> Treatment Inner   : PRI 15,VID 0, TPID 0
> AssociatedMePoint: 0x101
> DscpToPbitMapping:
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
> =================================
> =================================
> EntityId: 0x02
> AssociationType: 10
> ReceivedFrameVlanTagOperTableMaxSize: 0
> InputTPID: 0x8100
> OutputTPID: 0x8100
> DsMode: 0
> ReceivedFrameVlanTaggingOperTable
> INDEX 0
> Filter Outer   : PRI 14,VID 4096, TPID 0
> Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
> Treatment Inner   : PRI 15,VID 0, TPID 0
> INDEX 1
> Filter Outer   : PRI 15,VID 4096, TPID 0
> Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
> Treatment Inner   : PRI 15,VID 0, TPID 0
> AssociatedMePoint: 0x601
> DscpToPbitMapping:
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
> =================================
> ```
> 
> With OMCI_FAKE_OF 0:
> 
> ```shell
> # omcicli mib get 84
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> VlanTagFilterData
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> =================================
> EntityID: 0x03
> FilterTbl[0]: PRI 0,CFI 0, VID 10
> FwdOp:  0x10
> NumOfEntries: 1
> =================================
> # omcicli mib get 171
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> ExtVlanTagOperCfgData
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> =================================
> EntityId: 0x01
> AssociationType: 2
> ReceivedFrameVlanTagOperTableMaxSize: 0
> InputTPID: 0x8100
> OutputTPID: 0x8100
> DsMode: 0
> ReceivedFrameVlanTaggingOperTable
> INDEX 0
> Filter Outer   : PRI 15,VID 4096, TPID 0
> Filter Inner   : PRI 8,VID 10, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
> Treatment Inner   : PRI 8,VID 10, TPID 0
> INDEX 1
> Filter Outer   : PRI 14,VID 4096, TPID 0
> Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
> Treatment Inner   : PRI 15,VID 0, TPID 0
> INDEX 2
> Filter Outer   : PRI 15,VID 4096, TPID 0
> Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
> Treatment Inner   : PRI 15,VID 0, TPID 0
> AssociatedMePoint: 0x101
> DscpToPbitMapping:
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
> =================================
> =================================
> EntityId: 0x02
> AssociationType: 10
> ReceivedFrameVlanTagOperTableMaxSize: 0
> InputTPID: 0x8100
> OutputTPID: 0x8100
> DsMode: 0
> ReceivedFrameVlanTaggingOperTable
> INDEX 0
> Filter Outer   : PRI 14,VID 4096, TPID 0
> Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
> Treatment Inner   : PRI 15,VID 0, TPID 0
> INDEX 1
> Filter Outer   : PRI 15,VID 4096, TPID 0
> Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
> Treatment Inner   : PRI 15,VID 0, TPID 0
> AssociatedMePoint: 0x601
> DscpToPbitMapping:
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
>         0x000000
> =================================
> ```

Did you reboot after changing from 1 to 0?
Because if the VLANs are there it should work (Not Fake O5)

galvesribeiro, 2023-05-10:
You are right. I assumed it wasn't working too fast. If I leave it for a bit it will eventually connect. So yeah, `OMCI_FAKE_OK` doesn't seems to be a requirement.

galvesribeiro, 2023-05-10:
@ToTheCLI so, do you think getting the `FS.com GPON-ONU-34-20BI` would be a better/more stable option with 2.5G for my use case?

ToTheCLI, 2023-05-10:
I have an MA5671A with right.cn custom firmware and it has been working with me for months with Huawei OLT but YMMV.

ToTheCLI, 2023-05-10:
@galvesribeiro 
Have a couple of spare ones if you would like me to send you one on the condition that you return on your own expense or compensate me.

galvesribeiro, 2023-05-12:
@ToTheCLI 

During the course of the day, it kept dropping the connection and I got the MIBs:

```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
```

It seems it is not comming. I'm going to re-enable the FAKE_OK.

galvesribeiro, 2023-05-17:
@ToTheCLI / @Anime4000 After a lot of try and error with multiple combinations of setups, one weird behavior I've found.

If I plug the ODI directly to the UDMP/UDW WAN port, the connection works 100% of the times.

If I plug it on a switch port in the CRS305-1G-4S+ switch, it _sometimes_ work, but eventually the connection is dropped and the VLANs doesn't come down to the MIBs regardless of the `FAKE_OK` and the `LAN_SDS_MODE` value.

The switch is configured with a bridge on ports sfp3 and sfp4 where one is with the ODI and other one a 10G SFP+ module with the fiber cable going to the UDMP/UDW WAN port.

Do you guys have any idea why it would behave differently on the switch?

galvesribeiro, 2023-05-23:
Ok, I finally found what the problem is. The 2.5G works perfectly fine. That switch works perfectly fine, and the laser was being powered just fine.

The problem was a misconfiguration in the Bridge on the switch. The port from which the ODI is connected needs to be tagged on VLAN 10 and untagged on VLAN 1. If I just leave it only on the tagged VLAN 10, the PPPoE connection wouldn't work. If I leave it untagged without the VLAN10, as soon as the local network sends traffic thru the fiber, the OLT shut us down with an alarm and we have to wait a bit to try again.

After the proper config on the CRS305-1G-4S+ switch bridge, I've being using the ODI for a week with 0 disconnections. I've manually rebooted the switch and the ODI multiple times and it always reconnect perfectly fine.

TL;DR; you can use the CRS305-1G-4S+ with the ODI in 2.5G mode

ToTheCLI, 2023-05-23:
> untagged on VLAN 1

There is Only one VLAN in mib 84 that is VLAN 10 which you get PPPoE from, post the relevant config on mikrotik switch so we can understand what you mean by untagged on VLAN 1

galvesribeiro, 2023-05-23:
[image]

[image]

Look at the PVID on the `sfp4-WAN-In` which is the port where the ODI is connected.

galvesribeiro, 2023-05-23:
Another update, I just got my hands on another brand new FS.com GPON-ONU-34-20BI that just arrived. All I had todo was to set this:

```
sfp_i2c -i8 -s "MSTC365C75F1"
sfp_i2c -i11 -s "0010975712"
sfp_i2c -i7 -s "MSTC"
fw_setenv ipaddr 192.168.50.2
```

I didn't had to set port speed tho and since the switch port was already set to 2.5G it just worked.

```
root@SFP:/home/ONTUSER# onu ploamsg
errorcode=0 curr_state=5 previous_state=4 elapsed_msec=4294725788 
root@SFP:/home/ONTUSER# onu lanpsg 0
errorcode=0 pport=0 mode=15 enable=1 link_status=5 phy_duplex=1 
```

I plugged it on the same port on the switch and it worked first attempt. It also suffered with the same VLAN misconfiguration issue (I forced it just to make sure).

galvesribeiro, 2024-09-14:
@speedyb0y sim, sou Brasileiro, but lets keep the convo in English so everyone can follow up.

So I've never looked back and used the ODI again after the first one I had basically bricked. I've been using the FS.com `GPON-ONU-34-20BI` for an eye with 0 problems so far. I have two links (Vivo and Claro) and both are working perfectly fine.

They are connected to a Mikrotik Switch (CRS510-8XS-2Xq but any other should be fine) and then I have two other cables from that switch to the WAN ports on my Ubiquiti Router (EFG).

serhii-nakon, 2024-12-11:
Hello everyone did somebody get working 2.5Gbps mode on regular sfp+?

Anime4000, 2024-12-11:
> Hello everyone did somebody get working 2.5Gbps mode on regular sfp+?

Based on SFF, HiSGMII or 2.5G Base-X has no valid standard.

Thus, only few chip can support HiSGMII

serhii-nakon, 2024-12-11:
> > Hello everyone did somebody get working 2.5Gbps mode on regular sfp+?
> 
> Based on SFF, HiSGMII or 2.5G Base-X has no valid standard.
> 
> Thus, only few chip can support HiSGMII

@Anime4000 did it possible to switch to sfp+ regular not 2.5gb but 10gb

arnold0405, 2025-02-02:
i need help please help me extend the loid to atleast 50 characters because its too short to my pppoe username the odi dfp-34x-2c2 is only have 28characters... i want to edit please help... thank you very much

[image]

Anime4000, 2025-02-03:
> i need help please help me extend the loid to atleast 50 characters because its too short to my pppoe username the odi dfp-34x-2c2 is only have 28characters... i want to edit please help... thank you very much
> 
> [image]

I think you can do it via `flash set`