Anime4000/RTL960x #142: No phone vlan

timetragit, 2023-03-28:
Hi!
I've got a problem.
I found that my ISP does not send VOIP VLAN settings via OMCI when I use a modem without POTS ports like this pon stick. Thus, all VLANs come to the SFP stick, except for the voip one. I see 2 ways to solve the problem. The most correct, but difficult, is to teach the stick to transfer the presence of POTS ports as onu capabilities - then the OLT will configure the correct vlan via OMCI. But I have no idea, where I can do that. The second option is to manually set the settings for processing the GEM port. I found the way, how to create them:
RTK.0> gpon show ds-flow 
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     4095 |  ETH |         * |    
      1 |     3962 |  ETH |           |    
      2 |     3964 |  ETH |           |    
      3 |     3960 |  ETH |           |    
     64 |       23 | OMCI |           |    
===========================================================
RTK.0> gpon show us-flow 
============================================================
    GPON ONU MAC U/S Flow Status
Flow ID | GEM Port | Type | TCont
      0 |     3962 |  ETH |     0
      1 |     3964 |  ETH |     1
      2 |     3960 |  ETH |     1
     64 |       23 | OMCI |    16
============================================================
RTK.0> diag
RTK.0> gpon add us-flow flow-id 4 gem-port 3961 ether
RTK.0> gpon add ds-flow flow-id 4 gem-port 3961 ether
RTK.0> gpon show ds-flow 
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     4095 |  ETH |         * |    
      1 |     3962 |  ETH |           |    
      2 |     3964 |  ETH |           |    
      3 |     3960 |  ETH |           |    
      4 |     3961 |  ETH |           |    
     64 |       23 | OMCI |           |    
============================================================
RTK.0> gpon show us-flow 
============================================================
    GPON ONU MAC U/S Flow Status
Flow ID | GEM Port | Type | TCont
      0 |     3962 |  ETH |     0
      1 |     3964 |  ETH |     1
      2 |     3960 |  ETH |     1
      4 |     3961 |  ETH |     0
     64 |       23 | OMCI |    16
============================================================

Where 3961 is gem port with my VOIP vlan (vid 43)
However, there is still no any traffic with 43 vlan tag.

Here is some info from my **HG8245H5**, where vlan43 is present and can be bridged manually to lan port:
============================================================
+++++++++++++++++++++++ Gemport      +++++++++++++++++++++++
============================================================

GemportObj Num: 6
GemportEntry:

GemID  TcontID  PqUp  PqDn  GemType  Loopback  PirUp  CirUp  PbsUp  CbsUp  PirDn  CirDn  PbsDn  CbsDn
-----------------------------------------------------------------------------------------------------
4095   0        0     0     1        0         0      0      0      0      0      0      0      0
3963   0        32775 0     0        0         0      0      0      0      0      0      0      0
3961   0        32775 0     0        0         0      0      0      0      0      0      0      0
3962   1        32783 0     0        0         0      0      0      0      0      0      0      0
3964   2        32791 0     0        0         0      0      0      0      0      0      0      0
3960   2        32791 0     0        0         0      0      0      0      0      0      0      0

============================================================
+++++++++++++++++++++++ UniPort      +++++++++++++++++++++++
============================================================

UniPortObjNum: 5
UniPortEntry:

Index  PortID  BridgeID  UniPort  UniType
-----------------------------------------
0      1       1         1        0
1      2       1         2        0
2      3       1         3        0
3      4       1         4        0
4      5       1         255      1

******************************
********* Index:2   **********
******************************
PortID: 7
BridgeID: 1
ForwardOper: 16
VlanNum: 1
VlanEntry: 43,
MapGemID: 65535
UntagMapGemID: 3961
TagFrameOption: 0
PbitMap:

Pbit     GemID
--------------
0        3961
1        3961
2        3961
3        3961
4        3961
5        3961
6        3961
7        3961

MappingEntry:

Index  MappingID  BridgeID  GemID  MapMode  UniPort  Vlan  Pri  UniType  Status  VBridgeID  Pir(0x)  Sir(0x)
------------------------------------------------------------------------------------------------------
0      1          1         3961   1        255      43    255  0        1       2         00000000 00000000

Any ideas?

rajkosto, 2023-03-29:
I get voip VLAN created automatically as wan when i use HGU firmware(which is annoying), but i think my Olt doesn't check mib tree instead just base what to send on HwVer

timetragit, 2023-03-30:
Hi!
What is HGU firmware? Did you mean original Huawei ONU with HGU firmware, not rtl960 stick? I also tried to use huawei smartax ma5671a - same result, no voip vlan.
Yes, I tried to change mib, however, there was no success, because I had no enough samples/skills to do it correctly. 
About gems. Creating of gems may be useful, because gems are existing even if onu is not configured to use them.
For example, on my MA5671A I can see increasing packet counter after manually adding gem port:
```
GEM port
GEM PORT ID  GEM PORT IDX  TX FRAMES   TX BYTES  RX FRAMES   RX BYTES   
         23           255       2461     118128       2451     117648    // OMCI
       3960             3       4207    1472450          0          0    // VLAN 44
       3961             4          0          0        350      22464	 // VLAN 43 VOIP
       3962             1       4187    1465450          0          0    // VLAN 42 (MCC enabled)
       3964             2     547484  739421444     384087   56464525    // INTERNET (VLAN 40)
       4095             0          0          0    5549740  7580944840   // MULTICAST

```

DFP-34X-2c2 also displays downstream packets after adding gem:
```
# diag
RTK.0> gpon add us-flow flow-id 4 gem-port 3961 ether
RTK.0> gpon add ds-flow flow-id 4 gem-port 3961 ether
RTK.0> gpon show us-flow 
============================================================
    GPON ONU MAC U/S Flow Status
Flow ID | GEM Port | Type | TCont
      0 |     3962 |  ETH |     0
      1 |     3964 |  ETH |     1
      2 |     3960 |  ETH |     1
      4 |     3961 |  ETH |     0
     64 |       23 | OMCI |    16
============================================================
RTK.0> gpon show ds-flow 
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     4095 |  ETH |         * |    
      1 |     3962 |  ETH |           |    
      2 |     3964 |  ETH |           |    
      3 |     3960 |  ETH |           |    
      4 |     3961 |  ETH |           |    
     64 |       23 | OMCI |           |    
============================================================

RTK.0> gpon show counter flow 4
============================================================
     GPON ONU MAC Device Counter: D/S Flow 4 Gem
D/S GEM packets : 66
D/S GEM bytes   : 4256
============================================================
============================================================
     GPON ONU MAC Device Counter: D/S Flow 4 Eth
RX Eth packetts : 66
Fwd Eth packets : 66
============================================================
============================================================
     GPON ONU MAC Device Counter: U/S Flow 4 Gem
U/S GEM counts  : 0
U/S GEM bytes   : 0
============================================================
============================================================
     GPON ONU MAC Device Counter: U/S Flow 4 Eth
U/S Eth packets   : 0
============================================================
```

So, stream seems to be alive, and you rightly said, that bridge and mib must be also correctly configured to get traffic. I believe, that it can be done also manually, however, I didn't find out right way to do that.

rajkosto, 2023-03-30:
by HGU firmware i mean the HGU firmware available for rtl9601D in this repo... the ones that do not support 4 port emulation in the table (you must eraseall mtd3 after switching between SFU to HGU firmware on ODI stick or vice versa, just restore your MAC and MAC_KEY)
the HGU firmware supports more features like respecting mib 171 in some cases, it can also run its own pppoe/DHCP client/server/and NAT, but this is not recommended, luckily it can still bridge created WANs, either untagged (by checking port binding), or leaving it tagged (not checking port binding)

timetragit, 2023-03-30:
SFU/IGD firmwares? Yes, I tried them both at the beginning of my research - there was no voip vlan in both cases.
In the meantime, I found a command to add an entry to the MIB table, but I can't figure out the format of the input:
Usage: omcicli mib [cmd]

  create    : create MIB ME
    {classId, entityId, "value of all SBC attributes"}

Command will be like:
omcicli mib create 84 05 "???"

But any my values instead of "???" brokes table