Anime4000/RTL960x #144: CPGOS03-0490v2 firmware

BittorB, 2023-03-31:
about this module anybody has the firmware to crossflash it in another sfp module? is based to rtl9601c1

Anime4000, 2023-03-31:
what I have tried is, V2801F to TWCGPON657, it just works since using same SoC

BittorB, 2023-03-31:
im goint to try to crossflash to V2801F to test if it works better than the original firmware from this module. i made a backup from this

Anime4000, 2023-03-31:
yes, please make full dump backup, read SPI Flash directly

nguyenthanhhong-tg, 2023-04-02:
How to make full dump backup @Anime4000 ?

Anime4000, 2023-04-02:
you need SPI Flash reader, clip into SPI chip and save as bin file

Anime4000, 2023-04-04:
it useful when know partition location and length, but if STICK not able to boot, SPI Flash is only way

Delitants, 2023-04-06:
Hi, I have the same stick, it's quite problematic.
I was able to get it in O5 operational state by setting legit SLID (Ploam password) supplied by tech, but that's pretty much it, I can't obtain public IP over DHCP from ISP. My router is Mikrotik (it does not show any info from the stick but checkmark module present and link OK).
Original box from ISP is Nokia G-010G-A.
Any help??

nguyenthanhhong-tg, 2023-04-07:
Your original box is NOKIA, this means your OLT must be NOKIA or ALCATEL LUCENT. You got O5 and no interntet, this means OLT already blocked due to OMCI not matching during negotiation. With this case, I usually use GPON G-010S-A stick.

Anime4000, 2023-04-07:
@Delitants 
have you try set 1 at `OMCI_FAKE_OK`? this should reply `OK` when incompatible OMCI being negotiate.

as @nguyenthanhhong-tg said:
> OMCI not matching during negotiation

OLT is blocking or not establish the link due to OMCI mismatch, not stright ban the ONU, no no-link is set at OLT side

nguyenthanhhong-tg, 2023-04-07:
Just wonder @Anime4000, in case I set value of OMCI_FAKE_OK to 1. How to see OLT reply is OK?

Anime4000, 2023-04-07:
> Just wonder @Anime4000, in case I set value of OMCI_FAKE_OK to 1. How to see OLT reply OK?

Nokia Fake O5 will log link is established, but, Nokia OLT it self block you from connecting to ISP, this impossible to fix unless reverse engeringing Nokia

Delitants, 2023-04-07:
> > Just wonder @Anime4000, in case I set value of OMCI_FAKE_OK to 1. How to see OLT reply OK?
> 
> Nokia Fake O5 will log link is established, but, Nokia OLT it self block you from connecting to ISP, this impossible to fix unless reverse engineering Nokia

Why such a conclusion? I’m sure I’m just missing some values to be copied. There is lots of proofs on internet you can pretend nokia gpon with no reverse engineering at all.

Delitants, 2023-04-07:
> > Just wonder @Anime4000, in case I set value of OMCI_FAKE_OK to 1. How to see OLT reply OK?
> 
> Nokia Fake O5 will log link is established, but, Nokia OLT it self block you from connecting to ISP, this impossible to fix unless reverse engineering Nokia
OMCI_FAKE_OK to 1 makes no difference, no internet.
Tried to identify OLT and got no output from this command:
omcicli mib get 131

Delitants, 2023-04-12:
> When you got LINK OK on SFP, is SFP stable (keep UP) or UP and DOWN loop?

Stable

Delitants, 2023-04-14:
> Your original box is NOKIA, this means your OLT must be NOKIA or ALCATEL LUCENT. You got O5 and no interntet, this means OLT already blocked due to OMCI not matching during negotiation. With this case, I usually use GPON G-010S-A stick.

I just got GPON G-010S-A and I have the same problem, it reports status O5 and no DHCP working, no internet access.

nguyenthanhhong-tg, 2023-04-14:
Your connection type is DHCP, not PPPoE client, is it? Can you check if there is VLAN setting on old device? Some time VLAN is set by default in firmware from ISP and you will not see in original device.

nguyenthanhhong-tg, 2023-04-14:
> I just got GPON G-010S-A and I have the same problem, it reports status O5 and no DHCP working, no internet access.

Your connection type is DHCP, not PPPoE client, is it? Can you check if there is VLAN setting on old device? Sometimes VLAN is set by default in firmware from ISP and you will not see in original device.

nguyenthanhhong-tg, 2023-04-15:
> > > > Your original box is NOKIA, this means your OLT must be NOKIA or ALCATEL LUCENT. You got O5 and no interntet, this means OLT already blocked due to OMCI not matching during negotiation. With this case, I usually use GPON G-010S-A stick.
> > > 
> > > 
> > > I just got GPON G-010S-A and I have the same problem, it reports status O5 and no DHCP working, no internet access.
> > 
> > Your connection type is DHCP, not PPPoE client, is it? Can you check if there is VLAN setting on old device? Sometimes VLAN is set by default in firmware from ISP and you will not see in original device.
> 
> Yes it’s DHCP. How do I suppose to check that? Old device has no management page or SSH, it’s Nokia ONT G-010G-A. And if I connect old box to the router, I don’t set up any VLAN, dhcp works, so obviously no vlans there.

Please check VLAN on your NOKIA G-010S-A if it is available by command:

/opt/lantiq/bin/gtop -b -g 'GPE FID assignment' | sed '1,5d' | cut -f 2 -d ';'

[image]

Delitants, 2023-04-15:
> Please check VLAN on your NOKIA G-010S-A if it is available by command:
>
> /opt/lantiq/bin/gtop -b -g 'GPE FID assignment' | sed '1,5d' | cut -f 2 -d ';'
>
> [image]

Empty output

BittorB, 2023-06-08:
> know partition location and length, but if STICK not able to boot, SPI

for firmware you can check my github