Anime4000/RTL960x #145: ODI DFP-34X-2C2 & 2.5GbE Mode question

charlez79, 2023-04-02:
Hi,

Will the ODI DFP-34X-2C2 put itself automatically in 2.5GbE Mode (autodetect), or should it always be set manually?

br
Charlez

Anime4000, 2023-04-02:
what I experience with RB5009 with auto nego off, force 2.5Gb

ODI DFP-34X-2C2 will automatically nego with 2.5Gb without set `LAN_SDS_MODE`

also, set to 2.5Gb is also make ODI sync to RB5009 much faster!

dnnp362, 2023-04-03:
BTW I'm in 7.9rc1

ToTheCLI, 2023-04-03:
LAN SDS Mode 4 Only Auto-Neg Off Force 2.5Gbps

ToTheCLI, 2023-04-03:
Also try frimware 7.7 if that doesn't work.

ToTheCLI, 2023-04-03:
> BTW I'm in 7.9rc1

LAN SDS Mode 4 Only Auto-Neg Off Force 2.5Gbps

Wait for a 2 Minutes or 3

Anime4000, 2023-04-04:
I am on 7.8 (latest)
and I use Auto Neg on STICK, and Force 2.5Gb at Mikrotik

dnnp362, 2023-04-04:
It doesn't work for me in 7.9rc1

When you say you use Auto Neg in stick do you mean?
# flash set PON_MODE 3
# flash set FIBER_MODE  2

charlez79, 2023-04-05:
Was playing around and did put my DFP-34X-2C2 in LAN_SDS_MODE 0 and now it's not responding anymore. While in LAN_SDS_MODE 3 it was.
So LAN_SDS_MODE 0 can also make you stick unresponsive.
In my case it's put in a Fortigate firewall.