Anime4000/RTL960x #150: Same `GPON_SN` among the WAN network in your area may interrupt the WAN connection

ccy, 2023-04-20:
I have few `DFP-34X-2C2` GPON to connect to few internet connections.

Beside setting the PLOAM (ONU password) password for these GPONs, I perform these settings too:
```
flash set OMCI_TM_OPT 0
flash set OMCI_CUSTOM_RDP 4
flash set GPON_SN HWTC12345678
```
All GPONs using are using same `GPON_SN`.  When I connect a fiber into one of the GPON, I notice other GPON connections will drop.  After some i trial and error, I try using different `GPON_SN` for each GPON device and all connections work perfect without interrupt each other.

Just curious if you have similar experience?  I suspect if we try using other `GPON_SN` may cause network disruption in your area.

Anime4000, 2023-04-21:
I can't try using same SLID/GPON Password, since I don't know my neighbor password.

if anyone here work with ISP, they can try

rajkosto, 2023-04-23:
Of course the same SN on multiple ONU connected to the same OLT port will make the other one drop connection, that's now they are "uniquely" identified !