Anime4000/RTL960x #153: Instructions for when configuring from NIC

User-McUserface, 2023-04-25:
I followed the instructions and got the XPON stick, but I don't have a router with SFP slot, only a NIC, which is plugged in a Windows PC.

I can't figure out how to get to the XPON config web interface when it's plugged into the PC.

If there was a small mention about that in the instructions, that would really help.

Thanks.

jjeziorny, 2023-04-25:
It's the same as using a router, you need to configure a static IP on the NIC which is in the same range as the SFP then you can access the SFP web interface from the computer

Anime4000, 2023-04-26:
You need buy SFP Media Converter that convert SFP to Standard Ethernet

User-McUserface, 2023-04-30:
> You need buy SFP Media Converter that convert SFP to Standard Ethernet

Ok, but just to clarify, theoretically, after the stick is configured through the media converter, will I be able to plug it into the NIC to continue the operation there?

What I'm trying to figure out is, whether I need to buy the media converter necessarily, or if I can just borrow one to perform the configuration.

Anime4000, 2023-04-30:
if your PC has SFP Slot, you just config there via 192.168.1.1 WebGUI

use for router without SFP, you need an router that has SFP,

or... media converter to router via RJ45

I would suggest to buy Router with SFP like RB5009

physx2494, 2023-05-02:
Can you please post your initial configuration to get the webui using Mikrotik?

ToTheCLI, 2023-05-03:
> Can you please post your initial configuration to get the webui using Mikrotik?

https://github.com/Anime4000/RTL960x/issues/54