Anime4000/RTL960x #154: ODI xPON stick which firmware to update

xbipin, 2023-05-05:
I bought the ODI xPON stick from aliexpress, connected it to mikrotik Hex S and configured it and internet works, cpuinfo shows RTL8672 chipset and it came with V1.0-220923 firmware, what I'm confused about is what firmware do I need to install or do I just use the firmware it came with, I want vlan mapping to work and the default firmware the temperature is a bit too high, almost 62 degrees which I want to reduce

Anime4000, 2023-05-05:
PON Stick are known to very hot when run, most all SFP are like that, it should survive at 90 degree

If current firmware works, it's better leave it as is, let router do the VLAN.
Doing PVID on ODI Stick are quite complicated, unlike those Huawei MA5671a Stick

xbipin, 2023-05-06:
secondly mikrotik doesnt report sfp temperature

xbipin, 2023-05-06:
i was given the nokia router by my isp, where can i get the PLOAM password from, the web gui doesnt seem to show it anywhere, currently using DEFAULT012