Anime4000/RTL960x #155: TWCGPON657

Forge411, 2023-05-08:
Hi ! Please tell me ! After flashing TWCGPON657 to C00R657V2801F_V1.9.0-220404.tar, the stick stopped turning on! What could I have done wrong ? And how can this be fixed ?

Anime4000, 2023-05-09:
did you follow this steps before flashing it?
https://github.com/Anime4000/RTL960x/tree/main/Firmware/TWCGPON657#using-v2801f-on-twcgpon657

Forge411, 2023-05-09:
Yes of course! Before the firmware, I did this! 

Before flashing the V2801F firmware, you need to do it via telnet first

installing LAN_SDS_MODE 1 flash
installing the flash PON_MODE 3

The stick is stitched! It rebooted and didn't turn on anymore! I couldn't do any more according to your instructions!

Forge411, 2023-05-09:
I couldn 't do these items anymore ! 

Once the firmware update is complete, do this to prevent a reboot with the wrong key.:

ELAN_MAC_ADDR 6CEFC6000000 Flash Memory Set

HW_HWVER RTL960X flash Memory Kit VS_AUTH_KEY 00CF646955CCBDB88AB3B68922DB810F flash Memory Kit
Set PON_MODE back to 1 GPON or 2 EPON

Anime4000, 2023-05-10:
unfortunate, way to fix this is via UART 😢

Forge411, 2023-05-10:
Will this do ?
[image]

Anime4000, 2023-05-10:
yes that should do just fine, you might just switch the boot partition back to original,
I suggest that you just buy ODI stick instead

Forge411, 2023-05-10:
The stick costs 85$ and the UART is 7$ 😂😂😂

Anime4000, 2023-05-10:
yea, it's good to have UART tool, you can login many embedded system

Forge411, 2023-05-10:
Don 't you have the original eeprom for the stick ?

Anime4000, 2023-05-10:
> Don 't you have the original eeprom for the stick ?

unfortunately I dont, my best guess it just boot normally and boa is crashed perhaps, you just swap boot partition and reflash again

Forge411, 2023-05-10:
When you don't understand it, it's very difficult 🫣

Anime4000, 2023-05-10:
well, that's the experience I been though since 2020, when world in lock down, much free time, I have broke many PON Stick just to make this guide 🤣

even blown the PON Laser 🤣

I recommend you get the ODI PON Stick instead, it has stable firmware between full bridge and HGU mode

I am sorry about your stick, I haven touch RTL9601CI long time after found RTL9601D Sticks

Forge411, 2023-05-11:
there is  .bin from b13 ?
[image]

Forge411, 2023-05-11:
I downloaded a file from eeproma . But there 's a format .bin . Perhaps you have firmware in this format ?

Forge411, 2023-07-15:
> 

Thanks ! Flashed the eeprom. The stick 's indicator started to glow ! It has not been possible to enter the settings panel yet! The standard ip for logging in is 192.168.1.1 ?

Forge411, 2023-07-15:
> > I downloaded a file from eeprom . But there 's a format .bin . You have firmware in this format ?
> 
> Maybe this will help, dump from working stick. [C00R657V2801F_V1.9.0-220404.bin](https://drive.google.com/file/d/1KBEZ3lQILjnizftc43nzuSfD2hfL8G1g/view?usp=drive_link)

Thanks ! Flashed the eeprom. The stick 's indicator started to glow ! It has not been possible to enter the settings panel yet! The standard ip for logging in is 192.168.1.1 ?

tantien11, 2023-07-15:
> Thanks ! Flashed the eeprom. The stick 's indicator started to glow ! It has not been possible to enter the settings panel yet! The standard ip for logging in is 192.168.1.1 ?

The ip for loggin is 192.168.255.2
telnet to stick with this account
user: user
pass: user

`flash all` to find admin password

Forge411, 2023-07-15:
> The ip for loggin is 192.168.255.2 telnet to stick with this account user: user pass: user
>
> `flash all` to find admin password

Thank you very much! 🙏 
Tomorrow I 'll try to log in to the stick by ip !

Forge411, 2023-07-16:
> The ip for loggin is 192.168.255.2 telnet to stick with this account user: user pass: user
>
> `flash all` to find admin password

[image]

The stick is working ! 
Thank you very much 🤪
 login: admin 
password: admin

Forge411, 2023-07-16:
> I try buy TWCGPON657 at taobao again and make full SPI Flash dump between original and V2801F based
> 
> but, my gut tell me, you can fix it via UART 😄
> 
> if micro solder is hard for you, ODI Stick is 😉

Maybe dump fail eeproma should be saved for other users ? There is also an IP login and password from dumpa.