Anime4000/RTL960x #159: Just bricked a DFP-34X-2C2 flashing a custom firmware

cpsoares, 2023-05-14:
Hi,

Tried to modify original firmware (tarball found here: https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M110_sfp_ODI_220923.tar) with this procedure documented here: https://github.com/Anime4000/RTL960x/blob/main/Docs/Modify_Firmware.md. Basically I injected the updated files needed for the "vlan fix" here: https://github.com/Anime4000/RTL960x/tree/main/Tools/etc. So, packed everything into a new tarball and updated firmware via web portal, after that, I could not access the stick anymore, ethernet interface of the stick does not come up. Is there a way to switch to the alternate image, which should be still good? Any other alternatives?

BTW, I did not use the firmware with that "fix" (found here: https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M110_sfp_ODI_220817.tar) because it has outdated scripts that are not so configurable and even can wear the flash chip over time.

In case the only way is to spi-reflash using an external programmer, I saw different information regarding flash layout which is confusing (here: https://github.com/Anime4000/RTL960x/issues/126 and here: https://hack-gpon.github.io/ont-odi-realtek-dfp-34x-2c2/).

What would be the easiest path to recover access to the stick without buying a new one?

PS: I don't have a SPI programmer but I have a 3.3V USB-TTL converter and I may try to access the UART, but I would like first to get your opinions if I can solve anything using that strategy.

Many thanks.

Anime4000, 2023-05-15:
the stick have two boot partition, r0, k0 and r1, k1

use UART and switch boot, reboot

you will boot another partition and reflash again with correct firmware

cpsoares, 2023-05-16:
Thanks @Anime4000! Just soldered tx and rx wires to the pads, ground to the metal cage and connected everything through the ttl to serial adapter, I presume baud rate is 115200, N81. All I receive is garbled chars, even press some keys at the beginning and nothing different, unless there is some magic key sequence to activate the bootloader options. I think I may have messed up with the bootloader. I will have access to an spi flash programmer soon, is there any chance someone gets me the 8Mbyte full image of the flash chip of a working and reset to defaults DFP-34X-2C2 stick? Many thanks.

Anime4000, 2023-05-17:
baud rate is correct and everything is default serial parameter. it should be have text output, I use putty to manage it

Voidstranger, 2023-06-27:
Hi. How can I disassemble it? I need to fix wrong LAN_SDS_MODE