Anime4000/RTL960x #162: VSOL V2801F configuring question

Voidstranger, 2023-05-29:
I apologize for writing here. It is not really an issue, just some noob questions.
I have pretty rare router BPI-R3. It has two SFP cages and I want to replace ISP's ONU. So I decided to buy VSOL V2801F. But I have some questions about configuring it.
1) How can I choose LAN_SDS_MODE after installing custom firmware? My router gives this ethtool info about sfp port:
[image]
So correct LAN_SDS_MODE parameter for 2,5 GbE is 6, right?
2) Can someone have tech specs of this module, especially about its power consumption? My current OpenWRT build supports 3W or less SFPs, but I think, I can raise it, since max current for 2 SFP ports on the router is 3A. And I don't plan to use second SFP for LAN (RJ-45)

Anime4000, 2023-05-29:
by default V2801F always force 1G, need a modified firmware that can set,
yes, LAN SDS 6 is 2500base-X
https://github.com/Anime4000/RTL960x/blob/main/Docs/2.5Gb.md

closed 2023-05-29 by Voidstranger

KYTECHNGAMING, 2025-05-06:
@Anime4000 Master,

I flashed your modded FW of [V2801F_V1.9.0-240614.tar](https://github.com/Anime4000/RTL960x/raw/main/Firmware/V2801F/V2801F_V1.9.0-240614.tar) into my Carlitoxx V2. 

Can I change LAN_SDS 1 to 6 and use with 2.5G converter or SFP ports? I dont wanna brick my sfp because of lan sds mode.

Anime4000, 2025-05-06:
use Media Converter that use Realtek chip inside,

however, SoC that VSOL use have partial 2.5G support, better get ODI instead

KYTECHNGAMING, 2025-05-06:
> use Media Converter that use Realtek chip inside,
> 
> however, SoC that VSOL use have partial 2.5G support, better get ODI instead

Yep, I have converters which are all have the RTL8221B chip. I just want to learn about compatibility.