Anime4000/RTL960x #167: ODI DFP-34G-2C2 - No OLT VendorID, reaches O5 by default but no connectivity

koge97, 2023-06-22:
Hello.

After hours of trying to set up my Stick, I give up.

My service provider is Movistar Chile, very similar to VIVO Fibra in Brazil or Telefonica Spain. They both use the Askey RTF3505VW and EdgeCore GG-11000 ONT models.

Originally, my ISP installed an RTF3505 and I replaced it with a used GG-11000 that I bought. I only had to change the PLOAM.

I'm using the data provided in the repository, but my PLOAM in ASCII format is only 7 digits. Shouldn't it normally be 10 digits?

I have cloned the GPON_ONU_MODEL, GPON_SN, PON_VENDOR_ID, HW_HWVER, and OMCI_SW_VER data, but I haven't had any results.

I have also tried adjusting the ELAN_MAC_ADDR of the original ONT.

Currently, no matter what value I enter in the specified parameters, the stick always reaches O5 state, but I don't have connectivity.

I have even factory reset the Stick (and by default, it also reaches O5 state).

If I try to view the OltVendorId, it always returns 0x00.

Currently, I am on firmware version 210702, the stick has Realtek chipset, tested on RB4011 and 5009.

Any help would be greatly appreciated; it's quite a challenge.

Anime4000, 2023-06-22:
this well known issue with Nokia OLT, same with my country Malaysia: when Nokia OLT found invalid Hardware Version, it give O5 without anything.

I called this Fake O5,

there something that Nokia OLT want to look at...

have you tried to set OLT MODE 21 ?

koge97, 2023-06-23:
After several attempts, I managed to establish connectivity with the module.
Indeed, some parameters were missing, so I tried cloning the data from another ONT I had on hand.

flash set GPON_PLOAM_PASSWD 1234567
flash set GPON_SN EDCR12345678
flash set HW_HWVER R1
flash set PON_VENDOR_ID EDCR
flash set OMCI_FAKE_OK 1
flash set GPON_ONU_MODEL EDGG11000
flash set OMCI_SW_VER1 1.2.8
flash set OMCI_SW_VER2 1.2.8
flash set OUI 7072CF
flash set HW_SERIAL_NO EDCR12345678
flash set OMCI_OLT_MODE 21
flash set OMCC_VER 160
flash set OMCI_TM_OPT 1
flash set HW_CWMP_PRODUCTCLASS SFU

[image]

closed 2023-06-23 by koge97