Anime4000/RTL960x #169: DFP-34X-2C2 disappeared after changing LAN_SDS_MODE

Voidstranger, 2023-06-27:
Hi. I changed LAN_SDS_MODE to 4 and can't reach sfp-module anymore.
I use BPI-R3 router with OpenWRT and other user confirmed, that he is using DFP-34X-2C2 with BPI-R3 router too and he hasn't any problems with 4, 5, 6 option of LAN_SDS_MODE. Using M110_sfp_ODI_210702 firmware.

ethtool eth1 (eth1 is port, where sfp-module is located) output:
[image]
It seems, that there is link on the interface.

No Rx to eth1. Why?
[image]

Also I changed sfp's ip address to 192.168.2.1. But this isn't a problem, right?

Anime4000, 2023-06-28:
the ODI Stick have auto negotiate which LAN Mode to use, perhaps your OpenWRT need the tweak first.

try revert back to 1G on OpenWRT if your new SFP are working...

if working, then try adjusting the OpenWRT router,

if is not, you need reset it back via TTL UART

Voidstranger, 2023-06-28:
Also, my stick is in epon mode. Can this cause an issue?

As for  UART.
I can see two little Rx and Tx spots near the chip on the stick. Ground is ground, I can connect it to metal cage, for example, right? But do I need to connect 3.3 cable? Where should I connect it?

Anime4000, 2023-06-28:
you can use `flying probe` if don't want do soldering, this much cleaner.
[image]

I suggest you do a factory reset, and try not change default IP Address, leave it as 192.168.1.1/24, this known issue most of the RTL based stick if not configured properly

closed 2023-06-28 by Voidstranger