Anime4000/RTL960x #171: 3mdeb Open Source Firmware vPub Conference

moriel5, 2023-07-04:
While I realize that the main subject of the conference is different from the subject here, there will be an open discussion, probably at 20:00 (8:00PM) UTC tomorrow, regarding the viability of open source firmware on G.PON ONTs, where they be SFP or otherwise, since I was asked to provide an idea for one of the discussions (this is not a lecture, and the is no dedicated host for this).

@Anime4000, I had mentioned your website, and since you have done more research into these modules, even though less regarding the open source part, I was wondering whether you would want to participate? If so, it will be online via Jitsi.

moriel5, 2023-07-04:
This is also open to anyone else who is interested in open source firmware development, and more details can be found on 3mdeb's website on the event's page (they are currently finishing up preparing the timetable, and will update the contents of the page soon, before the event actually starts): https://vpub.dasharo.com/e/7/dasharo-user-group-2

Disclaimer: I am unaffiliated with 3mdeb, however being an enthusiast who really likes what they are doing, am also trying to help out getting more people to know about these things.

Anime4000, 2023-07-06:
I would be nice if have Open Source GPON ONT especially in SFP module, easily adapt via Media Converter who want a Ethernet Port...

timing kind bad time for me to join 😅

moriel5, 2023-07-06:
Sorry about that, I just checked your timezone, and I can understand what you mean.

If you wish to join at the end (last haker standing), which will last however long people can stay up (so it could reasonably last a few hours), that may be manageable, as that would be in the morning for you, starting from 23:00 (11:00PM) UTC, or 7:00AM for you.

If so, I'll make sure to bring up the subject again so that you could add your insights.