Anime4000/RTL960x #173: Decrypt config file form ISP ONU

maidlinux01, 2023-07-13:
I would like to ask for some help.
My current ISP ONU is based on the same chip RTL960x so most of the config to make the SFP is very straight forward, however i found that as a good ISP Provided ONU, the "config.xml" is encrypted.
I was able to extract the web directory by using UART and i found this function in  "saveconf.asp"

```
function on_submit(obj)                                                                                                                                                                       
{                                                                                                                                                                                             
 obj.isclick = 1;                                                                                                                                                                             
 postTableEncrypt(document.saveCSConfig.postSecurityFlag, document.saveCSConfig);                                                                                                             
 return true;                                                                                                                                                                                 
}      
```
The postTableEncrypt function can be found here
https://gist.github.com/maidlinux01/23b09d79c6666b7781f9d3b86c8c90a8.js

Unfortunately  i dont have much experience dealing with this kind of code, so i would like to ask you for some help to create a decrypt python script, so it be easier to dump config from this ISP modem. 
ISP ONU:
PING Communication
7962v1

Regards

maidlinux01, 2023-07-14:
Looks like its ONU uses 9607c chip
```
# cat /proc/cpuinfo 
system type		: Formosa
machine			: rtk,9607c
processor		: 0
cpu model		: MIPS interAptiv (multi) V2.0
BogoMIPS		: 598.01
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 64
extra interrupt vector	: yes
hardware watchpoint	: no
isa			: mips1 mips32r2
ASEs implemented	: mips16 dsp mt
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VPE			: 0

```

And here is the disk layout

```
# df -P
Filesystem           1024-blocks    Used Available Capacity Mounted on
/dev/root                12032     12032         0 100% /
devtmpfs                 56068         0     56068   0% /dev
tmpfs                    56192         0     56192   0% /run
ubi0:ubi_Config           8140       348      7344   5% /var/config
# 
```

crrodriguez, 2023-07-15:
> I would like to ask for some help. My current ISP ONU is based on the same chip RTL960x so most of the config to make the SFP work is very straight forward, however i found that as a good ISP Provided ONU, the "config.xml" is encrypted. I was able to extract the web directory by using UART and i found this function in "saveconf.asp"
> 
> ```
> function on_submit(obj)                                                                                                                                                                       
> {                                                                                                                                                                                             
>  obj.isclick = 1;                                                                                                                                                                             
>  postTableEncrypt(document.saveCSConfig.postSecurityFlag, document.saveCSConfig);                                                                                                             
>  return true;                                                                                                                                                                                 
> }      
> ```
> 
> The postTableEncrypt function can be found here https://gist.github.com/maidlinux01/23b09d79c6666b7781f9d3b86c8c90a8
> 
> Unfortunately i dont have much experience dealing with this kind of code, so i would like to ask you for some help to create a decrypt python script, so it be easier to dump config from this ISP modem. ISP ONU: PING Communication 7962v1
> 
> Regards

there is no cryptography in there, it is obfuscated and should be possible to reverse.