Anime4000/RTL960x #175: Strange behavior causing slow speed on wireless when tested on 10 G servers

hugleo, 2023-07-21:
The stick is bridged to a wireless router.

Scenario 1
Speedtest server interface: 1 Gb
In tests on the cable, we get almost 1 Gb.
In tests on the wireless, we get almost 1 Gb.

Scenario 2
Speedtest server interface: 10G
In tests on the cable, we get almost 1 Gb.
In tests on the wireless, we get a slow speed of about 600 Mb.

A little weird is that the slow speed only happens if tested on a wireless device with the speedtest server using a 10 G interface. In theory, it should be better since Speedtest has more capacity.
I do not know if it is something with the OLT or some configuration we can do on the stick.

Anime4000, 2023-07-21:
this issue looks similar to RB5009 from SFP+ to 2.5G Ethernet

hugleo, 2023-07-25:
I've been testing it on other devices, mikrotik ccr, ubiquity routers and switches, same behavior. The only one that worked was a big switch Juniper EX. I think it could be something with buffers or flow control on the stick but I don't know how I can try to change these parameters if it's even possible.