Anime4000/RTL960x #176: V2801F & untagged vlans

byte123, 2023-07-24:
My ISP is using untagged IP traffic from the OLT toward the ONUs.
When I transmit a (broadcast) packet, the OLT can see my MAC address, but I do not receive any vlan untagged traffic.
VLAN transparent mode does not work.

When the ISP changes the configuration with vlan tagging toward ONUs, everything works fine.
Is V2801F capable to work with untagged WAN traffic ?

Anime4000, 2023-07-25:
maybe not, try check omcicli command if there any Q-in-Q

Anime4000, 2023-07-25:
have you checked EntityId ? is existed

V2801F, ODI SFU firmware doing EntityId hack to map all Entity into single interface, I guess it collided with others entity

Anime4000, 2023-07-25:
oh I forgot that, V2801F omcicli didn't work on telnet/ssh, maybe try buy ODI (china) or Leox if you in Europe