Anime4000/RTL960x #178: Internet Speed Very Slow When Using Nokia G-010S-A

Tuluminati, 2023-07-28:
I have 2 ISP's and tried both with the same Nokia G-010S-A and same other hardware. With the modems provided by the ISP I get a speed of 500MB/s with both. With 1 ISP (Telmex Mexico) in combination with the G-010S-A I get the same or a slightly higher speed, but with the other (TotalPlay Mexico) my speed is consistently maximum 50MB/s.

- ISP provided modem: FiberHome HG6145F
- Verification method used by ISP: LOID, LOIDPW, PLOAM, SerialNumber (and possibly MAC-Address), not PPPoE credentials.
- Connection is stable (but slow). First the speed was only 20 MB/s but after changing the mode to "2.5 Gbps / HSGMII with auto-neg on" the speed improved to 50MB/s

- I used the custom firmware uploaded here to be able to set LOID and LOIDPW: https://github.com/hwti/G-010S-A/issues/17#issuecomment-1073027251

No variables were changed other then the ones needed for succesfull verification.

Are there any variables I didnt set that wouldn't effect succesfull authentication but would have such a drastic impact on speed?

Anime4000, 2023-07-28:
I not really sure about that GPON ONU SFP since it's not using Realtek.

firmware is stock or Chinese OpenWRT?

Tuluminati, 2023-07-28:
Stock firmware.

Anime4000, 2023-07-30:
The Nokia G-010S-A is not using Realtek SoC like ODI, Nokia G-010S-A is using Intel Lantiq SoC which is very different

tantien11, 2023-07-30:
`[G-010S-A.bin](https://drive.google.com/file/d/1hadkBpTAlN7IPFz1A12gB4nv4mJDM-ZN/view?usp=drivesdk) `

tantien11, 2023-07-30:
[g010sa.bin](https://drive.google.com/file/d/1hadkBpTAlN7IPFz1A12gB4nv4mJDM-ZN/view?usp=drivesdk)

mshrem, 2023-09-23:
@Tuluminati can you take a look at [this ](https://github.com/Anime4000/RTL960x/issues/197)? 

I am also trying to set it up with telmex

PnoT, 2023-11-29:
@Tuluminati have you had any luck as a friend of mine is trying to bypass his HG6145F on TotalPlay.  We're running into issues getting into the superadmin to find out what values are required.  We're using the DFP-34X-2C2.