Anime4000/RTL960x #181: Update firmware V2801F ONU Stick

rabbanigithub, 2023-08-05:
My current firmware is very old. There is not option to upload firmware using WebGUI. Only http and telnet is working.
Is there any alternative way to upgrade firmware, or transfer firmware.tar file into the ONU Stick?

Model: V2801F
current firmware: V1.9.0-191015

vuducdong, 2023-08-05:
> My current firmware is very old. There is not option to upload firmware using WebGUI. Only http and telnet is working.
> 
> Is there any alternative way to upgrade firmware, or transfer firmware.tar file into the ONU Stick?
> 
> 
> 
> Model: V2801F
> 
> current firmware: V1.9.0-191015

You can try with:

192.168.1.1/upgrade.asp

rabbanigithub, 2023-08-05:
> You can try with:
>
> 192.168.1.1/upgrade.asp

Thanks, it worked.