Anime4000/RTL960x #184: Getting into Huawei HG8040H

shrmnk, 2023-08-17:
Hi folks, I am having trouble getting into my Huawei HG8040H - the sticker has no login credentials or IP addresses, just 3 lines/barcodes of Mac, SN and Prod ID.

I have attempted to clone Mac, generate related MACKEY and set GPON SN as HWTC+last 8 hex characters of sticker SN, but PON status stays between O2 and O3.

provider is Starhub in Singapore, hoping someone can give some insights into obtaining PLOAM password and other details from the Huawei ONT.

ToTheCLI, 2023-08-17:
> I have already tried plugging into ONT port and settting static IP to access 192.168.1.1 and 192.168.100.1 - both did not work, no pings, no http response.

Try while fiber is disconnected

Anime4000, 2023-08-17:
it's best to login stock ONU is via Disconnected Fiber and Cold Boot.
So, OMCI wont override any ISP specific that prevent you from login

shrmnk, 2023-08-18:
I have done that too, cold boot and fiber disconnected. Tried all ports 1 to 4 and both 192.168.1.1, 192.168.100.1, 192.168.1.254, didn't work.

Is it possible that the ONU was locked down by ISP that survives reboots? Anyway I could possibly do a hard reset then?

Anime4000, 2023-08-18:
@shrmnk I think only way to get inside is via UART TTL,

have you tried telnet or ssh it? maybe your ONU has disabled webGUI

shrmnk, 2023-08-19:
> @shrmnk I think only way to get inside is via UART TTL,
> 
> have you tried telnet or ssh it? maybe your ONU has disabled webGUI

I did try telnet when I was on those addresses (in addition to simply ping-ing them), but to no avail.

I'm not familiar with UART TTL - do you have a guide handy to setting this up with a similar ONU?