Anime4000/RTL960x #187: VLAN Setup on DFP-34X-2C2

Tuluminati, 2023-08-22:
Trying to replace my Fiberhome modem I first used a Nokia G-010S-A, I got this working but only at low speed. Now I'm trying the DFP-34X-2C2. I finally have O5 with hardware version 220923 and these settings:

<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="adsl"/>
  <Value Name="SUPER_PASSWORD" Value="realtek"/>
  <Value Name="TEST_MAC_ADDR" Value="realtek"/>
  <Value Name="ELAN_MAC_ADDR" Value="685811c75568"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="MAC_KEY" Value="669a571d27f72f2214b41389c38fc7b4"/>
  <Value Name="RESTORE_DEFAULT_EN" Value="1"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="LASER_POLARITY_TYPE" Value="0"/>
  <Value Name="PON_DETECT_ENABLE" Value="1"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="FHTT9BC75568"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="HW_WLAN0_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN1_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN0_COUNTRYSTR" Value="US"/>
  <Value Name="HW_WLAN1_COUNTRYSTR" Value="US"/>
  <Value Name="HW_CWMP_MANUFACTURER" Value="FHTT"/>
  <Value Name="HW_CWMP_PRODUCTCLASS" Value="HGU"/>
  <Value Name="HW_HWVER" Value="WKE2.094.373A0"/>
  <Value Name="GPON_ONU_MODEL" Value="HGU"/>
  <Value Name="LAN_SDS_MODE" Value="1"/>
  <Value Name="PON_MODE" Value="1"/>
  <Value Name="PON_LED_SPEC" Value="0"/>
  <Value Name="OUI" Value="68,58,11"/>
  <Value Name="PORT_REMAPPING" Value="00,01,00,00"/>
  <Value Name="PON_VENDOR_ID" Value="FHTT"/>
  <Value Name="GPON_SN" Value="FHTT9BC75568"/>
  <Value Name="EPON_ONU_MODEL" Value="FHTT"/>
  <Value Name="EPON_EXTONU_MODEL" Value="HGU"/>
  <Value Name="EPON_SILENT_MODE" Value="1"/>
 </Dir>
</Config>
<Config Name="ROOT">
 <Dir Name="MIB_TABLE">
[... 277 lines cut ...]
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x601
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
=================================`

Can anyone spot a mistake in my VLAN settings?

Anime4000, 2023-08-24:
what type of Firmware you are using currently? IGD or SFU ?

Tuluminati, 2023-08-24:
SFU

Anime4000, 2023-08-25:
maybe you can try IGD firmware then can use VLAN mapping inside WebGUI