Anime4000/RTL960x #188: Tataplay fiber ODI

chinmaythosar, 2023-08-23:
I recently got an ODI and was able to clone all my settings from a Nokia G-2425G-A by telnet and ritool dump after its connected to the internet.

the Format:01
the MfrID:ALCL
the Factorycode:02
the HardwareVersion: XX
the ICS:01
the YPSerialNum:        XX
the CleiCode:0000000000
the Mnemonic:G-2425G-A
the ProgDate:220719
the MACAddress:XX:XX:
the DeviceIDPref:0000
the SWImage:005e
the OnuMode:0003
the G984Serial:xx
the HWConfiguration:0000000000000000
the PartNumber:XX
the OperatorID:ALCL
the SLID:00000000000000000000000000000000
the CountryID:i1
the GroupID:00000000

However I think I also have the fake O5 issue. Can someone help me out. I am using M110_sfp_ODI_220923.tar 

My ISP has put my device in 'bridge' mode and I can use port 1 connected to my modem for PPPoE without any VLAN on my router (so the ONU must be doing some kind of VLAN tagging). Without bridge mode my ISP can set it to router mode (but it's a hassle to contact the ISP and change things) where the router itself sets 4051 as the VLAN. But before I get to connecting the right VLAN I probably need to fix the fake O5. Do I need to use a different firmware?

Anime4000, 2023-08-24:
have you try to execute `omcicli mib get 84` to print all VLAN ?

chinmaythosar, 2023-08-24:
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

This is the output of it. I believe there is no real O5 based on what it's outputting. Status in GUI does show O5 though.

d5aqoep, 2023-09-07:
VLAN of Tataplay Fiber is 1502

chinmaythosar, 2023-09-08:
It's not a VLAN issue. This command 'omcicli mib get 84' shows all the VLANs exposed if your ONU has O5 stable state. The fact that it's not showing any output means either VLAN isn't there (highly unlikely there is no VLAN) or the fact that it has fake O5 (Nokia OLT issue). So I need to first get proper O5 before I can get to VLANs.

Strykar, 2023-12-05:
You've changed your MAC address, did you regenerate and update the `MAC_KEY`?