Anime4000/RTL960x #189: DFP-34X-2C2 becomes unresponsive after a few days

labuwx, 2023-08-24:
I have a DFP-34X-2C2 with firmware 220923. It gets progressively unresponsive after 5 days of uptime.
First the web interface started to respond slowly, then not at all. Soon after that it would not even respond to ping. Access to my ISP and to the internet is unhindered (bandwidth and latency wise) but I do not know if OMCI communication is affected.

Suspecting a memory leak, I made a light version of 220923 with no boa, igmpd, dhcpd, ssh, ... . I only had telnet, omci_app, pondetect, configd running. I was monitoring the memory usage and did not see any increase over the days. /proc/kmsg did not show anything.
Yet after 5 days of uptime it started again. First diag decided to take 10 seconds to start up, while other command line tools did work normally. Then I could not even connect with telnet, finally no response to pings.

Has anybody experienced this? Anybody from the opposite, still stable after months of uptime group? 

The stick is in a well ventilated rack, max 45C transceiver temperature.
The traffic is not extreme either. There is a constant ~150 Mb/s IPTV downstream that the ONU should forward. Besides that, the internet traffic is usually below 10 Mb/s with less than 100 Mb/s peaks.

ToTheCLI, 2023-08-25:
> It gets progressively unresponsive after 5 days of uptime.

Similar issues happened to me (Packet loss with pings), switched to Lantiq based GPON SFPs and never observed this issue.

What OLT do you have and what is the SFP plugged in to?

labuwx, 2023-08-27:
> What OLT do you have and what is the SFP plugged in to?

The OLT reports itself as HWTC (and probably really is). I use a gigabit switch (TL-SG2424) but the same happens when I bypass the switch and plug the ONU into my router via a gigabit media converter.

> Similar issues happened to me (Packet loss with pings), switched to Lantiq based GPON SFPs and never observed this issue.

Just to reiterate, the ONU did not stop forwarding anything between my ISP and my LAN, it's just the user interface (and other higher level services) that went down. However I am afraid that complete loss of connection would happen if I left the ONU running for long enough. Either because the lower layers of the RTL9601 would need some input from the locked up upper layers, or the ONU would stop responding to OMCI and my ISP would terminate my connection.

I used to have a UFiber NanoG (Broadcom chipset) but I could not get the IPTV multicast forwarded. Is it accessible with Lantiq ONUs?
My current setup with the DFP-34X-2C2 is not ideal either. I would like to use IGMP snooping on the ONU since the full IPTV downstream takes up a significant portion of the gigabit link to my switch but I suspect (have not confirmed) that would need igmpd, which dies with the rest of the services.

ToTheCLI, 2023-08-27:
[image]
Well there are settings for IGMP in Lantiq SFP's Chinese firmware, don't have IPTV to test with though.

ToTheCLI, 2023-08-27:
> The OLT reports itself as HWTC

Best for them is MA5671A or fs.com SFPs.

labuwx, 2023-09-12:
I think I have found the culprit. I had OMCI_LOGFILE and OMCI_LOGFILE_MASK set.
With the aforementioned parameters cleared, my ONU has been up and responsive for 20 days now.

closed 2023-09-12 by labuwx

rndm2, 2023-11-17:
I have exactly the same problem. But instead of days, I have hours, or even minutes.
And it causes random packet loss.

Here is the `top` from stick:
[image]

Config: 
`
# flash all
<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="adsl"/>
  <Value Name="SUPER_PASSWORD" Value="realtek"/>
  <Value Name="TEST_MAC_ADDR" Value="realtek"/>
  <Value Name="ELAN_MAC_ADDR" Value="00e04c867001"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="MAC_KEY" Value=""/>
  <Value Name="RESTORE_DEFAULT_EN" Value="1"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="LASER_POLARITY_TYPE" Value="0"/>
  <Value Name="PON_DETECT_ENABLE" Value="1"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="000000000002"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="HW_WLAN0_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN1_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN0_COUNTRYSTR" Value="US"/>
  <Value Name="HW_WLAN1_COUNTRYSTR" Value="US"/>
  <Value Name="HW_CWMP_MANUFACTURER" Value="MSTC"/>
  <Value Name="HW_CWMP_PRODUCTCLASS" Value="HGU"/>
  <Value Name="HW_HWVER" Value="GPT-2541GNAC"/>
  <Value Name="GPON_ONU_MODEL" Value="HGU"/>
  <Value Name="LAN_SDS_MODE" Value="1"/>
  <Value Name="PON_MODE" Value="1"/>
  <Value Name="PON_LED_SPEC" Value="0"/>
  <Value Name="OUI" Value="11,11,11"/>
  <Value Name="PORT_REMAPPING" Value="00,01,00,00"/>
  <Value Name="PON_VENDOR_ID" Value="MSTC"/>
  <Value Name="GPON_SN" Value="MSTC13DA1366"/>
[... 139 lines cut ...]
  <Value Name="VPI" Value="0"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> 
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="43"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> 
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="51"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> 
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="PPPOE_SESSION_TBL">
 </Dir>
 <Dir Name="RIP_TBL">
 </Dir>
 <Dir Name="DHCP_RESERVED_IPADDR_TBL">
 </Dir>
 <Dir Name="EPON_LLID_TBL">
 </Dir>
</Config>
`