Anime4000/RTL960x #191: Lu Lee GPON SFP stick and TP-Link BE900 router 2.5Gbps compatibility

d5aqoep, 2023-09-07:
I have Lu Lee GPON SFP stick which is rebrand of ODI with same realtek chipset. 

If I use the 2.5G media converter and connect it to TP-Link BE900's 10G WAN port to dial a PPPoE connection, I get 1100/1010 Mbps down/up on my Tata Play Fiber connection. 

However, if the GPON stick is in SFP cage of BE900, I am restricted to 940 Mbps which means the link speed is 1 gigabit only. I tried to set the WAN port speed to 2500 Mbps from BE900's web GUI but it remains at 1 Gbps.

What would be the ideal 2.5Gbe mode for this stick? 4, 5 or 6? I don't want to set to something wrong and make the stick inaccessible.

Please help.

Anime4000, 2023-09-07:
I think TP-Link BE900 can't support HiSGMII and or at least 2500Base-X

chinmaythosar, 2023-09-08:
Try and raise a request to TP-Link to see if their SFP+ port supports HiSGMII . Most SFP+ ports only support 10G and 1G but if it's possible they could enable it with a firmware update.

Anime4000, 2023-09-08:
if hardware it self supported SerDes RGMII, SGMII, HiSGMII

d5aqoep, 2023-11-14:
@Anime4000 

Qualcomm has enabled HiSGMII support on Asus' RT-AX89X SFP Port

Test firmware here
https://www.asuswebstorage.com/navigate/a/#/s/0CAB26175EF8440FBD6117E4076095924

That gives us hope that TP-Link would also be able to provide 2.5G SFP port capabilities in the future for BE800, BE900, BE95, BE85 models which are having Qualcomm SOC platform.

Anime4000, 2023-11-14:
@d5aqoep Nice! I link this to 2.5G compatibility list