Anime4000/RTL960x #193: Any specific settings for multicast IPTV?

Eyeless77, 2023-09-14:
I have DFP-34X-2C2 SFP module. My ISP provides me at least two VLANs: one for the Internet (VLAN 30) and the second for IPTV (VLAN 20). SFP module is connected into L2 managed switch (port 9) and operates in "Manual Transparent Mode" in terms of VLAN settings of SFP module.

[image]

The switch is set up with two additional VLANs (20 and 30). The uplink SFP port 9 is set up as tagged port for both VLAN 20 and 30. Also I have a downstream router which is connected to port 8 of the switch and this port is set up as tagged port for VLAN 30 (Internet). I have a subinterface on a router for Internet traffic and it is working fine.

[image]

My goal is to have ISP provided IPTV box working in a separate switch port (port 1 for example). In order to achieve this I set up port 1 as untagged port for VLAN 20 and assign PVID 20 for port 1.

[image]

After connecting IPTV box into port 1 it successfully gets an IP address from ISP IPTV subnet and gets access to on-demand services such as movies, tv shows and so on. The only issue occurs with multicast IPTV, it shows just a black screen. I played around with IGMP snooping settings on a switch (turned on and off globally, enabled for specific ports), it detects IGMP queries, but there is no picture on a screen.

[image]

In my opinion that should work like a charm, but there is an issue somewhere. I tried to use different firmwares for SFP stick (M114_sfp_ODI_hybrid_221209, M110_sfp_ODI_220304, M114_sfp_ODI_Vlan_220414), I tried Auto mode in VLAN settings, but that doesn't help. Maybe I'm missing something and there are some hidden settings for multicast or IGMP? The counters for multicast packets increase and that frustrates me.

[image]

Eyeless77, 2023-09-14:
Hope this diag info could help.

````
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 10
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x04
FilterTbl[0]: PRI 0,CFI 0, VID 20
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x05
FilterTbl[0]: PRI 0,CFI 0, VID 30
FilterTbl[2]: PRI 0,CFI 0, VID 32
FwdOp:  0x10
NumOfEntries: 2
=================================
=================================
EntityID: 0x06
FilterTbl[0]: PRI 0,CFI 0, VID 40
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x07
FilterTbl[0]: PRI 0,CFI 0, VID 50
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x08
FilterTbl[0]: PRI 0,CFI 0, VID 60
FwdOp:  0x10
NumOfEntries: 1
=================================
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x01
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 4096, TPID 0, RemoveTags 0
Treatment Inner   : PRI 0,VID 30, TPID 4
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 60, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 60, TPID 0
INDEX 2
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 32, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 32, TPID 0
INDEX 3
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 30, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 30, TPID 0
INDEX 4
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 5
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x101
DscpToPbitMapping:
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
=================================
=================================
EntityId: 0x02
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 4096, TPID 0, RemoveTags 0
Treatment Inner   : PRI 5,VID 10, TPID 4
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 10, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 10, TPID 0
INDEX 2
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 3
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x601
DscpToPbitMapping:
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
	0x000000
=================================

labuwx, 2023-09-15:
Hi,

This is how I have managed to make IPTV work. I use the M110_sfp_ODI_220923 firmware. The OLT is Huawei, the IPTV system is Mediaroom.

IPTV multicast packets use a different GEM port than the rest of the IPTV traffic. I have GEM ports for Internet, VoIP, TR-069, bidirectional IPTV and downstream IPTV multicast.
```
# diag gpon show ds-flow
gpon show ds-flow 
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     4095 |  ETH |         * |    
      1 |     1124 |  ETH |           |   *
      2 |     1121 |  ETH |           |   *
      3 |     1122 |  ETH |           |   *
      4 |     1123 |  ETH |           |   *
     64 |       12 | OMCI |           |    
============================================================
```
First, you should make sure that the ONU forward the multicast stream. I suggest you turn off IGMP snooping on the ONU and connect it directly to a PC to check the forwarded packets with Wireshark. Do you see a substantial amount (~150 Mb/s in my case) of multicast packets arriving? If not, you should tweak the settings. The combination I use is `VLAN_CFG_TYPE=1`, `DIRECT_BRIDGE_MODE=1`, `DUAL_MGMT_MODE=1`. If I change any of these, the multicast packets stop coming.

You should also check if the multicast packets have the same vlan tag as the rest of the iptv traffic. If so, you may skip ahead. If the multicast packets arrive without a tag, you may try `VLAN_MANU_MODE=1`, `VLAN_MANU_TAG_VID=<a random vid that is not used anywhere else>`.
If the multicast packets have a different tag (let's use VLAN900 from now on), you have to merge them with VLAN20. I do not yet know how to do that with the ONU. Possible solutions: a switch that can rewrite incoming VLAN tags; a switch with Multicast VLAN Registration support.
Third option is to untag both VLAN20 and VLAN900 on port 1, and set its PVID to 20. A problem with this setup was that my IPTV STB has only a 100 Mb/s port that cannot handle the full unfiltered multicast downstream. I could not do IGMP snooping since VLAN900 was to be filtered but IGMP queries and reports existed on VLAN20. What I did was to use a loopback cable and create a VLAN for the merged traffic that I could successfully filter.

However that was not enough for IPTV to work. My Tatung STB ignores IGMP queries unless they come from the default gateway so connecting it directly to my ISP's IPTV vlan is not an option. I had to put it into a separate subnet and run IGMP proxy on my router.

If anybody knows how to do the IGMP snooping on the ONU, I would love to hear that. I have a 1G/1G internet subscription that I cannot fully utilize because I only have a gigabit switch and the full IPTV broadcast reaches 200 Mb/s during peak hours.

Eyeless77, 2023-09-15:
Hi @labuwx, thank you for sharing your experience.

I have the following output for `gpon show ds-flow`:

```
gpon show ds-flow
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     4095 |  ETH |         * |
      1 |      260 |  ETH |           |
      2 |      388 |  ETH |           |
      3 |      516 |  ETH |           |
      4 |      644 |  ETH |           |
      5 |      772 |  ETH |           |
      6 |      900 |  ETH |           |
     64 |        4 | OMCI |           |
============================================================
```

and `gpon show us-flow`:

```
gpon show us-flow
============================================================
    GPON ONU MAC U/S Flow Status
Flow ID | GEM Port | Type | TCont
      0 |      260 |  ETH |     0
      1 |      388 |  ETH |     1
      2 |      516 |  ETH |     2
      3 |      644 |  ETH |     2
      4 |      772 |  ETH |     2
      5 |      900 |  ETH |     2
     64 |        4 | OMCI |    16
============================================================
```

I'll try using M110_sfp_ODI_220923 firmware with your config envs.

vuducdong, 2023-09-16:
Hi @labuwx 

I have almost similar problem.

The OLT is Huawei, the IPTV system is ZTE.

I have GEM ports for Internet (vlan11), TR-069 (vlan4000), IPTV (vlan12) and multicast (vlan9).

When I use Hisense sfp everything works perfectly.

When I use sfp ODI with M110_sfp_ODI_220923 firmware (internet ok_vlan11) but whenever I use IPTV the internet is lost (seems sfp freezes) so I need to unplug sfp and then reload it. I also tried with many different firmware versions (and still the same error).

This is a photo taken when using sfp Hisense

[image]

[image]

And this is from sfp ODI
[image]

Eyeless77, 2023-09-18:
Got it working via dedicated switch port. The main issue was the fact that multicast traffic is passing through a different VLAN 9. And VLAN 20 is used only for STB registration and on-demand services.

- My firmware is M110_sfp_ODI_220923, `VLAN_CFG_TYPE=1`, `DIRECT_BRIDGE_MODE=1`, `DUAL_MGMT_MODE=1`;
- IGMP Snooping on the switch is OFF;
- Port 9 (SFP ONU stick) on the switch is configured as tagged for VLANs 9, 20, 30;
- Port 1 (STB) is configured as untagged for VLANs 9, 20;
- Port 1 (STB) is configured as PVID 20.

This is applicable for ISPs MTS and MGTS (Moscow region).
The next step for me is to get it working via router and igmp-proxy.

skon77, 2023-09-18:
@Eyeless77 

> This is applicable for ISPs MTS and MGTS (Moscow region).

vlan 3 - Internet (for ZTE OLT)
vlan 5 - SIP (for ZTE OLT)
vlan 8 - Multicast and IPTV subscription management (for ZTE OLT)
vlan 9 - IPTV multicast
vlan 10 - SIP (for Huawei OLT)
vlan 20 - IPTV set-top box authorization server (for Huawei OLT)
vlan 30 - Internet (for Huawei OLT)
Multicast subnet 5 - encrypted by Verimatrix, used by Ericsson IPTV set-top boxes
Multicast subnet 4 - encrypted by Securmedia, used by Smartlabs IPTV set-top boxes.
 
Huawei OLT - vlan 9, 10, 20, 30, 40, 50, 60 are formed in GEM ports
ZTE OLT - vlan 3, 5, 8, 9

Eyeless77, 2023-09-18:
@skon77 Thank you for sharing your experience. VLAN assignment seems true for my case with Huawei OLT. One more step that I've made is turning off "anyservice" via ISP tech support. With "anyservice" turned on multicast traffic comes untagged. With "anyservice" turned off multicast traffic comes tagged with VLAN9.

There is one more caveat I'm struggling with. It seems that IGMP join requests are forwarded within routed VLAN20 and the actual TV stream returns via VLAN9 and multicast.

[image]

Every time I change TV channel, I can see IGMP queries and later then a new source of multicast traffic in VLAN9. I use OPNsense as a router and now the challenge is to forward IGMP queries via VLAN20. I cannot make a bridge of 3 interfaces (STB port, VLAN9, VLAN20), DHCP stops working, it's a known limitation.

labuwx, 2023-09-18:
> One more step that I've made is turning off "anyservice" via ISP tech support. With "anyservice" turned on multicast traffic comes untagged. With "anyservice" turned off multicast traffic comes tagged with VLAN9.

When I first set up the DFP-34X-2C2 my ISP did not push the Multicast operations profile (MIB 309). Then I upgraded my internet subscription, and without me changing anything on the ONU, the multicast frames started to come untagged.
My workaround is to specify a nonexistent vid in VLAN_MANU_TAG_VID, which I guess the ONU would untag if it existed, but this way the ONU keeps the multicast frames tagged in spite of the Multicast operations profile.

vuducdong, 2023-09-19:
@Eyeless77 

[image]

For your case, I think you just need to create a bridge between STB port and vlan20 (remember to turn off IGMP snooping option)

Have you tried: when using stb, does the router lose internet connection? (while using option VLAN_CFG_TYPE=1 (vlan transparent mode) with M110_sfp_ODI_220923 firmware)

Eyeless77, 2023-09-19:
Finally got it working in a routed environment with OPNsense. Described some details in OPNsense community: https://forum.opnsense.org/index.php?topic=35933.msg175397#msg175397