Anime4000/RTL960x #194: rtl960x

hmetro24, 2023-09-16:
Hello guys, I got my self xpon(ODI DFP-34X-2C2) to replace isp ont (adtran 611) , i am getting o5 state, but not able to get internet connection, the isp will not tell the vlan. so i did try to telnet and uart serial using ch341 to the adtran611 , hoping to get some information, but i am stuck because password is not known, user name is admin.
the adtran is using rtl9601B chipset  
this is what i was able to get during the boot up.
https://pastebin.com/WFW9mbFN
any body know what could be the password, or either way to crack it. 
thank you

ToTheCLI, 2023-09-16:
Use this command and see if any VLANs appear:

omcicli mib get 84

hmetro24, 2023-09-16:
i did use omcicli mib get 84
and i got 121. i used vlan 121 and it doesn't connect

ToTheCLI, 2023-09-16:
Also post output of this one

omcicli mib get 171

What method does your original router use to connect (PPPoE or DHCP)?

hmetro24, 2023-09-16:
i will see omcicli mib get 171, 
the adtran is a bridge connected to router using pppoe

ToTheCLI, 2023-09-16:
You might need to set FwdOp 0x02 for the internet VLAN with the following command:
omcicli mib set 84 "VLAN 121 Entity ID" FwdOp 0x02
replace  "VLAN 121 Entity ID" with omcicli mib get 84 Entity ID of VLAN 121

Anime4000, 2023-09-17:
can anyone here list all FwdOp value? I think I lost the PDF referring to that

labuwx, 2023-09-17:
> can anyone here list all FwdOp value? I think I lost the PDF referring to that

[OMCI specification](https://www.itu.int/rec/dologin_pub.asp?lang=e&id=T-REC-G.988-202211-I!!PDF-E) Table 9.3.11-1
| Forward operation | Received frame: Tagged | Received frame: Untagged |
| --- | --- | --- |
| 0x00 | Bridging (a) (no investigation) | Bridging (a) |
| 0x01 | Discarding (c) | Bridging (a) |
| 0x02 | Bridging (a) (no investigation) | Discarding (c) |
| 0x03 | Action (h) (VID investigation) | Bridging (a) |
| 0x04 | Action (h) (VID investigation) | Discarding (c) |
| 0x05 | Action (g) (VID investigation) | Bridging (a) |
| 0x06 | Action (g) (VID investigation) | Discarding (c) |
| 0x07 | Action (h) (user priority investigation) | Bridging (a) |
| 0x08 | Action (h) (user priority investigation) | Discarding (c) |
| 0x09 | Action (g) (user priority investigation) | Bridging (a) |
| 0x0A | Action (g) (user priority investigation) | Discarding (c) |
| 0x0B | Action (h) (TCI investigation) | Bridging (a) |
| 0x0C | Action (h) (TCI investigation) | Discarding (c) |
| 0x0D | Action (g) (TCI investigation) | Bridging (a) |
| 0x0E | Action (g) (TCI investigation) | Discarding (c) |
| 0x0F | Action (h) (VID investigation) | Bridging (a) |
| 0x10 | Action (h) (VID investigation) | Discarding (c) |
| 0x11 | Action (h) (user priority investigation) | Bridging (a) |
| 0x12 | Action (h) (user priority investigation) | Discarding (c) |
| 0x13 | Action (h) (TCI investigation) | Bridging (a) |
| 0x14 | Action (h) (TCI investigation) | Discarding (c) |
| 0x15 | Bridging (a) (no investigation) | Discarding (c) |
| 0x16 | Action (j) (VID investigation) | Bridging (a) |
| 0x17 | Action (j) (VID investigation) | Discarding (c) |
| 0x18 | Action (j) (user priority investigation) | Bridging (a) |
| 0x19 | Action (j) (user priority investigation) | Discarding (c) |
| 0x1A | Action (j) (TCI investigation) | Bridging (a) |
| 0x1B | Action (j) (TCI investigation) | Discarding (c) |
| 0x1C | Action (h) (VID investigation) | Bridging (a) |
| 0x1D | Action (h) (VID investigation) | Discarding (c) |
| 0x1E | Action (h) (user priority investigation) | Bridging (a) |
| 0x1F | Action (h) (user priority investigation) | Discarding (c) |
| 0x20 | Action (h) (TCI investigation) | Bridging (a) |
| 0x21 | Action (h) (TCI investigation) | Discarding (c) |

hmetro24, 2023-09-17:
> You might need to set FwdOp 0x02 for the internet VLAN with the following command: omcicli mib set 84 "VLAN 121 Entity ID" FwdOp 0x02 replace "VLAN 121 Entity ID" with omcicli mib get 84 Entity ID of VLAN 121

this got it working or might be something regarding to flash set DEVICE_TYPE  was giving me the problem, every time i switch it to bridge, reboot, check again it switch back to Router mode. currently i have internet connection not sure if it is on bridge or router mode. even it is DEVICE_TYPE=1 which is router.

ToTheCLI, 2023-09-17:
Keep in mind setting Fwops doesn't presisit after reboot, you would need custom frimware with script on bootup.

Anime4000, 2023-09-18:
@labuwx nice, I edit the README and link to there

hmetro24, 2023-09-18:
I am curious, if it is possible to set MTU in bridge mode, the stick i have is missing mtu settings, but this stick has MTU setting
[image]

vuducdong, 2023-09-18:
> I am curious, if it is possible to set MTU in bridge mode, the stick i have is missing mtu settings, but this stick has MTU setting [image]

What firmware are you using? I see there are 2 port mappings

Yukii1119, 2023-11-17:
> this is just a picture from another port not mine, my question is, is it possible to set mtu size in bridge mode if only have Lan_1?

The PON stick ONU is bridge mode and only supports forward function, but you can configure the mtu size of the onu eth port through olt.

hmetro24, 2024-12-27:
> Which settings did you change to make this work? I am in the UK but cannot get past O3 status

I don't remember much , its been a while, i got it working, but 2 days after i got a call from the ISP, telling me i either remove it and put back the Gpon they gave me, or they will disconnect me. They might find out due to wrong cloning the gpon.
what i remember is i found the correct Vlan the isp is using (omcicli mib get 171), did the cloning with same Mac address, FW and other info of the ISP Gpon. 
and it worked using Bridge mode. 
Other than that , i don't remember.
I would reset it to default/factory settings and start clean. Also every Isp is different

tszyszko, 2026-08-18:
@hmetro24 what isp was it?

hmetro24, 2026-08-18:
> [@hmetro24](https://github.com/hmetro24) what isp was it?

It is bezeq, or partner

closed 2026-08-18 by hmetro24