Anime4000/RTL960x #195: VSOL 2802RH O5 status no internet connection

Faruolo2, 2023-09-19:
Hello, im trying to replace Nokia G-240W-C ONT to VSOL 2802RH and after set all configurations, i could get a VLAN return on omcicli mib get 84 as of:

AP#omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

VlanTagFilterData

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

=================================

EntityID: 0xe102

FilterTbl[0]: PRI 0,CFI 0, VID 1107

FwdOp: 0x10

NumOfEntries: 1

=================================

=================================

EntityID: 0xe103

FilterTbl[0]: PRI 0,CFI 0, VID 1107

FwdOp: 0x10

NumOfEntries: 1

=================================

=================================

EntityID: 0xe10b

FilterTbl[0]: PRI 0,CFI 0, VID 1107

FwdOp: 0x10

NumOfEntries: 1

=================================

AP#omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

ExtVlanTagOperCfgData

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

=================================

EntityId: 0xe01

AssociationType: 10

ReceivedFrameVlanTagOperTableMaxSize: 0

InputTPID: 0x8100

OutputTPID: 0x8100

DsMode: 0

ReceivedFrameVlanTaggingOperTable

INDEX 0

Filter Outer : PRI 8,VID 220, TPID 5

Filter Inner : PRI 8,VID 4096, TPID 0, EthType 0x00

Treatment Outer : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner : PRI 9,VID 1107, TPID 3

INDEX 1

Filter Outer : PRI 15,VID 4096, TPID 0

Filter Inner : PRI 8,VID 220, TPID 5, EthType 0x00

Treatment Outer : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner : PRI 8,VID 1107, TPID 2

INDEX 2

Filter Outer : PRI 8,VID 210, TPID 5

Filter Inner : PRI 8,VID 4096, TPID 0, EthType 0x00

Treatment Outer : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner : PRI 9,VID 1107, TPID 3

INDEX 3

Filter Outer : PRI 15,VID 4096, TPID 0

Filter Inner : PRI 8,VID 210, TPID 5, EthType 0x00

Treatment Outer : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner : PRI 8,VID 1107, TPID 2

INDEX 4

Filter Outer : PRI 15,VID 4096, TPID 0

Filter Inner : PRI 15,VID 0, TPID 5, EthType 0x00

Treatment Outer : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner : PRI 15,VID 0, TPID 2

INDEX 5

Filter Outer : PRI 14,VID 4096, TPID 5

Filter Inner : PRI 14,VID 4096, TPID 0, EthType 0x00

Treatment Outer : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner : PRI 15,VID 4096, TPID 3

INDEX 6

Filter Outer : PRI 15,VID 4096, TPID 0

Filter Inner : PRI 14,VID 4096, TPID 5, EthType 0x00

Treatment Outer : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner : PRI 15,VID 4096, TPID 2

AssociatedMePoint: 0x601

after configuring WAN with 1107 vlan still no internet connection, any1 know what i could do?

ToTheCLI, 2023-09-19:
Similar Issue
 
https://github.com/Anime4000/RTL960x/issues/121

ToTheCLI, 2023-09-19:
Try setting FwdOp to 0x02 for each entityID Alone with the following commands then check for Internet then reboot after no connection
omcicli mib set 84 0xe102 FwdOp 0x02
Then check for connection on VLAN 1899 and 210
No connection reboot
then
omcicli mib set 84 0xe103 FwdOp 0x02
Then check for connection on VLAN 1899 and 210
No connection reboot
then
omcicli mib set 84 0xe10b FwdOp 0x02

Faruolo2, 2023-09-19:
Tried all, still no connection :(

Anime4000, 2023-09-20:
I also experimenting
[image]

may I know, did your ISP have forced specific LAN port of ONT for Internet Access?

here in Malaysia (TM Unifi), this ONT doesn't wok when have LAN Port Provisioning (aka: 4 LAN port ONU Emulation)

Faruolo2, 2023-09-20:
No, just wan port and 4 lan ports to connect

Anime4000, 2023-09-20:
What I can guess is, the `FwdOp: 0x10` is very problematic, 
Your ISP uses `FwdOp: 0x10` for special bridging, since VoIP only available that mean VSOL accept that and ignore the rest.

my fiber wholesale set their OLT to force ONU LAN 2 and disable other ONU LAN, only can hack the bridge is ODI SFU firmware

sad to say, I can't reverse engineering much on VSOL V2802RH.

you need to use ODI Stick with SFU Firmware + Media Converter

d5aqoep, 2023-09-23:
You get a 10.xx.xx.xx IP means your ONU is working perfectly. You just need the correct VLAN. Maybe you can get it from your ISP or ask a service engineer to get it from their backend.

Faruolo2, 2023-09-23:
another user experiencing the same problem said that if he set:
omcicli mib set 84 0xe102 FwdOp 0x02
omcicli mib set 84 0xe103 FwdOp 0x16
omcicli mib set 84 0xe10b FwdOp 0x02

he gets internet via IpoE but no connection on bridge

Faruolo2, 2023-09-23:
> You get a 10.xx.xx.xx IP means your ONU is working perfectly. You just need the correct VLAN of internet. Maybe you can get it from your ISP or ask a service engineer to get it from their backend.

yeah, aparently provisioning is OK. The problem is that this IP range is for VOIP. i confirmed this because if i check directly on ONT provided by ISP, VOIP is coming from a Ip from that range on Vlan 220.

Faruolo2, 2023-09-24:
updating

after executing

omcicli mib set 84 0xe102 FwdOp 0x02
omcicli mib set 84 0xe103 FwdOp 0x16
omcicli mib set 84 0xe10b FwdOp 0x02 

i could get internet conection trough IpoE routing.

Faruolo2, 2023-10-05:
updating 2:

setting FwdOp 0x15 and 0x00 on entity 0xe103 bridge kinda works...

Vsol isnt get any IP trough bridge, returning only 0.0.0.0, but i can connect to internet and some websites like youtube, google works, but other websites like speedtest.com, waveform bufferbloat test doesnt connect.

also download speed seems ok, but upload speed stuck at 0.08mb

Faruolo2, 2023-10-16:
updating 3:

after almost a month using Vsol like this, i has to reboot, and after that i cannot connect again, everytime i set the right FwdOp 0x15 im getting only ipv6 connection and no ipv4.

Any1 know what may be causing this?