Anime4000/RTL960x #197: 05 but problem with vlans

mshrem, 2023-09-23:
So i finally got my DFP-34X-2C2 , got it to show 05 but i have no internet connection. 

2 things to consider 
- i am still using stovk firmware from ODI
- I know for sure my isp uses Vlan 881 for the internet 
```
MIB: 00084

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x1102
FilterTbl[0]: PRI 0,CFI 0, VID 1314
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x110b
FilterTbl[0]: PRI 0,CFI 0, VID 1314
FwdOp:  0x10
NumOfEntries: 1
=================================
```

```
MIB: 00171

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x101
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 8,VID 881, TPID 5
Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 1314, TPID 3
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 881, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 1314, TPID 2
INDEX 2
Filter Outer   : PRI 14,VID 0, TPID 5
Filter Inner   : PRI 14,VID 0, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 4096, TPID 3
INDEX 3
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 0, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 4096, TPID 2
INDEX 4
Filter Outer   : PRI 15,VID 0, TPID 0
Filter Inner   : PRI 15,VID 0, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 2
AssociatedMePoint: 0x101
DscpToPbitMapping:
	0x000000
	0x249249
	0x492492
	0x6db6db
	0x924924
	0xb6db6d
	0xdb6db6
	0xffffff

```

Should i try another firmware ?????

i read [FwdOp](https://github.com/Anime4000/RTL960x/issues/194#issuecomment-1722497606) but i am confused as how to proceed. 

I have pfsense so i know i can tag vlan to WAN if necesary

Any suggestions????

mshrem, 2023-09-23:
If anyone else stumbles upon this , i added the Vlan 1314 on the client side so i can establish the PPPOE connection.

[Here ](https://github.com/Anime4000/RTL960x/issues/26)

closed 2023-09-23 by mshrem

pmcnano, 2024-05-03:
@mshrem I'm struggling with the same (Telmex). Did you end up changing the firmware or did you work with the original firmware?

Running the MIB commands gets me nothing.

mshrem, 2024-05-03:
> @mshrem I'm struggling with the same (Telmex). Did you end up changing the firmware or did you work with the original firmware?
> 
> Running the MIB commands gets me nothing.

Hi there yes i was able to get it to work using the original firmware , no need for changing the firmware at all. 

Are you using any type of media converter ? How are you hooking up the SFP module ?

pmcnano, 2024-05-03:
@mshrem IT JUST WORKED RIGHT NOW ...yaaassss. Thanks man, it was all you and the 1314. I changed the VLAN settings (in the stick) to auto, and then the command 84 actually showed the vlan 1314, then I was able to use it to connect.