Anime4000/RTL960x #199: Get status 05 but immediately errors(DFP-34X-2C2)

dinamicus, 2023-10-05:
Hi guys!
After I get status 05, errors immediately appear.
Parent ME class is not supported in ethernet_frame_extended_pm_handler: EthExtPmData, 0x149
Instance not found in fec_pmhd_pm_handler: Anig, 0x0
the issues are repeated 
it doesn't matter which FW  I use : SFU of  IGD
omcicli mib get 84

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x02
FilterTbl[0]: PRI 0,CFI 0, VID 1672
FwdOp:  0x02
NumOfEntries: 1
=================================
=================================
EntityID: 0x137
FilterTbl[0]: PRI 0,CFI 0, VID 1672
FwdOp:  0x10
NumOfEntries: 1

omcicli mib get 171

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x20
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 3, RemoveTags 0
Treatment Inner   : PRI 0,VID 1672, TPID 6
AssociatedMePoint: 0x101
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
seems like that's a weird issue
has anyone encountered something similar and can help?
spent a lot of time changing settings, etc. but no luck

Anime4000, 2023-10-07:
This could be the first time ever happened... have tried FAKE OK 1 ?

maybe time to dump OMCI and analyzing what cause 
https://github.com/tdmadam/OMCI-for-SFP-Stick

dinamicus, 2023-10-08:
> This could be the first time ever happened... have tried FAKE OK 1 ?
> 
> maybe time to dump OMCI and analyzing what cause https://github.com/tdmadam/OMCI-for-SFP-Stick

there is no difference with FAKE OK 0 or 1
as for OMCI dump, everything looks fine there. there are requests from OLT for creating MIBs and answers from ONU that commands processed successfully.btw as result I get this{RTM_OMCI} OMCI configuration is complited!
so I found MIBs entries that are causing the errors above
1st one is here:(Parent ME class is not supported in ethernet_frame_extended_pm_handler: EthExtPmData, 0x149)
omcicli mib get 334
seems like there is no support for VEIP, Parent Class is 329 .  If delete both entry for mib 334, the error disappears) It’s not clear what to change to, what class to set.
EntityId: 0x0008
IntEndTime: 0
ControlBlock:
        ThresholdDataID: 0x0000
        ParentMeClass: 329
        ParentMeInstance: 0x0601
        AccumDisable: 0x0000
        TcaDisable: 0x0000
        CtrlFields: 0x0001
        TCI: 0x0000
        Reserved: 0x0000
DropEvents: 0
Octets: 0
Frames: 0
BroadcastPackets: 0
MulticastPackets: 0
CrcErrPackets: 0
UndersizePackets: 0
OversizePackets: 0
Octets64: 0
Octets65to127: 0
Octets128to255: 0
Octets256to511: 0
Octets512to1023: 0
Octets1024to1518: 0
=================================
=================================
EntityId: 0x0009
IntEndTime: 0
ControlBlock:
        ThresholdDataID: 0x0000
        ParentMeClass: 329
        ParentMeInstance: 0x0601
        AccumDisable: 0x0000
        TcaDisable: 0x0000
        CtrlFields: 0x0003
        TCI: 0x0000
        Reserved: 0x0000
DropEvents: 0
Octets: 0
Frames: 0
BroadcastPackets: 0
MulticastPackets: 0
CrcErrPackets: 0
UndersizePackets: 0
OversizePackets: 0
Octets64: 0
Octets65to127: 0
Octets128to255: 0
Octets256to511: 0
Octets512to1023: 0
Octets1024to1518: 0

2nd error(Instance not found in fec_pmhd_pm_handler: Anig, 0x0) is due to this MIB 312, 
omcicli mib get 312
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
FecPmhd
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0000
IntervalEndTime: 0
ThresholdData12Id: 0x0000
CorBytes: 0x00000000
CorCodeWords: 0x00000000
UncorCodeWords: 0x00000000
TotalCodeWords: 0x00000000
FecSeconds: 0x0000
=================================

this is from OMIC dump ,no idea what to check regarding this Instance "0" or how to set it to correct one
    Message Identifier, ME Class = OLT-G   (ME=131), Instance = 0
        Managed Entity Class: OLT-G   (ME=131) (0083)
        Managed Entity Instance: 0x0000
    Attribute Mask (0xa000)

therefore by deleting entries in MIB 312 and MIB 334 It makes errors disappear

Anime4000, 2023-10-08:
> therefore by deleting entries in MIB 312 and MIB 334 It makes errors disappear
if delete these, internet is working?

dinamicus, 2023-10-08:
> > therefore by deleting entries in MIB 312 and MIB 334 It makes errors disappear
> 
> if delete these, internet is working?

unfortunately it's down internet isn't working

Anime4000, 2023-10-08:
have you tried `EntityID: 0x137` using `FwdOp:  0x02`

dinamicus, 2023-10-08:
no luck as well

Anime4000, 2023-10-08:
I can't be sure how to solve this, as anyone to replicate the issue, mine quite different either:

```
# omcicli mib get 334
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
EthExtPmData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 329
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VEIP
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0601
AdminState: 0
OperState: 0
InterDomainName: 0x00000000000000000000000000000000000000000000000000
TcpUdpPtr: 0x0000
IansAssignPort: 0
CustomizedSlotId: 0xff
=================================
# omcicli mib get 312
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
FecPmhd
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
```

perhaps try Huawei GPON Stick ?

sweetpotato133, 2023-10-10:
> Hi guys! After I get status 05, errors immediately appear. Parent ME class is not supported in ethernet_frame_extended_pm_handler: EthExtPmData, 0x149 Instance not found in fec_pmhd_pm_handler: [...]

I'm getting the same problem, I've been trying to make it to work for 2 days now. It's giving me a headache.. My ISPs OLT is fiberhome.

dinamicus, 2023-10-11:
> I'm getting the same problem, I've been trying to make it to work for 2 days now. It's giving me a headache.. My ISPs OLT is fiberhome.

could you share here your settings received from ISPs OLT just to compare
omcicli mib get 312
omcicli mib get 334
omcicli mib get 171
omcicli mib get 84