Anime4000/RTL960x #200: ODI DFP-34X-2C2 - Does not boot/start with fiber connected

luizosorio, 2023-10-07:
Firstly, I apologize if there is a topic about this but, I wasn't able to find it.

I have problems with electrical fluctuations in my region and sporadic power outages. This occasionally leads to a reboot of my devices. In these cases, I ended up encountering a situation I don't know if it was a problem with my SFP, a bug, a misconfiguration, or expected behavior.

When my device is restarted, it seems to me SFP does not boot (when I mention SFP doesn't seem to boot/start, it's because the device's physical interface never gets UP). After several tests, I noticed this behavior only occurs when the ISP fiber is connected, i.e.: 

Every time my device restarts for any reason, I need to disconnect the fiber from the ISP, and by some means force the SFP to physically restart (either by restarting the device or disconnecting the SFP and reconnecting again) and wait for the SFP to boot/start to then finally connect again ISP fiber.

SFP has the same behavior on my mini pc or directly on a media converter.

SFP: ODI DFP-34X-2C2
Firmware: 220923

----------------------------------------------------------------------
I'm new here in the community but thanks to your work I finally managed to get rid of my ISP damn ONT. (Country: Brazil, ISP: Oi, ONT: G-240W-C)

akw28888, 2023-10-15:
Use this command to check lan sds mode: ```cat /proc/lan_sds/lan_sds_cfg``` when you can connect to the stick.
Then change the LAN_SDS_MODE value to make sure the stick will not stuck in other modes when booting.

Anime4000, 2023-10-15:
acceptable answer by @akw28888

Yukii1119, 2023-11-17:
Here is direct factory of ODI module, further question you can contact +86 18138129117