Anime4000/RTL960x #206: ODI DFP-34X-2C2 with device name as ODI DFP-34G-2C2

fvp, 2023-11-12:
Hi,

I bought a DFP-34X-2C2, the stick on the SFP states that it is DFP-34X-2C2, but the device name in the UI is DFP-34G-2C2.

Is there a way to check that I got a real DFP-34X-2C2 ?

[image]

Thanks,
Felipe

longthanhtran, 2023-11-12:
you can try to update `Product Class` in settings section to see if the Device Name is updated. (remember to reboot after update)

fvp, 2023-11-12:
Thanks for the suggestion, but what I am looking for is if there is a way to check that the device that I got is really a 34X-2C2 and support both gpon and epon. Or they mislabeled the device 34G-2C2 as a 34X-2C2

longthanhtran, 2023-11-12:
Got it, I think if the label on your stick is 34G-2C2 (compare to the detail from [gpon-hack](https://hack-gpon.org/ont-odi-realtek-dfp-34g-2c2/)) then it's fine. Just use as usual

fvp, 2023-11-12:
Hi, it is the opposite, the label is 34X-2C2 and the UI and SPF info is 34G-2C2

sayantanguharoy, 2024-01-28:
Any upload issue with this stick ?

Anime4000, 2024-01-29:
> Any upload issue with this stick ?

slow upload speed?