Anime4000/RTL960x #208: DFP-34X-2C2 becomes unresponsive after a few minutes

rndm2, 2023-11-17:
Hi 

I have exactly the same problem as described here: https://github.com/Anime4000/RTL960x/issues/189 
At a certain point (minutes-hours), the CPU load goes sky-high, and the stick stops responding.

And stick also causes random packet loss.

That topic is closed, so I am creating a new one. 

Here is the `top` from the stick just before it stops responding:
[image]

Config: 
```
# flash all
<Config Name="ROOT">
 <Dir Name="HW_MIB_TABLE">
  <Value Name="SUPER_NAME" Value="adsl"/>
  <Value Name="SUPER_PASSWORD" Value="realtek"/>
  <Value Name="TEST_MAC_ADDR" Value="realtek"/>
  <Value Name="ELAN_MAC_ADDR" Value="00e04c867001"/>
  <Value Name="WLAN_MAC_ADDR" Value="00e04c076802"/>
  <Value Name="MAC_KEY" Value=""/>
  <Value Name="RESTORE_DEFAULT_EN" Value="1"/>
  <Value Name="WAN_PHY_PORT" Value="4"/>
  <Value Name="LASER_POLARITY_TYPE" Value="0"/>
  <Value Name="PON_DETECT_ENABLE" Value="1"/>
  <Value Name="HW_RF_TYPE" Value="7"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_CCK_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_A" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_HT40_1S_B" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT40_2S" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_HT20" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_TX_POWER_DIFF_OFDM" Value="00,00,00,00,00,00,00,00,00,00,00,00,00,00"/>
  <Value Name="HW_WLAN0_11N_TSSI1" Value="0"/>
  <Value Name="HW_WLAN0_11N_TSSI2" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER" Value="0"/>
  <Value Name="HW_WLAN0_11N_THER_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_PA_TYPE" Value="0"/>
  <Value Name="HW_WLAN0_REG_DOMAIN" Value="1"/>
  <Value Name="HW_WLAN0_11N_TRSWITCH" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_C9" Value="0"/>
  <Value Name="HW_WLAN0_11N_TRSWPAPE_CC" Value="0"/>
  <Value Name="HW_ANT_DIVERSITY" Value="0"/>
  <Value Name="HW_TX_ANT" Value="0"/>
  <Value Name="HW_CS_THRESHOLD" Value="11"/>
  <Value Name="HW_CCA_MODE" Value="1"/>
  <Value Name="HW_PHY_TYPE" Value="0"/>
  <Value Name="HW_WLAN_LED_TYPE" Value="7"/>
  <Value Name="BYTE" Value="255"/>
  <Value Name="WORD" Value="65535"/>
  <Value Name="DWORD" Value="4294967295"/>
  <Value Name="INT1" Value="2147483647"/>
  <Value Name="INT2" Value="-2147483648"/>
  <Value Name="WIFI_SUPPORT" Value="0"/>
  <Value Name="HW_SERIAL_NO" Value="000000000002"/>
  <Value Name="HW_FON_KEYWORD" Value="63,62,68,4a,75,3c,65,cd,7b,75,e5,2d,a9,80,72,8c,97,fc,bd,9b,37,51,d9,e3,82,92,68,a1,07,ae,a0,b5"/>
  <Value Name="HW_WLAN0_11N_XCAP" Value="0"/>
  <Value Name="HW_WLAN0_11N_XCAP_2" Value="0"/>
  <Value Name="HW_WLAN0_11N_TARGET_PWR" Value="0"/>
  <Value Name="HW_WLAN0_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN1_COUNTRYCODE" Value="1"/>
  <Value Name="HW_WLAN0_COUNTRYSTR" Value="US"/>
  <Value Name="HW_WLAN1_COUNTRYSTR" Value="US"/>
  <Value Name="HW_CWMP_MANUFACTURER" Value="MSTC"/>
  <Value Name="HW_CWMP_PRODUCTCLASS" Value="HGU"/>
  <Value Name="HW_HWVER" Value="GPT-2541GNAC"/>
  <Value Name="GPON_ONU_MODEL" Value="HGU"/>
  <Value Name="LAN_SDS_MODE" Value="1"/>
[... 145 lines cut ...]
  <Value Name="VPI" Value="0"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=5-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="43"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=6-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="51"/>
 </Dir>
 <Dir Name="AUTO_PVC_SEARCH_TBL"> <!--index=7-->
  <Value Name="VPI" Value="8"/>
  <Value Name="VCI" Value="59"/>
 </Dir>
 <Dir Name="PPPOE_SESSION_TBL">
 </Dir>
 <Dir Name="RIP_TBL">
 </Dir>
 <Dir Name="DHCP_RESERVED_IPADDR_TBL">
 </Dir>
 <Dir Name="EPON_LLID_TBL">
 </Dir>
</Config>
```

Anime4000, 2023-11-18:
Have you tried using other firmware? by default ODI ship switch bridge firmware, try HGU one and see what happen

rndm2, 2023-11-18:
You mean that last in a list? Hybrid?

rndm2, 2023-11-18:
Same with M114_sfp_ODI_hybrid_221209 
It works 5-15 minutues. And then everything freezes.

Anime4000, 2023-11-18:
I am sorry can't be any help, I can't replicate your issue.
Perhaps someone can help

rndm2, 2023-11-23:
It looks like erasing the config partition and reverting to defaults fixed the problem. 
I also didn't change management IP this time. Read through issues that could cause problems (I don't really understand how). 

Firmware is 220923. Btw, I don't know if it is best or not

closed 2023-11-24 by rndm2