Anime4000/RTL960x #211: DFP-34X-2C2 change EEPROM info

shuher21, 2023-11-23:
Hello, 
How to change EEPROM info for DFP-34X-2C2?

"DFP-34X-2C2" it not present in router supported list (Optical Det), here is part of router console log:
```
Opticaldet: optical module detected on i2c bus 0:
Opticaldet Unknown Transceiver
Module Form Factor: SFP/SFP+
Module Type       : UNKNOWN
Vendor Name       : OEM             
Part Number       : DFP-34X-2C2     
Part REV          :     
************************************************************************
* Opticaldet: Unknown optical module - using default configuration     *
* Please make sure the optical module is correct for your connection   *
************************************************************************

Optical detection module loaded.
DDMI data is not ready!
Failed to read DDMI EEPROM2 through I2C.
```

Anime4000, 2023-11-26:
like this, you need use SFP Programmer, change Vendor, etc.. and recalculate checksum

shuher21, 2023-11-27:
Thank you,

At the time I made the post, I thought that DFP-34X-2C2 has EEPROM emulating software (sfpapp) like V2801F/CIG G-97S/Nokia G-010S-Q and similar. 
But looking at other Issues I found the post https://github.com/Anime4000/RTL960x/issues/27
where got the confirmation, that DFP-34X-2C2 has ordinary EEPROM.

Unfortunately, I have no SFP Programmer, but I will try to run i2c-tools binary directly on my router. If I have success, I will share the steps.

Anime4000, 2023-11-29:
yes, ODI has normal EEPROM,

please do share the step how to change all SFP info value

shuher21, 2023-11-30:
As I do not have SFP Programmer, first i tried to use ethtool, but In ethtool eeprom dump I didn't find vendor-related information, it seems ethtool has access to other EEPROM pages.

What I tried to do next it to use a modified script from the post "https://github.com/Anime4000/RTL960x/issues/27#issuecomment-1115266208" to change vendor information:
using "diag i2c set ..." commands

it shows changed Vendor related Information inside ODI SFP, but will not take effect when using ODI SFP, in the router.

So, I guess I have to find SFP Programmer...

shuher21, 2023-12-01:
While I try to find i2c programmer or other way to change EEPROM 

I'm just wondering what I changed with the "diag i2c set..." command? Because these changes are saved even after the module is turned off.

shuher21, 2023-12-04:
I finally managed to use my router as I2C programmer using the router SFP+ port.

But the way no such straight as just use specialized I2C programmer devices. (Now I understand why nobody suggests other ways except specialized I2C Programmer device+soft, like RasberyPI and so on)

Will try to briefly describe a long story of the possible ways how to change EEPROM contents on SFP module for whom who don't have a specialized I2C Programmer device, but have time and patience just to change some bytes on SFP EEPROM one time.

Checked boot log for my router and found the following messages: 
`bcm_i2c: sfp i2c device found registered for bus 0 address 0xa0
bcm_i2c: sfp i2c device found registered for bus 0 address 0xa2`

this suggested that my SFP module can be reached by I2C interface

As my router software is based on OpenWRT branch, first I tried to setup openwrt i2c-tool package, but later found that i2c command already present in Busybox which you can download from official site. 

https://busybox.net/downloads/binaries/1.28.1-defconfig-multiarch/

(just choose the corresponding architecture)

Form that moment you can check device on I2C bus:
`/busybox-armv8l i2cdetect -y -r 0

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 

10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

50: UU UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

70: -- -- -- -- -- -- -- -- `

The "UU" typically means that a driver is bound to the device at that address. 
Further, it was a challenge to find a way how to make the kernel module to release 0x50 address. Because bcm_i2c module doesn't appear in the "lsmod" output and can't be unloaded using the "rmmod" command.

Headcrabed, 2024-03-03:
Well, if you are using mainline openwrt, it would be much easier @shuher21 :
https://github.com/Anime4000/RTL960x/discussions/250

shuher21, 2024-03-05:
No the router I using is not a mainline openwrt, it is Zyxel AX7501-B0 with proprietary firmware, and one of the ways to make the router work with DFP-34X-2C2 is to change DFP-34X-2C2 EEPROM device information for one of supported by the router (I chose UF-RJ45-1G).

and what I describe in the issue is how to change EEPROM on DFP-34X-2C2 without having a Programmator.